sinara-hw/sinara #166: Sayma Signal Integrity

gkasprow, 2017-03-03:
I'm trying to optimize JESD signal chain.
I use HL9.4.1, models for GTX, all connectors and DAC.
Here is how the eye diagram looks like without optimisation and via matching

[image]

And with matched vias and optimised traces

[image]

but still I cannot meet BERR requirement, even using FR408 PCB material
it seems I have to experiment with other trace widths - there is always trade-off between skin effect losses and dielectric losses.
So the without pre-emphasis on Tx side the eye diagram looks like this:
[image]

With simulated 3-tap  pre-emphasis  BERR is 3e-8
[image]
5-tap gives BER=9e-14 assuming  jitter = 1% of UI
[image]

So theoretically we meet the rquirement, but the question is haw GTX pre-emphasis has to do with HL simulation and how many taps GTX has.

gkasprow, 2017-03-03:
I can specify GTX taps/parameters, simulation was made with "synthesize optimal values" option enable.

gkasprow, 2017-03-03:
Here is bathtub plot
[image]
Some JESD204B specify mask at 55mV like [here](http://archive.eetasia.com/www.eetasia.com/ART_8800685415_499501_TA_28e33ab7_3.HTM)
Our DAC specifies eye mask 
[image]
So if we apply its settings we get (without pre-emphasis and with 1% UI jitter)
[image]

gkasprow, 2017-03-03:
So the question is - shall I continue to meet JESD204B channel specification or we should only satisfy our DAC :)

jordens, 2017-03-03:
A few questions from my side:

* How do the dB values of pre-emphasis in the GTH transcievers relate to the tap numbers you are working with?
* Can you localize the cause for the degradation? The long traces, impedance steps due to corners/proximity, non-linearities/losses in the material, or is it the connector?
* What about the ADC JESD link?

gkasprow, 2017-03-03:
A few questions from my side:
>
>    - How do the dB values of pre-emphasis in the GTH transceivers relate
>    to the tap numbers you are working with?
>
> This is a question I want to ASK. I don't know GTX/GTH transceivers in
such detail

>
>    - Can you localize the cause for the degradation? The long traces,
>    impedance steps due to corners/proximity, non-linearities/losses in the
>    material, or is it the connector?
>
> I removed piece by piece parts of my model and it seems it is combination
of all these components.
This simulation is after optimisation of transmission channel

>
>    - What about the ADC JESD link?
>
> It is the same model, didn't analyse it yet. The speed rate is much slower
so I don't expect any issues.

gkasprow, 2017-03-03:
And here is the model extracted from both PCBs
[image]

gkasprow, 2017-03-03:
For ADC channel running at 8Gbps I do not observe any issues. With 1% UI jitter, no pre-emphasis I got perfect plot
[image]
[image]

gkasprow, 2017-03-03:
Even with standard JESD204B mask BER is 7.7e-13
[image]

jordens, 2017-03-03:
@gkasprow sorry for the confusion. To (re)confirm: we only need 10 Gb/s lane rate (not 12.5 as I said) for the DACs. You can choose from your experience what margin you add to that. But the base rate is 10 Gb/s (for 1 GHz sample data rate).

gkasprow, 2017-03-03:
@jordens that's great. I did analysis for 14G and it looked very bad.

abyszuk, 2017-03-04:
GTH transceivers allow for directly setting pre/post-emphasis in [dB] units.
PREcursor has (0; 6.02) dB range with 20 steps.
POSTcursor has (0; 12.96) dB range with 31 steps.

Number of filter taps is defined only for RX path.

jbqubit, 2017-03-14:
@gkasprow Is this Issue now resolved? If so please close.

closed 2017-04-16 by hartytp