sinara-hw/sinara #169: Upconverting AFE Mezzanine

hartytp, 2017-03-04:
Initial design notes for an upconveritng AFE mezzanine likely to be funded by Oxford:

* RF input from DAC: 50MHz to 500MHz
* Microwave output: 2.5GHz to 3.5GHz using LO at 3GHz
* 1 microwave output channel per mezzanine (2 DAC channels) using IQ mixer (2nd output SMA is DNP/blanked off on front panel)
* LO fed in through SMPs from Sayma RTM (can use either clock mezzanine, or external source fed to the clock SMA on Sayma/Baikal front panel). This will be a PECL square wave @~3GHz
* Inputs: space allowing we will have 3 paths, with resistor jumpers to select between
  * 2 x InAmps as Allaki
  * 2 x direct inputs to ADC as Allaki
  * 1 x demodulated input to ADC, using IQ mixer (no InAmp for this path)
* Otherwise, design as close to Allaki as possible
* Codename TBD

We also considered using 90-deg hybrids + IQ mixers to have 1 microwave output per 1 DAC channel. This improves channel cost/density at the expense of flexibility. For the time being, we decided to go for improved flexibility.

Please post if you have any comments/questions.

closed 2017-03-08 by hartytp

jordens, 2017-03-08:
Just as a sidenote: You also get twice the bandwidth for this type of IQ mixing.

hartytp, 2017-03-29:
Now funded by Oxford. @sbourdeauducq @jordens do you want to suggest a codename?

hartytp, 2017-03-29:
One other comment on the design: use a "nice" IQ mixer for this, meaning good isolation, balance, etc. (i.e. don't just pick the cheapest one...)

sbourdeauducq, 2017-03-29:
Nanoga

sbourdeauducq, 2017-03-29:
And great news about the funding!

hartytp, 2017-03-29:
Wiki updated to include Nanoga. This issue is now defunct.