sinara-hw/sinara #170: AC couple clock balun inputs

hartytp, 2017-03-08:
Please can we AC couple (1nF, ideally C0G) all inputs to TCM2-43+ baluns used for external clock inputs?

During our PLL tests, we tried the following: Wenzel oscillator conencted to TCM2-43+ balun input. 50Ohm terminate one output, and connect the other to a phase noise meter. The close-in phase noise was significantly worse than without the Balun, and contained a lot of content at offset frequencies from the carrier that matched power line harmonics. AC coupling the Wenzel oscillator into the Balun fixes this. 

The most plausible explanation is that the Balun presents a short between input and ground at DC, which the Wenzel oscillator doesn't like. 1nF presents a suitable high-pass filter/DC block to resolve this.

This should be applied to all clock/LO inputs on Baikal/Sayma. (Obviously, changes on Sayma can wait until the next revision).

jordens, 2017-03-08:
I can also see the balun non-linearly mixing DC and carrier through e.g. saturation-like effects.
You can test your hypothesis by putting an attenuator between balun and oscillator.
Also, I remember reading something about different levels of imperfections of the different balun configurations due to common mode effects like this. But I can't find the reference now.
But yeah. AC couple it.

hartytp, 2017-03-08:
> I can also see the balun non-linearly mixing DC and carrier through e.g. saturation-like effects.
> You can test your hypothesis by putting an attenuator between balun and oscillator.

We did consider this, and haven't taken enough data to rule it out. But, IIRC the TCM2-43X+ are non-ferrite based and so should have pretty minimal saturation affects at this kind of power level....

gkasprow, 2017-03-08:
All clock inputs and outputs on Sayma are already AC-coupled

>
>

hartytp, 2017-03-09:
> All clock inputs and outputs on Sayma are already AC-coupled

Yes, sorry, see my edit. Can you reduce the AC coupling capacitors to 1nF (ideally C0G) though to high-pass filter audio frequency noise? With the current 100nF coupling caps, too much of this still gets through.

gkasprow, 2017-03-09:
done

hartytp, 2017-03-13:
@sbouhabib Please close this issue once you're updated Baikal

jbqubit, 2017-03-14:
Recap: two modifications needed before this can be closed. 
- [ ] update to Sayma v0.2 
- [ ] update to Baikal

gkasprow, 2017-03-14:
already did it on Sayma.

jbqubit, 2017-07-25:
@sbouhabib Please update Baikal and close Issue.

closed 2017-08-01 by sbouhabib