sinara-hw/sinara #174: PCB_3U_DAC noise

gkasprow, 2017-03-19:
In the spec on Wiki there is specified noise density of the DAC: <1nVrtHz
Isn't there a typo? 50Ohm resistor noise is 0.9nVrtHz. Multichannel DACs have ~250nVrtHz.
We could expect 1uVrtHz, not nV

jordens, 2017-03-19:
That's not a typo and not exceedingly hard to achieve. It's at >1MHz, a decade from the filter knee at 100 kHz.

hartytp, 2017-03-19:
> @ In the spec on Wiki there is specified noise density of the DAC: <1nVrtHz Isn't there a typo 50Ohm resistor noise is 0.9nVrtHz. Multichannel DACs have ~250nVrtHz. We could expect 1uVrtHz, not nV

As @jordens says, this isn't a typo, it's our (target) specification. This was previously discussed in #142 (see posts about heating rate). We have a little room to slacken this off, but not much.

AFAICT, this should be achievable using a combination of appropriately designed active and passive filters. 

The drift and noise specifications will make design for this DAC somewhat challenging.

gkasprow, 2017-03-20:
OK, i got it.
This noise spec is above 1MHz.
The thing that worries me is low frequency noise where flicker noise is
dominant. And not all components are specified for such low frequency.
The only way is to verify it with real hardware.

hartytp, 2017-03-21:
> @gkasprow The thing that worries me is low frequency noise where flicker noise is dominant. The only way is to verify it with real hardware.

Agreed, the current specification of 10ppm noise in 10Hz-20MHz bandwidth is a little ambitious. But, even if we can't quite meet this, I do think we should be able to get fairly close. 

I'm happy with a "best effort" approach for the low frequency noise: make it as good as conveniently possible in the design through choice of DAC, reference, etc and then we'll see where we are.

This seems resolved, so I'm closing this issue. Feel free to reopen if there is more to discuss.

closed 2017-03-21 by hartytp