sinara-hw/sinara #176: AMC box with RTM

gkasprow, 2017-03-25:
We are designing AMC box that holds single AMC and single RTM board.
It is for another project but as I promissed, will include ARTIQ requirements as well.
Here is brief specification

dimensions: width 10U, length = RTM+AMC, heigth: tbd
supply: 12V external with DIN jack, there are plenty of suitable COTS AC/DC supplies 
Power switch
Interfaces:
2x SFP (port 0, port 1)  specified for 10G
1x QSFP (Fat Pipe) specified for 40G
2x RJ45 (MLVDS), compatible with 3U RJ45. Optional.
2x RS232 RJ45 routed to MLVDS  (required by my project). Can be physically same connectors as MLVDS.
1x RJ45 with management Ethernet port implementing IPMI. Optional.
LPC1764FBD100 for management. Can be not mounted for completely passive operation
JTAG header
2x SMA with baluns routed to TCLKA and TCLKB
2x eSATA (port 2, port 3)
Small  4x10 LCD display connected to AMC I2C.
1x SMA routed to clock mux which selects between SMA or 100MHz internal clock. Connected to FCLK
the design will be based on http://creotech.pl/en/produkt/dual-amc-box-2/ with reduced functionality, weight and cost. Mechanically it would look similar, only height will be reduced by almost 50%

I'd like to open discussion - what are ARTIQ requirements for such box?
I imagine you want completely passive setup with QSFP of SFP
Would RJ45 LVDS ports be useful?

Below is some draft of the front panel. SATA ports would be connected by dedicated cable to the front panel socket.
[image]

jordens, 2017-03-25:
Nice!

Questions:

* Is there some kind of mini-rf backplane in there? If we can always use the SMA connector on Sayma RTM for clocking then not having one is fine AFAICT.
* Is that external/internal 100 MHz thing drving the RTM clock? What is it driving?
* If the IPMI controller behaves and does not need any configuration and handling for deployment or during operation then it does not need to be removed.

Remarks:

* So far we *need* almost nothing from those features.
* We could make use of one SFP on port 4 (fabric D) to drive a Sayma AMC DRTIO satellite. But since you seem to have a QSFP on there that would prevent us from using it and present a significant inconvencience because it makes a Sayma from a "big" uTCA rack bitstream-incompatible with this one. Maybe a capacitor mux to select between that QSFP and an SFP?
* Maybe we could make use of an SFP on port 5 (fabric E) for DRTIO as well. Very low priority.
* VHDCI would potentially be more useful for us than 4xRJ45. I would feed all 16 differential signals from AMC_P2P and AMC_MLVDS to one VHDCI connector (if there is space etc..). Medium Priority.
* If there is space, the same goes for FAT_PIPE2 (wire as VHDCI). Really low priority.

gkasprow, 2017-03-25:
I don't need QSFP in my application so can replace it with 2 or even 4 SFPs.
We have only 8 LVDS links so they fit to 2x RJ45.
Another two RJ45 are RS232
The 100MHz circuit is driving PCie clock (Fabric F). I need it for PCie over fibre.
We can implement mini-RF backplane but it would only have single 100MHz input which we have on the reare panel anyway. The RF connectors are quite costly. 
For FP2, P2P and all other stuff I can make dedicated VHDCI connector, but on the rear panel.

jordens, 2017-03-25:
@gkasprow ACK.

* An SFP on port 4 would go a long way towards simplifying things. It is AFAICT the only transceiver link between Metlino and Sayma that we have in a rack. We'd use that for DRTIO.
* There is no purpose allocated to any of other ports (P2P, MLVDS, port 5, FP2, etc.). You can wire them as you see fit. SFPs on transciever links are certainly welcome. QSFP is not immediately useful for us unless those QSFP-to-4xSFP fan-outs are known to be absolutely transparent and work fine with just one SFP filled and used. A bit of RJ45 doesn't hurt, nor does VHDCI. But I wouldn't want to drive cost with this.
* RF backplane does indeed seem useless if we have the SMA clock connector on the RTM panel.

jordens, 2017-03-25:
Na. The QSFP-to-SFP break-outs all seem to be direct attach, i.e. short distance copper.

gkasprow, 2017-03-26:
I will make it in following way using KISS approach:
- RJ45 would be either RS232 ( routed to LVDS signals) or LVDS. They are mutually exclusive so using resistor placement variant we can have one or another.
- I will install 4 SFPs for fat pipe nad 2 SFPs for Port0 and Port 1.
For my project I won't use SFPs for FP, for Sayma we will install only required SFPs
- if someone needs some other configuration, the PCB is very simple and can be easily modified.

gkasprow, 2017-03-26:
@sbourdeauducq where do you want me to commit AMC Box design files?

sbourdeauducq, 2017-03-26:
Where are the design files for those?
http://creotech.pl/en/produkt/dual-amc-box-with-rtm-2/
http://creotech.pl/en/produkt/dual-amc-box-2/
If you already have a place for those designs, I'd think this one should go there, since it is fairly generic and only loosely connected to ARTIQ/m-labs.

gkasprow, 2017-03-26:
they are here http://www.ohwr.org/projects/amc-ubackplane-sfp/repository

hartytp, 2017-04-16:
Nothing to do here, so closing. 

@gkasprow Please could you put a brief description + link on the Wiki once you've completed the designs (I've added a page for uTCA chassis)?

closed 2017-04-16 by hartytp

gkasprow, 2017-04-18:
done

jbqubit, 2017-04-27:
Thanks. Looks good.

hartytp, 2017-07-13:
@gkasprow Is this finished now? If so, how can I buy one?

gkasprow, 2017-07-13:
Not yet, still playing with mechanical issues. I have to fit 230V supply
inside because it is now required by another project.

hartytp, 2017-07-13:
Okay. I'm not in a hurry for this, but we'll probably buy a few once they're ready and tested.

gkasprow, 2017-12-06:
@hartytp the boxes arrived. After some mechanical fixes all boards fit well. Next thing is testing of the cooling.
[image]

hartytp, 2017-12-06:
Great!