sinara-hw/sinara #178: obsolete SDRAM memory chip in Sayma

gkasprow, 2017-04-11:
@sbourdeauducq please check if MT41K256M16TW-107:P can be soldered intead of MT41J256M16HA
Micron recommentds it as replacement but it has slightly different timings.
It is still available but soon will be difficult to get. So I want to use new part to keep the firmware compatible.

sbourdeauducq, 2017-04-11:
Fine, we can change the gateware to adapt to various SDRAM timings. What timing parameter are you worried about?

gkasprow, 2017-04-11:
Nothing particular in mind. Just wanted to make sure that you accept it.

closed 2017-04-11 by gkasprow

jbqubit, 2017-04-13:
@gkasprow Where do you plan to source MT41K256M16TW-107:P? Minimum quantity according to Octopart availability is poor and minimum quantity > 1000.

https://octopart.com/search?q=MT41K256M16TW-107%3AP

gkasprow, 2017-04-13:
We buy such quantities for other projects in the company and have them on stock

Main problem with SDRAM is that their availability changes every few months when new silicon revision is introduced and part number changes.

There are plenty of replacement chips from Hynix, Samsung, Nanya and other vendors.

In this case it can be easily replaced by  <https://octopart.com/mt41k256m16tw-107+xit%3Ap-micron-71345159> MT41K256M16TW-107 XIT:P since it has same timing, but commercial temp range. The one my coleagues from CTI proposed is industrial temp grade.

It is new part so are not available in popular catalogues yet.

jbqubit, 2017-04-13:
That's good to know. Thanks @gkasprow !