sinara-hw/sinara #179: Allaki SMA connector availability, cost

jbqubit, 2017-04-13:
I got an email from Tomasz Krzaczek at creotech.pl asking about specific components on Allaki. 

p/n J01151A0451 and J01151A0921 are SMA connectors. Some downsides of that choice appear to be
- not many suppliers says octopart
- high cost, >$20/ea

@gkasprow Are there alternatives?

gkasprow, 2017-04-13:
they are on stock in RS, didn't find any alternatives.
There are low cost similar ones in china stores but need to be tested

gkasprow, 2017-04-13:
Anyway, 20$ per connector working at 6Ghz is not much.

jbqubit, 2017-04-13:
# p/n J01151A0451
- uk.rs-online.com says stock is 2 and lead time is 7/6/2017 

# p/n J01151A0921 
- uk.rs-online.com says stock is 8 and lead time is 7/6/2017 

@jordens @sbourdeauducq do you have plans to use SMA on Kasli >> 500 MHz? I'm surprised there are not many options for SMA connectors.

gkasprow, 2017-04-13:
Allaki is space constrained and require quality SMAs for DAC outputs. 
For all other applications we use standard RA SMAs that are available from many vendors.

jbqubit, 2017-04-13:
Sounds good. Do you have access to small quantities for building prototypes? 

Broader story is we need to decide on expected production number ASAP so parts can be pre-ordered.

gkasprow, 2017-04-13:
Since we will need hundreds pieces, we will order them directly

jbqubit, 2017-04-13:
Closing Issue. Will use other mechanism to poll users for expected channel count.

closed 2017-04-13 by jbqubit

gkasprow, 2017-04-13:
We have a few pieces in my lab. Anyway, these connectors are very popular in RF world so will find another source of them.