sinara-hw/sinara #182: 3U ADC schematics

gkasprow, 2017-04-15:
@hartytp I sketched schematics of the ADC. Please look at it. All comments are welcome.
[link](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_ADC/3U_ADC.PDF)

hartytp, 2017-04-16:
Wow, you work quickly Greg!

I won't comment on the digital bits, as @jordens and @sbourdeauducq are better placed than me to do that.

Asside from the usual bits of schematic tidying (labelling filter poles, removing the out of date description of the board etc), which can be done later:

- [ ] Please can you add an RC filter between the InAmps and the ADCs. IIRC, for smaller input ranges (InAmp gain of 100 or 1k), the overall noise floor will be limited by the InAmp's white noise unless we limit the bandwidth here. I'd go for a 250kHz pole, as that should be plenty fast for our applications. Users who want more bandwidth can leave this filter 0R/NC.

- [ ] Please could you move the CMCs into the RFI filter section, and correct the filter annotations to include the extra poles due to the CMCs (these give an extra CM pole at 2pi * 1.5k * 6.5mH=61kHz, and a higher frequency DM pole due to the leakage inductance, right?).

- [ ] If there is room, please can you add an extra DIP switch to allow the input ground to be shorted to circuit ground? Since the InAmp amplifies the CM voltage before subtracting it, if we keep the grounds completely floating then even mV differences between input and circuits grounds will cause issues on the highest gain setting. If possible, adjust the LEDs to indicate both termination and ground-floating.

Other than that, looks good to me.

gkasprow, 2017-04-18:
@hartytp, the InAmp's first stage does not amplify the CM signals. Notice that the feedback loop works only for diff signals. For CM signals the gain is 1 for all gain settings. Second stage has also gain of one or so for diff signals.
The only place where shorting to ground would make sense is before the CMC.

hartytp, 2017-04-18:
> For CM signals the gain is 1 for all gain settings.

Apologies, I misremembered something from a different data sheet I'd read recently. Agreed, there is no problem here (of course).

> The only place where shorting to ground would make sense is before the CMC.

I don't see a good use-case for this, so let's leave it out (unless you have an application in mind?).

hartytp, 2017-05-02:
@gkasprow Should we add a screening can to this board as well (same as DAC)?

gkasprow, 2017-05-02:
@hartytp 
It makes sense in high-impedance circuits and where inductive components are used that can pickup external noise. Since our ADC has hi-Z input, it's worth adding such option. We don't have to install it.

hartytp, 2017-05-02:
Thanks @gkasprow!

gkasprow, 2017-05-02:
We can use same shields for ADC as well. They fit without serious board modification

hartytp, 2017-05-14:
Thanks for posting the revised ADC schematics @gkasprow. That looks really nice.

Some minor comments:

- Can you remove the "please add list of IC names" comment on the schematic. IMO, this design is simple enough that a list of IC names is not required/helpful.
- Out of curiosity, why did you go for +-15V here, and not copy the DAC voltages? I don't mind about this decision, I'm just curious.
- Can you make the power budget annotation a bit clearer, like the (very nice) one on the DAC schematic?
- Is it easy to remove the screening can to access the termination switches?

gkasprow, 2017-05-14:
@hartytp 
I think we can follow DAC voltages. I simply used same values as on a devkit.
The screening can will have openings and switches will protrude from it.

hartytp, 2017-05-14:
@gkasprow Great, that sounds good to me.

As with the DAC, let's give other users until the end of **_Monday_** to provide feedback on the design.

Thanks again for all the good work Greg.

gkasprow, 2017-05-14:
[image]

closed 2017-05-15 by hartytp

jbqubit, 2017-05-22:
@hartytp Please update the wiki to reflect the locked-in specifications for Novogorny.

hartytp, 2017-05-22:
It's on my to do list. I'll probably wait until we've finished the first prototype round incase there are any significant changes.

jordens, 2017-05-22:
Just FYI: Digital and analog concept for Novogorny look good to me.