sinara-hw/sinara #183: Future RTM card for sc qubit applications

dhslichter, 2017-04-20:
Analog Devices has come out with a new set of chips (AD916x) which support data update rates of 12 GSPS, with instantaneous bandwidths of >1 GHz and a variety of interpolation factors.  These are suitable for direct output of signals with frequencies up to ~8 GHz (depending on how much power one needs) using mix mode, same principle as the AD9154.  These capabilities would allow fully digital modulation and direct output of signals with frequencies and bandwidths relevant for superconducting qubit experiments.  The phase noise performance is somewhat better than that of the AD9154 as well.  

In the fullness of time, once the existing Sayma AMC/RTM combination is up and running, it would be interesting to approach superconducting qubit researchers about interest in funding design and construction of a microwave-frequency RTM card based on these chips.  One could fit 4 single-channel DACs on a new RTM board (4 lanes each at 10 Gbps gives 1 GHz instantaneous bandwidth, use 6x interpolation, 6 GHz DAC clock, 12 GSPS output update) for use with the existing Sayma AMC.  Resource usage on the AMC FPGA should be similar (~double the data rate per channel but half the number of channels).  One could also add an ADC if desired, e.g. a 500 MSPS 4-channel ADC (e.g. AD9694) for qubit readout signals (this would still require an external demodulation circuit).  

This is not an action item for 2017, but it's probably worth starting a discussion with potential users once we have performance details in hand from the Sayma system.

gkasprow, 2017-05-19:
@dhslichter Will we keep existing front end modules or introduce new ones? We could keep the physical size and connectors but treat them as single channel.

dhslichter, 2017-05-23:
The analog daughtercards (Allaki) would need to be redone. We could keep the physical size, could definitely reduce the number of connectors (only one channel per board instead of two).  We could decide about whether it makes sense to be able to swap daughtercards with existing Sayma RTM, or whether we should intentionally make them different to prevent users from accidentally populating the wrong cards.

gkasprow, 2017-05-24:
@dhslichter I'm supervisor of very clever MSC student who already built complex AMC board for his bachelor thesis. It was more complicated than Sayma. And he is willing to build DAC RTM board as upgrade of Sayma RTM. He is very excited about ARTIQ project. He will defend his thesis in 2018 so we fit the timeline. At the moment he helps me with HDL for other boards. He could start this summer.

dhslichter, 2017-05-25:
Sounds like a good candidate then!  Will he require external funding, or is this something he could do as part of his thesis research with the funding already secured for him?  Would funds need to be available for prototyping?

hartytp, 2017-05-26:
@dhslichter Out of curiosity, is this kind of JESD204 DAC really suitable for SC QC? I got the impression that the latency of this kind of DAC was a bit too high for them.

dhslichter, 2017-05-26:
@hartytp it depends on your application.  If you are trying to do feedback, then yes, the latency is problematic.  However, most SC qubit experiments operate in a "preload data and trigger playback" mode, which would be fine here.  

For feedback applications, one would want to have parallel input DAC and parallel output ADC, both coupled directly to an FPGA on the RTM card which analyzes the ADC readings and outputs the desired waveforms to the DAC.  If you're doing this, you're stuck with an analog upconversion stage.  This is not itself a problem necessarily, just adds complication.  

For feedback applications one would want something like AD9736 as the output DAC and an AD9484 for the input ADC (8 bits is usually plenty for doing the qubit readout).  You'd want two for each channel (I and Q), and they'd be best synchronized.

gkasprow, 2017-05-29:
@dhslichter @hartytp 
For such applications one can use general purpose AMC FMC carrier with FMC DAC and ADC.
If Kintex device is sufficient, one can consider AFCK board (open source, available commercially) or newly designed AFCZ based on ZynQ US+ chip. It is based on Sayma design.

gkasprow, 2017-07-26:
@dhslichter @hartytp @jbqubit There are new [SOCs](https://www.xilinx.com/products/silicon-devices/soc/rfsoc.html) (Zynq UltraScale+™ RFSoC) from Xilinx that integrate 12bit 4GSPS ADCs and 16bit 6.4GSPS DACs in same silicon. In such configuration there is no issue with JESD204B latency.
Look at ZU25DR which has 8 ADC and 8 DAC channels. IF you need more channels, there is ZU29DR with 16 slightly slower ADCs and 16 DACs
Maybe this is the right direction for new Sayma hardware?

dhslichter, 2017-07-27:
These look interesting, but it's still vaporware right now.  It does use the same mix-mode technique to boost the signal bandwidth, which would be useful.  Unclear how much internal latency there will be (massive slow parallel bus inside?), if there is indeed lower latency than running JESD204B links this could be compelling.  There is the issue of digital switching noise and how much of that appears on the output signals.  Anyway, worth keeping an eye on.  It would definitely simplify the board layout!!

gkasprow, 2017-07-27:
They say they are shipping, which means it is still 2 year perspective to get these chips.

hartytp, 2017-08-20:
@dhslichter AFAICT, this issue is stale. Can we close it? (We can reopen it once we get someone from the SC community interested in this!)

dhslichter, 2017-08-21:
@hartytp we can close it, but I will flesh out the Wiki entry with some of the material that we have discussed in here and in person so that people coming to check out the site are aware of the possibilities and discussions.

closed 2017-08-21 by dhslichter

jordens, 2017-10-09:
Note to self: There is [a new family of 12.6 G/s DACs][http://www.analog.com/media/en/technical-documentation/data-sheets/AD9172.pdf#page=4) (and ADCs) with channelizers. That's originally designed for multi-band and multi-standard telco applications. But it's also very much the same idea as SAWG, just pushed into the DAC, i.e. relevant to the typical physics use case. Saves lots of FGPA resources and allows for more features on the channels.

dhslichter, 2017-10-09:
@jordens these are cool-looking chips, but I think not as useful for the Blaster use case because many (if not most) relevant frequencies will probably be in the second Nyquist zone, assuming 12.6 GSPS.  The output sample rate is limited to 6.16 GSPS with no interpolation AFAICT -- one can only run at 12.6 GSPS if you use interpolation, which kills off the (in this case desired) Nyquist images.  

That said, I think these could be very relevant for a next-generation Sayma RTM for applications below ~4 GHz, by enabling much of the NCO and spline interpolator heavy lifting to be carried out on the DACs, as you pointed out.

dtcallcock, 2017-10-09:
The impression I got from talking to the Delft crew (who are building something roughly equivalent to Sinara for SC qubits) a few weeks ago was that JESD busses were a non-starter for SC work due to the latencies.  It would also be nice to have Sinara hardware capable of fast feedback loops.

In related news, I noticed the FRSoC advance product spec came out a few weeks ago and they claim to be shipping samples so perhaps not vapourware for much longer: https://www.xilinx.com/support/documentation/data_sheets/ds889-zynq-usp-rfsoc-overview.pdf

@jordens It looks like these have some of the same NCO and mixer goodies as those new AD chips.  Only 14-bit though.

dhslichter, 2017-10-09:
I think @dtcallcock is right, to enable flexibility of use for feedback applications, it may make sense to stick with analog up/downconversion so we can use a parallel DAC and ADC on Blaster boards.  Those RFSoC specs are interesting, but again the frequencies are too low for direct synthesis.  However, if it saves you having to route an obscene number of LVDS buses around a PCB...

dhslichter, 2017-10-09:
14-bit is probably fine for SC qubit applications, most people are working with 12-bit or 10-bit still.

gkasprow, 2017-10-09:
I asked Xilinx about these RFSoCs, so far my FAE didn't now even what are the latencies. They will be published in a month or so.
Another problem with Xilinx chips is that when they are shipping chips it means they do it to selected customers who are beta tesers. I'm waiting already 2-nd year for EV series of ZynQ US+ SoC. They postponed delivery date already a few times.

dhslichter, 2018-08-16:
Update about the RFSoCs - I am seeing prices for the XCZU25DR (8 DACs, 8 ADCs) of ~$11k each in the slowest speed grade.  You better hope that the assembly house doesn't botch the solder job! :)  I think that this kind of pricing means we should look elsewhere for at least the next few years until it comes down....

jordens, 2018-08-17:
Isn't that a wrinkle the SC people will get over very quickly once they realize that this gets them very low latencies coupled with parametrizable AWGs and a CPU to supervise and manage everything in one package? Even more so once the prices go down.

gkasprow, 2018-08-17:
I've noticed that. We are building right now SDR radio based on RFSOC for satellite. RFSOC seem to be the only option for sub-us feedback systems i.e. for superconducting qubit readout. This is another project I work on.

gkasprow, 2018-08-17:
If you want to have a look, there are even schematics with this chip. The only difference are additional ADC/DAC banks that look like GTH transceivers :)

sbourdeauducq, 2018-08-17:
I wonder how many horrid bugs and abysmal design antifeatures those chips have (that is, unless they are swept under the carpet by the tens of gigabytes of crap toolchain and "IP" that will configure all the obscure undocumented parameters in one of the few ways that makes the hippo dance).
What about e2v devices? They have very low latency and a simple interface.

gkasprow, 2018-08-17:
My colleagues are using them, so far using devkit only. There were some issues but they managed to make it working. I asked them to measure real delay. Nice thing is hard IP digital NCO and up and downconverter.
Real nightmare would be ARTIQ support for such complex and poorly documented chip.

dhslichter, 2018-08-17:
Anyway, the point of my comment was that I don't think that RFSoC devices are what I want to be working with for *our* group's stuff.  If superconducting people want to pay more and deal with all the pain because they need the loop bandwidth, sure.  I haven't looked at the latency spec on the ADCs or DACs, though; I could imagine it being nontrivial if bad design choices were made.  And I agree with @sbourdeauducq about the worries with debugging -- now you can't even put a scope on your pins to see if the signals are arriving ;)

gkasprow, 2018-08-17:
On the other hand - sum up 8 x  GHz-grade ADCs and 8  DACs + large FPGA and you will surely end up with more than 11k $.