sinara-hw/sinara #185: 3U DAC schematics review

gkasprow, 2017-04-23:
@hartytp I sketched initial version of schematics and component placement for 3U DAC [(link)](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_DAC/3U_DAC.PDF)
Please check it. There are also noise simulation of the DAC and the output filter.
Simulations are in /SIM_CALC directory. You can run them with Tina TI.

dhslichter, 2017-04-24:
It appears you have selected the AD5373, which is a 14-bit DAC.  We'd want the AD5372, the 16-bit version.

dhslichter, 2017-04-24:
Switching supply IC15 should be TEN 5-1223, not TEN 5-2443 (12 VDC nominal input).

gkasprow, 2017-04-24:
Sure, I placed wrong symbol, fixed.
I didn't use TEN 5-2443 but TEN 5-2423WI It has extended voltage range from 9 to 36V so it is OK.

gkasprow, 2017-04-24:
I changed to TEN 5-1223 - it has 2% better efficiency and is 1$ cheaper.

jordens, 2017-04-25:
* R9, R10, R25 don't need to be precision 0.1% resistors. They don't affect DC stability.
* What are the four mounting holes for? I don't think you need them for the BNC adapters. They would be well held in the front panel.
* With the R40+R68 you need to account for another 5 mA per channel draw.
* In the power budget, how much current is left for the actual outputs?
* Is there no sensible way to use SIGGND[3..0] to get more immunity against ground loops and other reasons for ground potential differences?

gkasprow, 2017-04-25:
@jordens 
I know they don't have to be 0.1%.  I have such in my libs and on stock in assembly company so it doesn't make sense to keep stock of 1% and 0.1% of same values. I use either 0402 E24  1% or precise 0603  0.1%. Since I buy reels in quantities of 5k, the price difference is negligible here. It is certainly lower than having all possible E192 1% and 0.1% values on stock.
The mounting holes are not needed here, I left them just for compatibility with other boards. One can use spacers to make the construction rigid when used in custom box. The ADC board can be screwed to BNC ones.
I assumed +/- 200mA from power supply but it seems that the margin is too low. All the amplifiers  and ADC consume in worst case ~110mA
The divider has indeed too low value, I will increase it 10 fold to keep the current low.
Is 2.5mA of margin per channel not sufficient? What average current is required to drive high impedance load at the end of 5m cable? All depends what we define as high impedance.

jordens, 2017-04-26:
Ack the resistors. The mounting holes also don't hurt.
For the current budget, AFAICT 100 mA (or around that number) for all channels would be plenty. Worst case loads I can think of look like 1k+10nF (in series) or 10k.
There is also the 8 W Traco in the same package and of apparently similar specifications.
Also what about a [common mode choke and some more filtering](http://assets.tracopower.com/20170426080912/TEN8/documents/ten8-filter.pdf) on the input of the Traco?

dtcallcock, 2017-04-26:
It would be useful if this board was capable, via selection of different components, of outputting higher voltages.   Several of our traps need more like +/-40V.  

One way of making this possible in the future would be to:
- Provide a header so that higher voltage op-amp supply rails can be supplied from an external source.
- Use an op-amp that is pin compatible with high-voltage op-amps.  The highest voltage part I can find that is pin-compatible with the OPA-2197 is the ADA-4522, a 55V part.  If a pair of OPA-197s were used instead of the OPA-2197 there would give us some 100V options.

Maybe it will make more sense in the long run to just design a proper HV version of the board.  As these changes are fairly minimal, I thought I would make the suggestion.

dhslichter, 2017-04-26:
I think it makes sense to design a real HV version of the board to do this.  You would want to have separate HV regulation on the board itself (so noise from external supplies doesn't hurt you), plus different gain resistors.  My feeling is that for those looking to do a high-voltage "hack" in the near term, the best way to do it would be to make a little daughterboard featuring HV supplies and amps, which just plugs into the IDC headers with the voltage outputs and offers its own connectors to the outside of the crate, and leave the current Zotino design alone.

dhslichter, 2017-04-26:
I would be strongly in favor of a metal shield to cover the switching supply and its associated filters, if possible space-wise.

gkasprow, 2017-04-26:
@jordens  Common mode filtering makes sense only when galvanic isolation is needed. In such case DC/DC converter transformer injects noise between converter input and output and this component is removed by common mode choke in cooperation with bypass capacitors (C3,C4). Since in our case input and output ground is shorted, such is common mode noise and only LC filter on input and output are needed to get rid of ripples.
@dtcallcock Having two version of same board with high voltage output may sooner or later lead to some catastrophe when someone plugs 40V version instead of 10V one and connects something really expensive on another end of cable. As @dhslichter said, it may be better to either add an daughterboard with additional amps and supply or design another board with different connector.

gkasprow, 2017-04-26:
@dhslichter I'm against covering power supply and amplifiers by same shield because it causes opposite effect . DC/DC converter already has metal shield, so I will add shield to analog part of design only.

dhslichter, 2017-04-26:
I wasn't suggesting a metal shield for the whole board; I was suggesting a metal shield covering just the DC/DC converter and the first stage of output filtering (C26, C27, L10, C40, C42, L12).  I agree that a metal shield over the whole board (analog and power supply both) would have the opposite of the desired effect.  The metal shield on the DC/DC converter itself will be helpful, but I worry about capacitive coupling of spikes/noise on the outputs of the converter, before the filters, to analog voltages elsewhere in the circuit.

gkasprow, 2017-04-26:
You are right - the filtering inductors have open magnetic circuits so can pick the noise from neighbouring boards. To limit capacitive coupling I use cut-outs in ground planes to make sure that spikes do not propagate outside certain area. I will reserve place for such shield.

dhslichter, 2017-04-26:
Another question @hartytp @gkasprow mainly: what about using 2x LTC2668 instead of AD5372?  Rationale includes:

- 3x lower voltage noise
- faster settling, lower glitching and channel crosstalk
- lower tempco
- lower uncalibrated offsets
- lower drifts

Not necessarily pushing for this, but one might consider the option.

jordens, 2017-04-27:
Careful. AFAICT they measured several of those specs in the default 0-5 V output range, i.e. one might see 4 times more noise, crosstalk, offsets and glitching for the +-10 V range (assuming those originate before the range gain). And crosstalk, glitching are all worse already at those small ranges, no?

gkasprow, 2017-04-27:
I already received AD5372 devkit. With AD5372 we still meet the specification. Every project can be made better but at certain stage we have to freeze it and finish it. In case or troubles with AD5372 performance, we can consider another DAC in second prototype run.

dhslichter, 2017-04-27:
I am fine with AD5372, I just wanted to put this out there in case @hartytp or others had any comments.  I agree with @gkasprow that at some point we need to freeze and build, I just figured that we are still early enough with this one to ask these question.  @jordens these are good points; I think in the end it's not worth switching over, basically just wanted to ask.

gkasprow, 2017-05-01:
@hartytp I updated the board layout and finished routing. DAC is placed under the shield on a thermal island together with reference, heaters and thermal sensors.

hartytp, 2017-05-02:
@gkasprow Thank you for doing this so quickly Greg! That looks really nice. 

Thanks everyone else for the useful feedback.

Some minor comments and questions:

* Unless anyone objects, I'd prefer to increase the output impedance to 330R|4.7nF. The current 47R|33nF makes the output buffers work quite hard, and would result in a large current draw if multiple channels were slewing at max rate simultaneously. I assume the motivation for using such a big output capacitor is to avoid loading from cabling capacitance etc. Assuming a maximum 5m SCSI cable with 150pF/m (does this sound about right?) we'd expect the cable capacitance to be <~750pF. If we decrease the output capacitor to 4.7nF, then the decrease in bandwidth due to loading is only 15%, which I'd be happy with (presumably, the filter capacitors are only 5% tolerance to start of with).

* What limits the output slew rate in this design? Is it the 1V/us of the DAC? AFAICT, this design won't meet our original specification of being able to drive a full-scale sinusoid at 100kHz due to the DAC's slew rate, which limits the full-scale BW to ~20kHz. In this case, I'm okay with relaxing the specification, but I do want to make sure that the output stage doesn't further degrade the slew rate.

* Why two voltage references on the schematic? Which one will be DNP by default? (Can you annotate the schematic with these details, please?)

* IIRC in the original Kasli specification (see the Wiki) we agreed that the +12V rail would supply 100mA per EEM.  In this design, you assume 0.5A. Are we happy increasing this in the Kasli/VHDCI_carrier specification? Will these carriers be able to supply 8*0.5A=4A of 12V power to the EEMs (e.g. are the power connectors they use rated to this current level?)

* Can you annotate the maximum current allowed for the +-15V rails supplied via the HD68/IDC connectors, please? AFAICT, this is currently limited to +-50mA by the ferrite bead used (if there is more current available from the supply, then maybe we should use a larger ferrite bead to increase the current that can be supplied by these rails). 

* What are the diodes before the CMCs for? (can you annotate this, please?)

* What is the overall temp co of the design?

* Can you label the output range per channel, please? (+-12V, right?)

* The OPA2197IDGKR you've used for the output filter looks nice. However, it only has a GBP of 10MHz. Isn't that a bit low for an active filter of this kind? I don't really mind as your simulations show that this design will meet the specification, but I'm a bit curious.

* What are the LEDs for (again, please annotate this on the schematic)?

* Should the NC pin of IC30 be connected to ground for better heat sinking?

* Could you add some extra annotation around the temperature regulator, please? e.g. "Fixed set-point and gain temperature controller." Give the max heater current, temperature set point etc. What does "okay" mean (i.e. +-?mK)? Did we originally agree to make the set point for this adjustable with a pot (I'm not really fussed here, so long as we put the set point somewhere sensible)?

* When we discussed this previously, the idea was to aim to meet the drift specification in good "lab" environment (say, +-0.5K) using passive stability only. The temperature regulator is only to be used as a fallback plan. Could we have a jumper to allow it to be disabled?

* There is a rough power budget on the schematic. Please could you fill it out a bit more (make it complete and a bit more readable).

* Add something on the silk screen for the IDC headers (e.g. "IDC_0: DAC_0 to DAC_15")

* Thanks for including the simulations! Please could you: move them to a separate page; include a caption that explains what is simulated (just the output filter? or the whole system including the DAC?); put the noise on a log scale, as currently one can't see 1nV/rtHz on that scale; either remove the markers, or add the marker values as well.

* Generally tidying up the schematic (some of this applies to the ADC board as well), including: move the overview page to the beginning of the schematic; label important ICs like the configuration EEPROM and voltage supervisor; remove the outdated description + specification (just replace with something like "32-channel, 16-bit 1MSPS DAC"); annotate the output RC pole frequency;

hartytp, 2017-05-02:
@gkasprow Thanks for the updated DAC schematic. I was writing while you posted that. 

I'll have a look over it in the morning and get back to you with any comments.

gkasprow, 2017-05-02:
@hartytp did you look at right schematic? There is single reference source.
https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/Kasli/3U/PCB_3U_DAC/3U_DAC.PDF

gkasprow, 2017-05-02:
*         Unless anyone objects, I'd prefer to increase the output impedance to 330R|4.7nF. The current 47R|33nF makes the output buffers work quite hard, and would result in a large current draw if multiple channels were slewing at max rate simultaneously. I assume the motivation for using such a big output capacitor is to avoid loading from cabling capacitance etc. Assuming a maximum 5m SCSI cable with 150pF/m (does this sound about right?) we'd expect the cable capacitance to be <~750pF. If we decrease the output capacitor to 4.7nF, then the decrease in bandwidth due to loading is only 15%, which I'd be happy with (presumably, the filter capacitors are only 5% tolerance to start of with).

This will somehow increase the noise, I will check it. We already hardly meet the specification with noise density above 1MHz

I also wanted to keep the output impedance low.

 

*         What limits the output slew rate in this design? Is it the 1V/us of the DAC? AFAICT, this design won't meet our original specification of being able to drive a full-scale sinusoid at 100kHz due to the DAC's slew rate, which limits the full-scale BW to ~20kHz. In this case, I'm okay with relaxing the specification, but I do want to make sure that the output stage doesn't further degrade the slew rate.

At the moment the low pass filter stage SR is 24V/12us = 2V/us

*         Why two voltage references on the schematic? Which one will be DNP by default? (Can you annotate the schematic with these details, please?)

In recent version of schematics there is one ref source.

*         IIRC in the original Kasli specification (see the Wiki) we agreed that the +12V rail would supply 100mA per EEM. In this design, you assume 0.5A. Are we happy increasing this in the Kasli/VHDCI_carrier specification? Will these carriers be able to supply 8*0.5A=4A of 12V power to the EEMs (e.g. are the power connectors they use rated to this current level?)

The DAC module supply current depends on output loading. At the moment it is about 300mA. We have to change the specification.

Power connector current depends on connector brand. Currently installed connectors handle 5A

*         Can you annotate the maximum current allowed for the +-15V rails supplied via the HD68/IDC connectors, please? AFAICT, this is currently limited to +-50mA by the ferrite bead used (if there is more current available from the supply, then maybe we should use a larger ferrite bead to increase the current that can be supplied by these rails).

OK

*         What are the diodes before the CMCs for? (can you annotate this, please?)

Protection diodes with low capacity. 

OK

*         What is the overall temp co of the design?

The highest influence is from DAC and reference. Opamps work with much higher range so their offset drift is negligible here. Main temp co of the output stage are gain setting resistors. 1ppm and 2ppm values are hell expensive (15$/pc) and since we need 64 of them it would be the most expensive part of the project.

At the moment I installed 10ppm values and hope that drifts will compensate each other.

I’ll update schematic with estimations.

*         Can you label the output range per channel, please? (+-12V, right?)

OK

*         The OPA2197IDGKR you've used for the output filter looks nice. However, it only has a GBP of 10MHz. Isn't that a bit low for an active filter of this kind? I don't really mind as your simulations show that this design will meet the specification, but I'm a bit curious.

It is the fastest precision opamp I manager to find with offset <200uV. Its GPB is taken into account in filter design. You are right, Normally one would need at least 30..50MHz to make amplifier GPB influence negligible.  But as I wrote, one can compensate (to some extend) limited GPB with component selection.

*         What are the LEDs for (again, please annotate this on the schematic)?

 

OK

There are just user defined LEDs, you can i.e. display activity, operation mode, number of channels, etc

*         Should the NC pin of IC30 be connected to ground for better heat sinking?

99% of the heat is transferred by thermal pad. They are not connected anyway in the package. 

But I can connect to GND if you like

*         Could you add some extra annotation around the temperature regulator, please? e.g. "Fixed set-point and gain temperature controller." Give the max heater current, temperature set point etc. What does "okay" mean (i.e. +-?mK)? Did we originally agree to make the set point for this adjustable with a pot (I'm not really fussed here, so long as we put the set point somewhere sensible)?

We didn’t specify any parameters here. OK means that temperature is within +/- 5 degrees window. We can change it.

At the moment the set point is at 55degrees. 

*         When we discussed this previously, the idea was to aim to meet the drift specification in good "lab" environment (say, +-0.5K) using passive stability only. The temperature regulator is only to be used as a fallback plan. Could we have a jumper to allow it to be disabled?

OK

*         There is a rough power budget on the schematic. Please could you fill it out a bit more (make it complete and a bit more readable).

OK

*         Add something on the silk screen for the IDC headers (e.g. "IDC_0: DAC_0 to DAC_15")

OK

[... cut ...] y one can't see 1nV/rtHz on that scale; either remove the markers, or add the marker values as well.

OK

*         Generally tidying up the schematic (some of this applies to the ADC board as well), including: move the overview page to the beginning of the schematic; label important ICs like the configuration EEPROM and voltage supervisor; remove the outdated description + specification (just replace with something like "32-channel, 16-bit 1MSPS DAC"); annotate the output RC pole frequency;

OK

hartytp, 2017-05-02:
@gkasprow 

> This will somehow increase the noise, I will check it. We already hardly meet the specification with noise density above 1MHz
> I also wanted to keep the output impedance low.

Okay, just pick whichever output impedance you think is best for this design, given the factors we've discussed.

> The DAC module supply current depends on output loading. At the moment it is about 300mA. We have to change the specification.

> Power connector current depends on connector brand. Currently installed connectors handle 5A

Okay, unless @jordens objects, let's change the specification to 0.5A at 12V per EEM and design Kasli accordingly.

How hot do you think this board will get when dissipating close to 6W while mounted in a standard rack? Do you think heatsinnks/fans will be needed?

> The highest influence is from DAC and reference. Opamps work with much higher range so their offset drift is negligible here. Main temp co of the output stage are gain setting resistors. 1ppm and [...]

Thanks for clarifying. 10ppm (or even 25ppm) are fine IME.

> We didn’t specify any parameters here. OK means that temperature is within +/- 5 degrees window. We can change it.
> At the moment the set point is at 55degrees.

Okay, let's choose a set point once we see how hot this board runs in a standard rack. 

The "okay" LED should be something like +-0.1K (if the temp varies by much more than that then there isn't any point having a temperature controller).

Some other really minor points I noticed when looking over the new version of the schematic:

* Pag 2 title should be "IDC to BNC" not "IDC to SMA"

* Quite a few of the component names/values overlap with each other on the schematic, making them hard to read. 

* Could you label the "clips" on the schematic as "screening can" or something, so it's clear what they are.

gkasprow, 2017-05-02:
This will somehow increase the noise, I will check it. We already hardly meet the specification with noise density above 1MHz
I also wanted to keep the output impedance low.

Okay, just pick whichever output impedance you think is best for this design, given the factors we've discussed.

OK, I will play with values.

The DAC module supply current depends on output loading. At the moment it is about 300mA. We have to change the specification.

Power connector current depends on connector brand. Currently installed connectors handle 5A

Okay, unless @jordens <https://github.com/jordens>  objects, let's change the specification to 0.5A at 12V per EEM and design Kasli accordingly.

How hot do you think this board will get when dissipating close to 6W while mounted in a standard rack? Do you think heatsinnks/fans will be needed?

If we pack 8 such boards to single box, this could be an issue. Luckily there are enclosures with forced cooling so we can mitigate it.

Standard racks are open and are convection cooled.

I see no way to reduce current consumption unless we get rid of the output buffers

The highest influence is from DAC and reference. Opamps work with much higher range so their offset drift is negligible here. Main temp co of the output stage are gain setting resistors. 1ppm and 2ppm values are hell expensive (15$/pc) and since we need 64 of them it would be the most expensive part of the project.
At the moment I installed 10ppm values and hope that drifts will compensate each other.
I’ll update schematic with estimations.

Thanks for clarifying. 10ppm (or even 25ppm) are fine IME.

There are also matched resistor divider pairs and quads in single package but are expensive.

The opamps may heat and their temperature is not stabilized so may influence DC stability

If measurements in the rack won’t satisfy us we can fill entire shield area with heat conducting compound and apply temperature regulation loop.

Take into account that even if we dissipate 1W in a rack without convection the temperature rise can be a dozen of degrees so IMHO temperature regulation is necessary

But it is the last resort.

We didn’t specify any parameters here. OK means that temperature is within +/- 5 degrees window. We can change it.
At the moment the set point is at 55degrees.

Okay, let's choose a set point once we see how hot this board runs in a standard rack.

The "okay" LED should be something like +-0.1K (if the temp varies by much more than that then there isn't any point having a temperature controller).

For such precise regulation I have to use low tempco opamps in regulator loop.

Some other really minor points I noticed when looking over the new version of the schematic:

*         Pag 2 title should be "IDC to BNC" not "IDC to SMA"

*         Quite a few of the component names/values overlap with each other on the schematic, making them hard to read.

*         Could you label the "clips" on the schematic as "screening can" or something, so it's clear what they are.

OK

gkasprow, 2017-05-02:
@hartytp 

> This will somehow increase the noise, I will check it. We already hardly meet the specification with noise density above 1MHz
 > I also wanted to keep the output impedance low.
 > Okay, just pick whichever output impedance you think is best for this design, given the factors we've discussed.

OK, I will play with values.

 > The DAC module supply current depends on output loading. At the moment it is about 300mA. We have to change the specification.
 > Power connector current depends on connector brand. Currently installed connectors handle 5A
 > Okay, unless @jordens objects, let's change the specification to 0.5A at 12V per EEM and design Kasli accordingly.
 > How hot do you think this board will get when dissipating close to 6W while mounted in a standard rack? Do you think heatsinnks/fans will be needed?

If we pack 8 such boards to single box, this could be an issue. Luckily there are enclosures with forced cooling so we can mitigate it.
Standard racks are open and are convection cooled.
I see no way to reduce current consumption unless we get rid of the output buffers

 > The highest influence is from DAC and reference. Opamps work with much higher range so their offset drift is negligible here. Main temp co of the output stage are gain setting resistors. 1ppm and 2ppm values are hell expensive (15$/pc) and since we need 64 of them it would be the most expensive part of the project.
 > At the moment I installed 10ppm values and hope that drifts will compensate each other.
 > I’ll update schematic with estimations.
 > Thanks for clarifying. 10ppm (or even 25ppm) are fine IME.

There are also matched resistor divider pairs and quads in single package but are expensive.
The opamps may heat and their temperature is not stabilized so may influence DC stability
If measurements in the rack won’t satisfy us we can fill entire shield area with heat conducting compound and apply temperature regulation loop. But it is the last resort.
Take into account that even if we dissipate 1W in a rack without convection the temperature rise can be a dozen of degrees so IMHO temperature regulation is necessary

 > We didn’t specify any parameters here. OK means that temperature is within +/- 5 degrees window. We can change it.
 > At the moment the set point is at 55degrees.
 > Okay, let's choose a set point once we see how hot this board runs in a standard rack.
 > The "okay" LED should be something like +-0.1K (if the temp varies by much more than that then there isn't any point having a temperature controller).

For such precise regulation I have to use low tempco opamps in regulator loop.

 > Some other really minor points I noticed when looking over the new version of the schematic:
 > •	Pag 2 title should be "IDC to BNC" not "IDC to SMA"
 > •	Quite a few of the component names/values overlap with each other on the schematic, making them hard to reaad.
 > •	Could you label the "clips" on the schematic as "screening can" or something, so it's clear what they  > are.

OK

hartytp, 2017-05-02:
@gkasprow Re screening cans: what noise sources are we trying to screen the analog outputs from? Internal (SMPs etc) or external sources (Kasli etc)?

Also, what is the plan for mounting this board in a 19'' rack? Will it be mounted in a metal enclosure like [this](https://schroff.pentair.com/en/schroff/plug-in-units---frame-type---kits/3074457345616861429)? If so, then the PCB mount screening cans shouldn't be needed, right? Or, is the idea that the screening cans allow us to mount the board without an enclosure (which would be nice)?

In general, I'd like to avoid adding screening cans unless they're necessary, as the increase the BOM cost and reduce air circulation. But, this is something we can test properly during prototyping.

hartytp, 2017-05-02:
> If we pack 8 such boards to single box, this could be an issue. Luckily there are enclosures with forced cooling so we can mitigate it.
Standard racks are open and are convection cooled.
I see no way to reduce current consumption unless we get rid of the output buffers

We will need 200DAC channels for some of our experiments, so this will be an issue for us. It sounds like using an enclosure with forced air cooling will be required in that case.

Agreed, we can't reduce the power dissipation for this design easily. I just want to make sure that we have enough thermal management/heat sinking to make sure that this doesn't cause problems.

> For such precise regulation I have to use low tempco opamps in regulator loop.

Okay, I don't want to over specify the temperature regulator. But, if the board's temp co is around 10ppm/K and we're aiming for drifts <~+-10ppm then "OKAY" should mean at least <+-1K.

hartytp, 2017-05-02:
> @gkasprow All depends what we define as high impedance.

For testing, let's assume that the load is purely capacitive, with a capacitance of <=1nF (i.e. a long SCSI cable).

gkasprow, 2017-05-02:
@hartytp 
> We will need 200DAC channels for some of our experiments, so this will be an issue for us. It sounds like using an enclosure with forced air cooling will be required in that case.

> Agreed, we can't reduce the power dissipation for this design easily. I just want to make sure that we have enough thermal management/heat sinking to make sure that this doesn't cause problems.

For many channel options you can use open racks that enable air convection like [this](http://www.farnell.com/datasheets/95177.pdf?_ga=2.203610960.1557258132.1493722129-162313899.1464604133) or [this](http://www.farnell.com/datasheets/95187.pdf?_ga=2.203610960.1557258132.1493722129-162313899.1464604133)

gkasprow, 2017-05-02:
@hartytp Anyway, I will test these boards for their DC stability both in open rack and in closed rack so we will know exactly what is the performance in such conditions.

hartytp, 2017-05-02:
> @dhslichter I am fine with AD5372, I just wanted to put this out there in case @hartytp or others had any comments. I agree with @gkasprow that at some point we need to freeze and build, I just figured that we are still early enough with this one to ask these question. @jordens these are good points; I think in the end it's not worth switching over, basically just wanted to ask.

Short answer: yes, we did consider other DACs. 

Longer answer: we chose the AD5372 because it's the simplest, cheapest DAC that will meet our specification. It's also widely used in labs, making it a "known quantity", which is nice.

To be honest though, there really isn't much choice in terms of high-performance MSPS 16+ channel DACs (basically just this and the LTC DAC you mentioned). Obviously, one could go to a 4 or 8 channel DAC where there are lots of options, but then the cost, complexity (mucking around with CSs etc) and power consumption tend to increase. And, since the AD5372 already meets our specification, it's not clear what we'd be trying to improve anyway.

hartytp, 2017-05-02:
> @gkasprow It is the fastest precision opamp I manager to find with offset <200uV. Its GPB is taken into account in filter design. You are right, Normally one would need at least 30..50MHz to make amplifier GPB influence negligible.  But as I wrote, one can compensate (to some extend) limited GPB with component selection.

Out of curiosity, why do we need the OpAmp offset to be below 200uV? IIRC, the DAC's uncalibrated offset is +-10mV.

gkasprow, 2017-05-02:
@hartytp 
Hm, good question. I simply assumed that we want to have offset below 1 DAC bit. I noticed this uncalibrated and calibrated values and simply assumed that this particular DAC has autocalibration feature which is not true in this case.
If we can calibrate the output buffer together with DAC when selection criteria for output buffere are relaxed and we can use something with lower noise and much higher GPB.
We also have to select opamp that has offset temp coefficient below 10uV/K to make sure that even at 20deg of temperature rise we still meet the specification. But most opamps meet this requirement.
In such case we could use i.e. OPA1612 which has GBW of 40MHz but is 3x more expensive than OPA2197. It also consumes 3x more power which means that we will dissipate really a lot of power.
Interesting candidate could b OPA1602 with 35MHz GBW, but it consumes twice more power.
Try to play with [this](http://www.ti.com/lsds/ti/amplifiers/op-amps/op-amps-products.page#p480=2;2&p1261max=30;105&p23typ=30;18000&p22typ=2;18000&p2max=0.001;7.9&p1typ=0.00032;3&p1130=0.06;2.9318183&p743typ=0;10&p7typ=0;5.84) tool. It seems that if you want to have anything faster than 10MHz and cheaper than 3$ per piece you have to double or triple the power consumption.

hartytp, 2017-05-02:
@gkasprow 
> Hm, good question. I simply assumed that we want to have offset below 1 DAC bit. I noticed this uncalibrated and calibrated values and simply assumed that this particular DAC has autocalibration [...]

I assume that accurate calibration is not required in many cases, so we can generally leave the DAC and OpAmp offsets uncalibrated. Accurate calibration can be left to the users in the few cases that it's required.

However, if many users require accurate absolute accuracy -- which I don't think they do -- then we could consider using a different DAC (that's probably easier/cheaper than doing production-time calibration).

> We also have to select opamp that has offset temp coefficient below 10uV/K to make sure that even at 20deg of temperature rise we still meet the specification. But most opamps meet this requirement.
In such case we could use i.e. OPA1612 which has GBW of 40MHz but is 3x more expensive than OPA2197. It also consumes 3x more power which means that we will dissipate really a lot of power.
Interesting candidate could b OPA1602 with 35MHz GBW, but it consumes twice more power.
Try to play with this tool. It seems that if you want to have anything faster than 10MHz and cheaper than 3$ per piece you have to double or triple the power consumption.

Agreed, cost and power dissipation are important considerations for this design. Unless using a faster OpAmp would allow you to remove a filter section, it's probably not worth the increased power dissipation/cost.

My point is just that you don't have to use an OpAmp with such low offset, so there might be other options (e.g. cheaper/lower power dissipation). But, if you feel that the OpAmp you've selected is still the best choice then I'm happy to keep it.

dhslichter, 2017-05-03:
Echoing several points, and adding a few more:

- Agreed that we can relax the offset voltage spec on the op amps to get better performance if desired -- this will be calibrated anyway by the user if desired when doing the DAC calibration.  
- Why not move the cutoff frequency of the Butterworth filter lower, say to 50 kHz?  This will help with removing glitching on the DAC outputs anyway, make it easier to meet the output noise spec at 1 MHz, and we will be slew-rate limited by the DAC anyway for full-scale sinusoids above ~16 kHz.  
- Thermal considerations seem really important here.  Given that simple devices like even the 3U BNC are running hot in open air, I can imagine that these enclosures are going to need forced air.  If we add shielding cans, we will definitely have to check what they do to the thermal performance (and I think we want the shielding cans, especially if there is a Kasli in the same rack).
- Important to note that all capacitors in the signal chain must be NP0/C0G -- should note this on the schematic.  Other dielectrics exhibit DC bias dependence, which will distort the output signal.

hartytp, 2017-05-03:
+1 to @dhslichter's points.

Agreed: given the DAC's slew-rate limitation, 50kHz is a more sensible filter cut-off frequency. If this allows us to remove a filter stage (and hence reduce cost/power dissipation) while still meeting the noise spec at 1MHz then great. If not, then it's still worth it to improve the high-frequency noise.

gkasprow, 2017-05-03:
@dhslichter @hartytp All caps in signal chain are already NP0 - that's why they occupy so much space.
We can make the shielding cans perforate on sides to make cooling possible from both sides.
Perforations do not decrease shielding efficiency in low frequency range.
The think that worries me the most is this voltage divider that sets the gain to 1.2. We could mitigate it by either using +/-10V range, stabilisation of output stage temperature or using compensated resistor networks (i.e. from vishay [link]( http://www.mouser.com/ds/2/427/mpm-65556.pdf) which are 5$ per piece.

dtcallcock, 2017-05-03:
The Vishay [ACAS 0606 AT](http://www.vishay.com/docs/28770/acasat.pdf) divider is a lot cheaper, albeit with only 5-15 ppm/K tracking (depending on grade).

Otherwise is a +/-10V range that much of a problem if there is to be an amplifier board designed that provides higher voltage ranges (as seemed to be the consensus)?

hartytp, 2017-05-04:
@gkasprow +-12V would have been useful for some of our applications, because it gives us a little extra head room. But, if it's going to cause that much difficulty let's go to +-10V outputs instead.

hartytp, 2017-05-04:
@gkasprow For testing and measuring cross-talk etc, we'll need to connect the DAC board to a long SCSI cable. We'll then need an Rx (load-end) common-mode filter/CMC.

One way of doing this might be to make a simple passive 3U HD68 to 4xIDC breakout board, which we can use in conjunction with the IDC to BNC board you've already designed. This board would essentially look like the last part of the DAC board.

If we do this, it would make sense to add the CMCs to the IDC to BNC board. It would also be worth adding a single-pole differential mode RC filter to the IDC to BNC board. This would removing noise like high-frequency pickup in the cabling etc. This pole should be placed at a high enough frequency to avoid degrading the DAC bandwidth significantly (say, 200kHz).

What do you think?

gkasprow, 2017-05-04:
> @gkasprow For testing and measuring cross-talk etc, we'll need to connect
the DAC board to a long SCSI cable. We'll then need an Rx (load-end)
common-mode filter/CMC

Sure. I've frgotten about it. We discussed it allready
>
> One way of doing this might be to make a simple passive 3U HD68 to 4xIDC
breakout board, which we can use in conjunction with the IDC to BNC board
you've already designed. This board would essentially look like the last
part of the DAC board.

We can also make place for active buffers or fully diff amplifiers

>
> If we do this, it would make sense to add the CMCs to the IDC to BNC
board. It would also be worth adding a single-pole differential mode RC
filter to the IDC to BNC board. This would removing noise like
high-frequency pickup in the cabling etc. This pole should be placed at a
high enough frequency to avoid degrading the DAC bandwidth significantly
(say, 200kHz).

It makes sense.
Anyway it would make sense to provide setup for DAC->ADC characterisation
so everybody could run such setup to quickly validate performance of both
boards.

hartytp, 2017-05-04:
> We can also make place for active buffers or fully diff amplifiers

@gkasprow In the first revision, let's stick to using a cheap and simple passive filter only. Hopefully, this should be enough to do the job. If it isn't then we can add active circuitry to the next revision (the IDC to BNC is a very simple board, so the redesign is relatively quick). That way, we can avoid overcomplicating things.

> Anyway it would make sense to provide setup for DAC->ADC characterisation
so everybody could run such setup to quickly validate performance of both
boards.

Shall we add a compatible IDC to the ADC board as well? That way, we can go: DAC -> SCSI cable -> HD68 to IDC -> ribbon cable -> ADC for easy testing/calibration of 8 channels at a time. We just need a jumper somewhere so that we don't connect the DAC and ADC power supply lines together.

dhslichter, 2017-05-04:
A few more thoughts:

- I agree with @dtcallcock that using +/- 10V for outputs makes things a lot simpler.  One additional factor to consider would be that you could reduce the number of regulators, running both the DAC and the op-amps on +/- 11.5 V (made with LDOs from +12 V and -12 V); you would just need a switching regulator to make the -12 V rail.  This cuts your power dissipation by 20% right off the bat.  The DAC needs 1.4 V headroom on the rails, the op-amps are rail-to-rail.  In addition, you don't have to worry about the drifts from the resistors setting the gain of 1.2.  
- A second, bigger consideration: how about moving the filter cutoff frequency lower and using a two-pole active filter followed by a single-pole passive filter?  The stopband cutoff will not be quite as steep nominally, but for our purposes (especially given the slew rate limits of the DAC) it should be fine.  The advantage here is that we cut the number of op-amps in two, and thus reduce the power dissipation by almost 50% as well, in addition to making the board less crowded.  I've attached some screen shots of an alternative set of filter components and the simulation results.
- The final passive filter component choices will impact the distortion from the op-amp.  For static voltages we don't care, but if you're trying to output low-frequency sinusoids or other ramping signals, the THD is substantially degraded by a 2 kohm load impedance, relative to a 10 kohm load impedance (see datasheet figs 21 and 22).  I understand that the cabling has some capacitance and so this will impact the filter cutoff, but it's probably worth considering these tradeoffs.  

Below are some outputs of simulations with the different filter design I proposed, a Sallen-Key unity-gain Butterworth with 25 kHz nominal 3dB cutoff, followed by a single-pole passive RC.  
## filter components
Changed the topology, all 1% component values, increased output impedance.
[image]
## filter transfer function
3dB point is around 22 kHz now, attenuation above ~500 kHz is similar to that with existing filter design (a bit worse at 1 MHz, better at ~3 MHz), but in any event attenuations this large are almost never practically realizable because of parasitic couplings.
[image]
## output noise spectral density
White noise floor of the op-amp is 5 nV/rtHz, input referred, so basically the output noise is dominated by the op-amp noise out past 100 kHz; the op-amp noise contribution is not calculated by SPICE here.  
[image]
## integrated noise from DC
This all meets the spec nicely.  Counting the op-amp noise gives an additional 15 uV rms integrated up from 100 kHz to 10 MHz.  
[image]
## transient response
Here the green trace is a 10 kHz "square wave", as realized by the DAC output with its 1V/us slew rate.  The red is the waveform after the filter.
[image]

dhslichter, 2017-05-04:
Here is the TINA SPICE file, if @hartytp @gkasprow are interested (zipped for GitHub compliance).
[OPA197_singlefilter.zip](https://github.com/m-labs/sinara/files/977511/OPA197_singlefilter.zip)

hartytp, 2017-05-04:
I'll leave the final design decisions to @gkasprow, but I don't object to any of @dhslichter's points: 25kHz BW is fine given the DAC's slew rate (no lower though, please); and, reducing the power dissipation will definitely help us achieving the stability we need.

cjbe, 2017-05-04:
I would vote against reducing the bandwidth to 25 kHz (and would ideally prefer a something greater than 50 kHz if possible). The DAC slew rate is specified at 1V/us, and the full-scale change settling time as 20 us. Using the 25 kHz BW filter gives a 1% settling time of 70us.

hartytp, 2017-05-05:
* We all agree on +-10V outputs, so let's freeze that for this design round. @gkasprow do the regulators/supplies however you think best, ensuring the OpAmps have enough headroom for our worst case load (i.e. slewing at 1V/us with a 1nF load).

* @gkasprow can you tell me what the maximum bandwidth you can achieve while still meeting our noise spec, but using only a single OpAmp (2-pole active + 1 pole passive) is? Once we know that, we'll make a final decision about the bandwidth + filter topology.

klickverbot, 2017-05-05:
> worst case […] with a 1nF load

I'm sure you would have thought about this, but it seems like you might end up with higher capacitive loads than that when e.g. using the DACs to drive ion trap electrodes (with a passive lowpass on the other side as well, or even just the typical on-trap-chip capacitors). Of course the resistors add to the total impedance, but having been bitten by this in the past (with an overly noise-averse choice of additional trap filter components leading to considerable load on the output drivers), I thought I would mention it.

dhslichter, 2017-05-05:
My proposal incorporates two features which are separable: the reduction in the number of filter poles (and thus op-amps, and thus power dissipation), and the reduction in filter bandwidth.  I'm fine with bumping the filter bandwidth back up if desired, recognizing that it will probably mean somewhat less noise attenuation at 1 MHz.  The key point is that one can still use a single two-pole op-amp filter, rather than two of them (see example below), thus halving the power dissipation.  I think controlling the temperature will be key for stability, so this is a big help if you have a rack full of these things.  Likewise, changing to +/- 10V output range lets one use a simpler regulator scheme and dissipates an additional 20% less power by not boosting to +/- 15V (actually it's even better than this because of the dissipation from the switching supply inefficiency, which is no longer an issue for the positive rail).  

Incidentally, I was wrong previously about the op amp noise -- I forgot that we have the passive RC filter, which will also filter out the intrinsic op-amp output noise, thus the TINA SPICE is in fact taking that into account.  

Changing the components slightly to boost the 3dB frequency to 77 kHz, we get the following performance.  Current draw for a full-scale sine at 20 kHz from the DAC (this is already sort of beyond its slew rate capability, but is about the most power-hungry signal one could realistically implement) is only about 750 uV peak to peak, so these components are not going to load the DAC outputs down.  Peak current draw for the op amps in this configuration is about 4 mA per rail peak, about 3 mA per rail time-averaged, so the average power dissipation would be 72 mW per channel from the op-amps, or about 2.3 W for the entire card's worth of op-amps if they all happened to be running this signal.  If you assume you are driving an additional 1 nF cable capacitance, with another RC lowpass 10k and 10 nF at the end (e.g. inside a trap), then the power requirement goes up to ~100 mW per channel, ~3 W per card.  Again, this is a very-worst-case kind of calculation, for normal operating conditions there will be much less current draw.

Budgeting for regulators etc, this means you probably want ~250 mA per rail available, so ~500 mA at 12V supplied by the crate.  When producing static voltages only, though, the card will only use ~50 mA per rail, so total power dissipation would be ~1.2 W.  

## filter design
[image]
## ac response
[image]
## settling time to 1%
With this filter, a full-range step (+10V to -10V) gives a 99%-1% transition time for the filter output of 30 us.  
[image]
## noise spectral density
Total noise in 1 Hz-20 MHz is 57 uV rms, well in spec.  We still meet the 1 nV/rtHz at 1 MHz, more or less exactly.  This noise level is set by the passive RC, which has a higher cutoff than the active filter.  
[image]

dhslichter, 2017-05-05:
@klickverbot you commented while I was still writing my post, but thanks, and yes, you want to keep the output capacitances small when you can to avoid overloading the drivers for ac signals.

@hartytp @gkasprow seems like we are on the same page.  Attaching TINA for the filter I just showed above, @gkasprow you can play with it but I think this is pretty well optimized already.
[OPA197_singlefilter.zip](https://github.com/m-labs/sinara/files/977884/OPA197_singlefilter.zip)

hartytp, 2017-05-05:
@klickverbot The DAC board will work with unlimited capacitance in the sense that it won't oscillate or anything nasty like that (the output resistor of the RC filter should be large enough to stabilise the OpAmp). The only effect of the extra load capacitance is a reduction in the signal bandwidth.

I assume that the only reason that a user would attach a huge capacitor to their DC electrodes is to try to reduce the bandwidth of DC signals (and hence noise) as far as possible; people who want to do shuttling generally don't (shouldn't) attach huge capacitors to their trap electrodes.

I don't want to decrease the DAC output impedance (increase the output capacitance) from where is is, as this increases the dynamic current draw (power dissipation). We need this DAC to have really low drift, which means it's temperature must be fairly stable. Aiming for a high slew rate on multiple channels simultaneously into large capacitors makes that harder to achieve.

hartytp, 2017-05-05:
@dhslichter ACK. I'll leave this to @gkasprow from here, but it seems like we've reached agreement on all the important points and can finish the design now.

dhslichter, 2017-05-05:
@hartytp agreed, specific component choices can be optimized but the single op-amp filter and +/- 10V output range address are what we need to address various concerns.

klickverbot, 2017-05-06:
@hartytp: Of course it wouldn't be unstable, but the reduction in signal bandwidth will happen in a rather nasty, non-linear-response (and potentially quite drifty) way if you overload the output stage. As for "huge capacitors", thermal noise considerations (in room temperature traps) and the capacitors typically designed into traps (e.g. 1 nF for the HOA2) tend to lead to a fairly low input impedance for a given filter topology. It's not like people doing shuttling would care less about drifts and motional coherence…

In either case, I agree the current design should be a good basis for the final tweaks @gkasprow  might want to do. (A design that completely falls over with 1.5 nF of load based on your 1 nF specification would have been less so.) For "serious" in-sequence shuttling, we are going to want a higher bandwidth/data rate design anyway.

jordens, 2017-05-07:
Those capacitors are behind a couple k of resistors when viewed from the DAC. You'd naturally increase the series resistor to make whatever you'd want to do with your ions possible given DAC output limitations while at the same time maximizing the filtering effect.

hartytp, 2017-05-10:
@gkasprow Do you have all the information you need to finalise this design? When do you estimate we'll be able to move onto prototype production?

gkasprow, 2017-05-10:
@hartytp @dhslichter @klickverbot 
Thanks for optimising the filter design.
Indeed, reduced version of the design is good compromise and I will implement it. We would save also money on components.
+/- 10V specification will make thermal stabilisation of the output stage unnecessary since there are no components that cause DC drift. The only drift comes from amplifiers offset but it is negligible.
But I'm afraid of using 12V (and generated -12V) to supply DAC and output amplifier. This would work providing that we have high quality power supply and only DAC board is plugged to the 3U chassis. Dedicated DC/DC converter does good job stabilising both rails since it works from 9V to 18V and can handle both noise, ripples and unstable 12V rail. So I'd keep this supply with output LDOs. I would only use 12V version of DC/DC converter and LDOs that work with very low dropout, i.e. 11.8V to filter out converter ripples.

hartytp, 2017-05-10:
@gkasprow Thanks for the update.

That all sounds like a good plan to me!

dhslichter, 2017-05-10:
@gkasprow @hartytp agreed, this sounds good generally. @gkasprow can we add a little more margin into the dropout of the LDO?  It seems to me that one tends to get much better performance (and we do care about very clean voltage rails here) with ~500 mV of dropout, and one does have to consider the tolerances and ripple on the power supply, which could easily mean the output of the dc/dc converter is 11.8 V and not 12V on the ten5-1222, for example.  .  Is it possible to get a switching dc/dc converter that will give +/- 13V instead of +/- 12V or +/- 15V?  For example [CC6-1212DF-E](https://www.digikey.com/product-detail/en/tdk-lambda-americas-inc/CC10-1212DF-E/445-2436-ND/920396) has a "trim" pin to allow setting the output between 12 V and 15 V using a resistor.  This would let us build in a little more margin for the LDOs while still reducing the power dissipation because we're not going all the way to +15 V.

[25 more replies]

closed 2017-05-15 by hartytp