sinara-hw/sinara #187: 3U BNC and 3U RJ45 tests

gkasprow, 2017-04-26:
Today I did tests of 3U BNC.
It works correctly but:
- there is an error in OE connection of the driver, it is caused by wrong schematic symbol. It should be grounded.
- I connected 2x 3U BNC baords in the loop, so the signal passes via all transceivers with all terminations on. It's getting warm, mainly due to termination current. The main heating component is DC/DC converter which heats up to 55degrees. I will pack them all with panels to the minirack and see how it works
- I also connected 3U RJ45 LVDS board in the loop with 3U BNC and noticed that RJ45 connectors are swaped. Numbering on the panel is 1-3-2-4 instead of 1-2-3-4. I don't know how relevant it is for users.
 Below are some photos

[image]

[image]

[image]

jbqubit, 2017-04-27:
@gkasprow Thanks for the test update. 

When 2 of the 3U BNC are adjacent in a rack is there room for fingers to attach/detach BNC cables?

Regarding panel numbering of 3U RJ45 let's cover incorrect numbering on panel with sticker showing 1-2-3-4 and fix silkscreen for future printing. 

What's the highest frequency pulsetrain the drivers can generate into 50 Ohm before significant degradation in pulse shape?

gkasprow, 2017-04-27:
@jbqubit I will check it but it is doable unless you have very thick fingers :) 
I tested it with 10MHz, but will connect better generator and see what is max frequency. It should work up to 100MHz easily but needs to be confirmed.

gkasprow, 2017-04-27:
@jbqubit, below is photo of 3U modules placed in Schroff mini rack (100$ per complete set)
[image]
Today I also designed and fabricated plastic front panels for all 3U modules to see how all fits. After verification I will order professional Al ones with stencil-printed graphics. 
[image]

hartytp, 2017-05-01:
> @gkasprow Numbering on the panel is 1-3-2-4 instead of 1-2-3-4. I don't know how relevant it is for users.

Do you mean the ordering of the RJ45 connectors themselves? i.e. does this mean that the two RJ45 connectors that are driven from the same IDC aren't adjacent on the front panel? That would be a bit confusing for users, so could you fix this for the next revision, please?

gkasprow, 2017-05-01:
They are not adjacent. OK, i will fix it.

jbqubit, 2017-05-09:
@gkasprow Any confirmation of what the pulse shape looks like coming out of these? I've had several prospective users inquire about the minimum pulse width for the BNC/SMA/RJ45 hardware.

gkasprow, 2017-05-11:
@jbqubit Today I made a setup and did pulse measurements.
For this purpose I connected DDS generator and programmed series of 4V pulses.
DDS works at 600MHz sampling rate. I used 7 series of pulses with widths starting from 1 to 7 clock periods (1.6ns)
The chain was 3U BNC -> IDC -> 3U BNC, terminated on input.
1.6ns pulse did not propagate but longer pulses did it easily. The shortest pulse did not achieve 2.1V threshold required by TTL so probably it was a reason why was not propagated.
Below are some screenshots. Yellow, upper plot is output, green, lower plot is input.
It seems that digital isolator bandwidth (~150MHz BW) is main limiting factor. I did not use any filters on DDS output nor on the scope input. Scope BW is 1GHz, sampling rate is 5Gs/s. Scope is 50Ohm terminated.

[image]
[image]
[image]
[image]
[image]

And the setup
[image]

hartytp, 2017-05-14:
Anything else to do here, or can we close this issue?

closed 2017-05-19 by gkasprow

jbqubit, 2017-05-22:
This looks good. Looks like latency is about 20 ns per device. Will be interesting to see addition of VHDCI Carrier and VHDCI cable when KC705 test setup is ready.

gkasprow, 2017-05-22:
THe FMC-VHDCI boards are already assembled. Hope to get them tomorrow.

gkasprow, 2017-05-26:
@jbqubit VHDCI 3U board works well. With 1m VHDCI cable we get 7s more delay
[image]

gkasprow, 2017-05-29:
And the setup...
[image]

jbqubit, 2017-05-30:
Is this driven using KC705?

gkasprow, 2017-05-30:
not yet but will be very soon :)

jordens, 2017-05-30:
@gkasprow If I have Kasli or Metlino driving SPI over an IDC cable to a EEM with an LVDS transciever module, the clock speed for SPI reads is limited by the round trip time (the SPI shift CLK edge from the FPGA, over the cable, through the transceiver, to the SPI slave device, and then the MISO data back out of the SPI slave device, through the transceiver, over the cable, into the FPGA, meeting the SPI CLK sample edge setup time half an SPI CLK cycle after the SPI CLK shift edge). What's a good number for that round trip? 50 ns?

gkasprow, 2017-05-30:
I achieved 50ns with 1m VHDCI cable (each meter is 7..8ns), two IDC cables
20cm long, dual isolators and LVDS buffers.
In case of our SPI over LVDS let's assume 20ns + 7ns*length of cable.
Actually, the SPI core can wait certain number of clock cycles for data and
achieve fast transfer on long links as well

jordens, 2017-05-30:
Tuning the sampling delay beyond the clock edges makes the gateware much more complicated. It breaks the shift register (shared for output and input) and falls apart if there are multiple devices with different speeds (different RTTs) on the bus. And it requires tuning for each link length. Let's not do this. We already have **multiple** plans for fast data rates: The fast Novogorny ADC emits a source-synchronous CLKOUT. And for higher sample rates people should really just use Sayma or start a project where the ADC-to-DAC loop is done locally on the EEM.