sinara-hw/sinara #188: EEM chassis standard

jbqubit, 2017-05-09:
The following is clearly defined in our standard for Sinara Eurocard Extension Modules [EEM](https://github.com/m-labs/sinara/wiki/EEM).
* 3U Eurocard format 
* PCB height is 100 mm
* PCB depth is >= 160 mm
* card pitch in chassis 4UP or 12 HP

Several example chassis are given. But I've found standard-by-example to be problematic when looking for an air-cooled, rack-mount chassis compatible with the EEMs. It would help to identify which standard we are compliant. To my eye it looks like the standard is "IEC 60297-3-102". @gkasprow, please confirm

Also, what is the standard PCB thickness for EEMs?

jbqubit, 2017-05-09:
Looking further, calling out such a standard may be insufficient. For example, there's a baffling variety of configuration options. See for example, p. 6.41 in the Schroff catalog for guide rails. 

https://schroff.pentair.com/wcsstore/ExtendedSitesCatalogAssetStore/Attachment/SchroffAttachments/06_e_subracks.pdf

Here are the high-level parts I've got for the 19" rack-mount, air-cooled enclosure I'd like to use for EEMs.

RatiopacPRO AIR, Case, Unshielded, Rack Mount with Handles
Part No.: 24572-670, qty 1

Radial fan unit for ratiopacPRO air
Part No.: 24572-407, qty 1

@gkasprow Do you know what other Schroff parts should be ordered to create complete enclosure?

gkasprow, 2017-05-09:
@jbqubit, The easiest way is to buy complete case with all accessories.
i.e. this one
https://schroff.pentair.com/en/schroff/14575-154
I just received it yesterday. If you like I can take a few photos.

gkasprow, 2017-05-09:
Standard thickness is 1.6mm

hartytp, 2017-05-13:
Anything else need doing here, or can we close this issue?

jbqubit, 2017-05-22:
To me this is still an open question. I've placed an order for the following parts from Schroff which I believe constitute a complete fan-cooled, 19" rack-mountable crate compatible with the Sinara 3U extensions. I'd like to leave the Issue open until this is a confirmed solution. 

24572-670, RatiopacPRO AIR, Case, Unshielded, Rack Mount with Handle, qty 1
24572-407, Radial fan unit for ratiopacPRO air, qty 1
24560-351, Guide rails, standard, 160 mm deep, qty 2
24560-141, Panhead screw, Torx (2.5x9.3), self-tapping, steel zinc plated, 1
24560-255, ESD clips  for guide rails, qty 1
34561-384, threaded strip, qty 4
24560-884, insulation strip, qty 1

hartytp, 2017-07-16:
@jbqubit Can we close this now?

jbqubit, 2017-07-17:
No not yet. Not all the parts have arrived! It turns out to be extremely annoying to source Eurocard crates in the USA...

hartytp, 2017-08-10:
@jbqubit How did this go?

gkasprow, 2017-08-10:
we should definitely provide option of ordering such crate together with EEM modules.

jbqubit, 2017-08-11:
The last shipment arrived about a week ago. I haven't yet put the parts together to know if something is missing. Will do after NACTI. 

@gkasprow Agreed that a standard crate ordering option would be excellent.

hartytp, 2017-09-03:
@jbqubit Have you tested your crate yet?

hartytp, 2017-09-03:
Does anyone know of a 4HP wide stand alone case for small EEMs like the clock buffer? Something like a narrow version of one of these: https://schroff.pentair.com/en/schroff/plug-in-units---frame-type---kits

jordens, 2017-09-03:
IIRC greg had a bunch of examples from other manufacturers somewhere.

hartytp, 2017-09-03:
Yes, but I'm ideally looking for something that is compatible with the panels that greg is making. The ones I saw were either too wide, or needed a different panel.

gkasprow, 2017-09-03:
@hartytp [This](http://uk.farnell.com/vero/114-52008l/case-eurocard-3u-6hp/dp/1277439) should do the job.

gkasprow, 2017-09-03:
There are even 4HP versions...

hartytp, 2017-09-03:
Thanks @gkasprow. Do you know if those will work with the panels you've designed without modification?

gkasprow, 2017-09-03:
@hartytp it looks like they accept standard panels and custom panels so should work.
Anyway, I will buy one such case and check.

hartytp, 2017-09-03:
Thanks! Let me know how it goes.

gkasprow, 2017-09-03:
@hartytp look [here](http://www.parkfield.co.uk/verotec_us/lbx-tac-print.jpg)
It will work in our case.

gkasprow, 2017-09-03:
@hartytp How many of such enclosures would community need? We can order some volume of them from manufacturer for most common widths - i.e. 4 HP and 8 HP and make them available together with EEMs.

hartytp, 2017-09-03:
> @hartytp look here
> It will work in our case.

Nice. I'll buy one and see how it works.

> @hartytp How many of such enclosures would community need? We can order some volume of them from manufacturer for most common widths - i.e. 4 HP and 8 HP and make them available together with EEMs.

Not sure what everyone else's plans are, but I only need a few of these (e.g. mounting standalone clock buffers etc). The vast majority of my EEMs will be mounted in 19'' chassis. I'll also have a few EEMs mounted with 1-2 EEMs + Kasli in a mini rack (part number TBD).

jordens, 2017-09-03:
I'd expect a reasonable demand for small 8, 20 or 40HP cases (Kasli + a few EEMs). Simply since most of the time Kasli will fill at most half of the HPs of a a 19" crate (84).

hartytp, 2017-09-03:
> I'd expect a reasonable demand for small 8, 20 or 40HP cases (Kasli + a few EEMs).

Yes, I'd certainly want a few of those for my lab.

gkasprow, 2017-09-05:
@hartytp You can even fit two EEMs in such case!
[image]

hartytp, 2017-09-05:
Looks good.

gkasprow, 2017-09-05:
Note that there are not ventilation holes, so you won't be able to pack too many DDS boards :)

hartytp, 2017-09-05:
ping @jbqubit can we close this?

jbqubit, 2017-09-05:
@arpitagrawal23 <https://github.com/arpitagrawal23> at UMD put the parts
together last week. Sounded like some parts were missing. @arpitagrawal23
<https://github.com/arpitagrawal23> please update. Consider posting a photo
showing air-flow path.

dhslichter, 2017-09-06:
How are we going to get the IDC cables in to these cases?

hartytp, 2017-09-06:
> How are we going to get the IDC cables in to these cases?

This would be more for Kasli + 1 EEM in the same box. Or, in my case, mounting a clock buffer EEM in a nice way.

arpitagrawal23, 2017-09-06:
I have partially assembled it, here are the pictures:

[image]

[image]

Air goes in from holes in the front bottom part and exits from holes in the back top part. Airflow is shown on this webpage:
https://schroff.pentair.com/en/schroff/24572-670
(please see picture titled "Airflow" in the section "Other views")

But @jbqubit and I were thinking that cards which are in between of those fans might not receive sufficient airflow. Though right now, I've assembled fans as per manufacturer's instruction manual (2nd picture in "Other views" section on this web page: https://schroff.pentair.com/en/schroff/24572-407), if we push fans little bit backwards, airflow might be better.

Another thing is, the rack is designed for cards of size 220 mm, but our cards are smaller it seems (160 mm I guess). So the 160 mm guide rails that we ordered has no support at the back. We might need to put something there, I will discuss with @jbqubit.

jordens, 2017-09-07:
One pair of the horizontal profiles is just for that. The cards are not supposed to go all the way to the back.
Reading the catalogue will be helpful.

dhslichter, 2017-09-07:
What is the cost of one of these chassis with all the bits?

hartytp, 2017-09-18:
@jbqubit: ping. Can we close this now?

jbqubit, 2017-09-20:
Cost of chassis plus fan is about $650. We're doing some rejigging to overcome a manufacturer design error that leaves some slots with no air flow. @arpitagrawal23 is still figuring out full bill of materials.  Please don't close yet.

michallgaska, 2017-10-13:
@jbqubit where did you buy all rack stuff?

jbqubit, 2017-10-17:
I'm still trying to find a good solution for fan cooling. Currently I have two varieties neither of which is satisfactory.

# Elma
Elma fan unit  [27-250-02](https://www.elma.com/products/enclosures-and-components/19-inch-rackmount-enclosures/product-pages/19-inch-fan-trays-detail/).
- good: draws air across all card slots
- bad: has a disposable dust filter that costs $56
- bad: roars like a jet engine
- bad: $455 crazy expensive
- bad: they sold me a $38 temperature controller that only works on DC units (mine is AC)
- bad: requires TBD exhaust manifold for routing air to front/back side of crate
- bad: parts shipped over course of 3 months in 6 separate packages
- bad: total crate cost > $1300

# Schroff
A better-looking design is by Schroff for their ratiopacPRO line of Euro crates [24572-407](https://schroff.pentair.com/en/schroff/24572-407). @arpitagrawal23 reported on this earlier in Sept. 
-  terrible: draws air across only 30% of slots. It has no plenium. I'm working with the Schroff engineers to figure out how to make their [fan unit](https://schroff.pentair.com/en/schroff/24572-407) work correctly. Outside of this problem I like their solution and service. 
- good: quite
- good: filters
- good: draws air from front to back side of chassis
- good: low profile
- good: low cost $650 inclusive (rack + fan + card slots)
 
Presently I don't have a positive recommendation for a 19" rack for EEM modules.

gkasprow, 2017-10-17:
@jbqubit what about using general purpose 19" fan trays? You can buy them in a corner shop.
Examples are [here](https://www.amazon.com/BUD-Industries-FT-1973-Assembly-Height/dp/B005T593WC/ref=sr_1_3?ie=UTF8&qid=1508268323&sr=8-3&keywords=fan+tray+19%22) or [here](https://www.amazon.com/kenable-Tray-Mount-Cabinet-Cooling/dp/B0055P2MV0/ref=sr_1_48?ie=UTF8&qid=1508268412&sr=8-48&keywords=fan+tray+19%22)

gkasprow, 2017-10-17:
[Here](http://www.rackmountmart.com/html/fS.htm) are ones with front intake

jbqubit, 2017-10-18:
Thanks @gkasprow. I'll give these a try.... 
1 of https://www.amazon.com/gp/product/B016A4CP3E/ref=ox_sc_act_title_4?smid=ATVPDKIKX0DER&psc=1
2 of https://www.amazon.com/gp/product/B06XCHJY7J/ref=ox_sc_act_title_3?smid=A3K929QT0IENFR&psc=1

The fan will be 1U offset from the top of the [RatiopacPRO AIR](https://schroff.pentair.com/en/schroff/Ratiopac-PRO/RatiopacPRO-AIR-Complete-Case-Unshielded-Desktop-ASIM0114) chassis you see in the above photos. Some duct tape will fix that in my setup. I'll report back if this works out.

jordens, 2017-10-30:
[These](https://www.bopla.de/fileadmin/products/Katalog_Pdfs/Ultramas_GB.pdf) should also work as small desktop cases.

hartytp, 2017-12-01:
ping @jbqubit can we close this yet?

jbqubit, 2017-12-04:
TRIPP LITE SRFAN1U Rack Enclosure Server Cabinet 1U Cooling Fan Tray 3 120V is available from Amazon $136.99 | [B016A4CP3E](https://www.amazon.com/Enclosure-Cabinet-Cooling-High-Performance-SRFAN1U/dp/B016A4CP3E). I've tested one and find that it works fine. Air flow to all slots at a single high rate of flow. It's a bit noisy but not objectionable. 

[Vector Electronics](https://www.vectorelect.com/subrack-emc-rugged.html) CCA72-84-01 is advertised to be a suitable, pre-assembled rack with card trays. It's available from [DigiKey](https://www.digikey.com/product-detail/en/vector-electronics/CCA72-84-01/CCA72-84-01-ND/6819333), Mouser and Gerber. At $225 this is far cheaper than Elma and Schroff. And its made in the US -- Los Angles. I've got one on order.

jbqubit, 2018-01-17:
For many months now I've searched for a Eurocard rack solution that meet the following criteria. 
- compatible with [EEM v1](https://github.com/m-labs/sinara/wiki/EEM#eem-mechanical) Sinara peripherals
- low cost
- excellent availability in USA (lead time < 1 week)
- forced air cooling of all EEM card slots

I evaluated offerings from several suppliers and found they [did not meet](https://github.com/m-labs/sinara/issues/188#issuecomment-337291750) these criteria. Here's what does meet the criteria. 

| Descriptor | Manufacturer | Cost | P/N | Qty |Extended |
|------------|--------------|------|-----|-----|------|
| fan tray| Tripp Lite | [$130/ea](https://www.amazon.com/Enclosure-Cabinet-Cooling-High-Performance-SRFAN1U/dp/B016A4CP3E) | SRFAN1U | 1 | Rack Enclosure Server Cabinet 1U Cooling Fan Tray 3 120V |
| rack | Vector Electronics [0] | [$225/ea](http://www.vectorelect.com) | CCA72-84-01 | 1 | pre-assembled 3U rack |
| card guides | Vector Electronics [0] | [$19/ea](http://www.gerberelec.com/) | CG3-160/12 | 4 | 160 mm card guide pkg of 12 |
| 1U blanks | Rosewill | [$15/ea](https://www.amazon.com/Rosewill-Horizontal-Vented-Cabinet-RSA-1UPA003/dp/B06XCHJY7J) | RSA-1UPA003 | 2 | 1U 19 Inch Horizontal Metal Blank Panel with Vented slots | 

[0] may be ordered direct from Vector
- 11115 Vanowen Street; North Hollywood, CA  91605
- www.vectorelect.com
- 800-423-5659 

[image]

[image]

NOTE: This information is provided for reference only and is not an endorsement of any particular vendor's product.

closed 2018-01-17 by jbqubit

hartytp, 2018-01-17:
Nice! Good work Joe.