sinara-hw/sinara #189: SMA EEM review

hartytp, 2017-05-14:
LGTM. 

The only minor observation I have is that the schematic is a bit dense and sparsely annotated, making it somewhat difficult to read. Maybe other people have different preferences, but I'm less bothered about fitting the schematic on a single page than I am about making it easy to follow.

A few things that could be annotated are:

- Grounds are isolated in banks of 4 IOs
- IOs are 5V CMOS logic levels
- IOs can be individually 50R terminated (could comment which switches do the terminating etc).
- Outputs can drive 25R

gkasprow, 2017-05-14:
@hartytp 
That's true, the schematic looks dense. It is caused by that we added more and more functionality and splitting into sheets takes some time. I fixed it a little bit.

hartytp, 2017-05-14:
>  It is caused by that we added more and more functionality 

Agreed, these somehow slipped from simple TTL buffer boards to quite complex designs. 

> I fixed it a little bit.

Thanks! Anyway, tidying up the schematics is something that can be done later during the documentation stage after the designs have been properly tested.

hartytp, 2017-05-15:
Closing as this seems resolved.

Feel free to reopen if you have other comments.

closed 2017-05-15 by hartytp