sinara-hw/sinara #190: EEM status

hartytp, 2017-05-14:
@gkasprow I'd like to buy some of the following boards (inc front panels):

- BNC EEM
- RJ45 EEM
- VHDCI carrier

AFAICT, these boards have now undergone preliminary (if not quite complete) testing and seem to work as expected. Is that correct?

Have all issues been resolved (inc #154), and are they in a state where I can order them from TechnoSystem? If so, I'll email the ARTIQ list to see if anyone else is interested in ordering at the same time to reduce costs.

gkasprow, 2017-05-14:
@hartytp The boards: BNC EEM and RJ45 EEM were quite well tested. I've just ordered another VHDCI cable and more IDC plugs to setup multiple boards together and test all under full load - all outputs toggling with max speed, packed to same enclosure with front panels.
In a few days I should have setup with KC705, FMC-VHDCI and all EEM boards fitted to same box.
I want to make sure that even under worst case conditions all works as expected before I run production and send anything abroad.
So far only one bug in BNC EEM was found - wrong polarity of OE pin caused by wrong setting of Altium display option - the input is active low while on schematic is active high unless you switch "single \ negation" option in Altium settings. I hate this option in Altium - It caused troubles in past. By default this option is disabled and I did project with new Altium installation.

hartytp, 2017-05-14:
> @gkasprow In a few days I should have setup with KC705, FMC-VHDCI and all EEM boards fitted into same box. I want to make sure that even under worst case conditions all works as expected before I run production and send anything abroad.

> The LEDs are added to SMA version so far. I will also add to BNC but it requires panel redesign. I want to makes sure it works on SMA board first.

Thank you for the update.

I'd like to place an order for these boards in the next couple of weeks or so if at possible, so it's good to hear that your tests should be finished soon. 

> I can send existing BNC prototypes after tests, I have 4 sets of boards in my lab.

I'm not sure who paid for these, but we'll gladly taken them if no one else wants them...

gkasprow, 2017-05-14:
So far I paid for them :)

hartytp, 2017-05-14:
If you send us a quote, we'd be happy to buy the BNC boards and a VHDCI carrier off you.

closed 2017-05-15 by hartytp