sinara-hw/sinara #192: Sayma timeline

hartytp, 2017-05-15:
@gkasprow Can I check what the timeline is for Sayma? I assume it's something like:

| Action | Complete week beginning | Status | 
| ------------- |:-------------:| -----:|
|  First prototypes arrive at WUT | May 25 | Waiting  |
| Testing at WUT | 12 June |  |
| Fix issues | 19 June |  |
| Ship prototypes to M-Labs to begin software development | 19 June |  |
| Place hardware order | 19th June | |
| Groups receive hardware | 10th July |  |

Is that correct? If not, please edit with a more realistic timeline?

hartytp, 2017-05-15:
@gkasprow Also, please can you confirm that we (Oxford) have already reserved parts for 2xSayma boards? Is it too late to increase this to 5 Sayma boards?

gkasprow, 2017-05-15:
We produce 10 Sayma sets in first proto round (2 more than planned) so everybody interested will get one or two sets providing that we make them running. On 25 May we should receive 2 sets then hopefully, one week later we will give green light to assemble remaining 8. These two could be sent to mLabs and week later we will get 8 pieces to do test and send abroad. But this is very optimistic approach with no margin at all.

hartytp, 2017-05-15:
@gkasprow Thanks for the update. That's much faster than I'd expected, even on an optimistic timeline.

hartytp, 2017-05-15:
> We produce 10 Sayma sets in first proto round (2 more than planned) so everybody interested will get one or two sets...

If it was possible to increase our order to 4 or 5 boards, that would be really helpful. No worries if it's not possible at this late stage.

gkasprow, 2017-05-16:
It's too late.

hartytp, 2017-05-16:
No problem, thanks for letting me know.

closed 2017-05-16 by hartytp