sinara-hw/sinara #193: Urukul CS demultiplexer

sbourdeauducq, 2017-05-22:
The logic around IC4 (demultiplexing of CS0..2 into CS_DDS0..4, LE_ATT and CS_REG) is not clear to me. Are you sure it works, and if it does, can you explain how?

gkasprow, 2017-05-22:
@sbourdeauducq The logic enables you to select either single CS (Y1..Y4) or all DDS CSn at the time (Y7).
First case is clear, second needs DDR SPI where each DDS receives its own data in parallel.
DDR functionality is not implemented yet - these will be 4 D flip-flops.
Since CSn is active low, single AND gate does the job.

sbourdeauducq, 2017-05-22:
I see. Would be nice to get @jordens's opinion on this DDR and multiple SPI data lines scheme.
Do we really need a full shift register (IC19) for just one signal? And why do we need the DDS powerdown signals?

gkasprow, 2017-05-22:
@sbourdeauducq the board consumes a lot of power so we may need to disable unused channels. But this is yet to be decided. And that's why it is just idea to discuss and not finished schematics.

sbourdeauducq, 2017-05-22:
If that power-down mechanism really needs to be done, maybe some DIP switches or jumpers on each board are easier for both users and developers.

gkasprow, 2017-05-22:
Sure, everything that is more convenient for user is very welcome.

hartytp, 2017-05-22:
IMO power-down of individual channels isn't required at all. But, if we do go for it then DIP switches are fine.

jordens, 2017-05-23:
* Power down can be done in software as well. We should do that instead.
* Please let's not do DDR SPI here. I don't even see how that would work as there is state (phase of the slow SPI clock on the remote end) then. We'd need to track and reset that.
* While nice, we don't need the "ALL DDS" CS.
* And I'd like a simpler assignment for the attenuator/shift register CS.

I updated the wiki entry to what I extracted from the comments and discussion. Looks like we are heading for two versions of this board in any case.

https://github.com/m-labs/sinara/wiki/Urukul#idc-usage

@gkasprow Does that updated wiki page make sense?

sbourdeauducq, 2017-05-23:
How do you want to do it in software? Clicking buttons or entering commands isn't necessarily better UX than DIP switches. Auto power down is slightly better UX if it is done correctly (and there are many ways to get power management wrong, see ACPI for a pathological example) but it is expensive.

jordens, 2017-05-23:
It's a DDS register. I'd just expose it and then the user can power down things in their startup kernels. Fewer screws and fewer surprises than dip switches. But if the dip switches are for free I'd be ok with them.

gkasprow, 2017-05-23:
@jordens If you want to synchronise  several DDS chis, you cannot use internal PLL. External clock PLL must be used instead

jordens, 2017-05-24:
@gkasprow correct. But that's even tricky without the internal pll. We would need the mechanism described in the wiki. Or am I misunderstanding what you are referring to?

jordens, 2017-06-01:
The CS decoder assignment can be simplified (drop ALL-DDS, no DDR SPI). Wiki updated. Needs schematics updates.

closed 2017-08-27 by gkasprow