sinara-hw/sinara #195: Urukul AM

hartytp, 2017-05-23:
I'd ideally like to avoid having multiple versions of Urukul if we can, as it's more work to support and maintain them. Ignoring the issue of IDC usage for now, as @jordens nicely summarised on the Wiki:

- OptiClock need a 48-bit FTW
- We need a convenient way of feeding back to the RF amplitude with >~100kHz analog bandwidth. I suspect that this is a feature that most users would like and many will require.
- Neither of us are keen on the AD9914 for the reasons previously discussed.

As there is no DDS that meets these requirements, how about adding some form of amplitude modulator after the DDSs? Adding e.g. a MSPS DAC + mixer can be done with minimal board area/cost/power consumption. These would be multiplexed onto the SPI bus so that either the DDS FTW/POW is updated or the AM is updated.

Not sure if that's really less work than maintaining multiple versions of the hardware with different DDSs, but I thought I'd bring it up...

jordens, 2017-05-23:
To be useful for us that external AM would need to have very low residual AM to PM conversion. But yes: then it would be great.

hartytp, 2017-05-23:
Specification would be something like:

- Analog BW: >~100kHZ (~1MSPS clock)
- Dynamic range: >~6dB (only needs to be an AM trim, since we also have the attenuators for coarse amplitude setting)
- Resolution/noise: around 1e-4 (14 bits)
- residual PM: ? 

@gkasprow Suggestions about the best way to do this?

hartytp, 2017-05-23:
@jordens If you're worried about AM to PM, you can always disable the noise eater during interrogation pulses, and only enable it during certain "servo pulses" (e.g. dummy pulses during cooling). 

Either way any RF modulator should have less PM than the laser current...

hartytp, 2017-05-23:
> Neither of us are keen on the AD9914 for the reasons previously discussed.

Having said that, while I'm not keen on the AD9914, I don't feel strongly about it versus an external modulator. Cost and power dissipation will still be dominated by the RF Amp etc and not the AD9914. I don't think we have the IOs to hope the utilise the profile pins in a sensible way, so we should leave them out.

Either way, if we can find an approach which we all agree on and is well supported in gateware/software then I'm happy. @jordens Would you consider using the AD9914 (no profile pins) as the default option instead of AD9912?

jordens, 2017-05-24:
Just figured it out (again): AD9912 does not have an ASF (before the CORDIC), but a 10 bit full-scale current auxiliary DAC that enables more than 10 dB range (8 to 31 mA) with 10 bit resolution. That should be sufficient for PI regulation with the digital attenuators then adding another 30 dB range.

Disabling the noise eater during interrogation pulses does not completely help as there would still be PM from shot to shot.

The AD9914 burns ~2.6 W while the amplifier would be just under 2 W. Using the AD9914 more than doubles the total power consumption.

hartytp, 2017-05-24:
@jordens Thanks for being on top of that. I'd forgotten about the aux DAC too. That looks absolutely fine in terms of resolution and dynamic range.

Assuming there are no nastys in terms of programming the aux DAC, or large glitches when it changes value, that sounds ideal. What would the total cycle time be for a complete update of FTW/POW/aux DAC?

jordens, 2017-05-24:
In your case you'd probably be happy with a 32 bit FTW. Then you can write FTW and POW to 0x01A8-0x01AD sequentially and then FSC to 0x040B-0x040C  with 2*16+8*(4+2+2) + 2=98 SPI cycles. With 50 MHz, that's 1.96 µs. Whether 50 MHz is practical is another question as it may not match the RTIO coarse frequency.

hartytp, 2017-05-24:
@jordens Okay, that's not so much worse than we'd do with the AD9910. For 125MHz RTIO clock divided by 3, we'd be up to 2.4us, which is a bit on the slow side, but manageable. We could potentially shave some time off by ditching the POW if people don't need that, but let's worry about that later.

jordens, 2017-05-24:
Conceptually we could also run the ADC and DDS SPI at 20 ns (derived from RTIO) with RTIO at 8 ns and then align the SPI transactions so that they start at the coincident clock edges every 40 ns.

hartytp, 2017-05-24:
ACK

dhslichter, 2017-05-25:
@gkasprow @jordens @hartytp Stepping back for a second -- what's wrong with the AD9910?  It's similar to the AD9912 in terms of power dissipation, rails, performance, but it allows you to do proper amplitude modulation, and if we're deciding that a 32-bit FTW is good enough anyway...

hartytp, 2017-05-25:
AD9910 would have been my preferred choice, but OptiClock require 48-bit FTW.

We did consider going for two population options (AD9910/AD9912), but it's just more hardware/software to maintain and support. If the AD9912 auxuiliary DAC works out for AM then it seems like the simpler solution.

gkasprow, 2017-05-25:
@hartytp 
I prepared schematic version with 9912/9914 because @dhslichter needs more GS/s and profiles.

dhslichter, 2017-05-25:
I wouldn't say "needs" -- we at NIST will probably not use Urukul anytime in the near future because we already have equivalent hardware.  @gkasprow if you replaced AD9914 with AD9910 on the dual schematic, I think that's fine with us, and it would help solve a number of these issues -- @hartytp can have his AD9910's and OptiClock can have a 48-bit FTW with the board variant, and the rest of the plumbing/power budget/etc is similar.  AD9910 supports profile mode, if we want to pursue that with a second IDC...

hartytp, 2017-05-26:
I agree that having an AD9910/AD9914 population option is a good idea for risk mitigation purposes/future-proofing. However, our plan "A" is to try to use the same population option as OptiClock if we can. We'll see what the outcome of the measurements discussed in #205 is, but my expectation is that the AD9912 will work absolutely fine for our "simple AOM driver" use case.

A note on my general approach to this: adding population options is easy, but ensuring that they are well tested and documented and have proper long-term support takes much more work. If we're not careful, we'll end up with a large array of poorly supported Sinara designs/population options. (e.g. in this case, will @gkasprow even test the AD9914 population option? Or, will Urukul go into production without this ever having been tested and characterised?)

If a group wants to support their own designs in-house, that's fine and they can do what they want. But, if we want proper long-term support/CI etc from M-Labs then we need to recognise that our time/resources are finite and we should try to limit the number of designs in play.

jordens, 2017-05-26:
I'd be ok with having the AD9910 instead of the AD9914 as the "underside option". Just that for us it would still be the AD9912 primary because of the 48 bit FTW.

hartytp, 2017-05-26:
@jordens if the AD9912 option doesn't work out for the servo, do want to:

A. Make AD9914 the main population option, so we can all use the same hardware/software
B. Use the AD9910 population option for the servo? 

I don't feel particularly strongly either way.

jordens, 2017-05-26:
If the AD9912 FSC servo won't work, from the optIclock perspective, we would most likely still use the AD9912. Therefore I don't have a strong preference for A or B.

hartytp, 2017-05-26:
> If the AD9912 FSC servo won't work, from the optIclock perspective, we would most likely still use the AD9912. Therefore I don't have a strong preference for A or B.

To clarify, I meant "would optIclock consider using the AD9914 as well", so that we can all share the same hardware. It sounds like the answer is "no". So, the plan is:

- If AD9912 FSC servo works out, we all use that.
- Otherwise, we make two hardware versions/population options: Urukul_AD9912 and Urukul_AD9910 and support both.

hartytp, 2017-05-26:
@gkasprow 

- [ ] Can you make the second (reverse-side) population option AD9910, please? Just ground the profile pins.
- What screening enclosures would you recommend for this population option?
- Do you anticipate any mechanical/thermal/electrical issues with using the reverse-side population option? (e.g. will the screening cans on the reverse-side interfere with the adjacent PCB?) 
- Is it going to be much more expensive/difficult to produce Urukul with the AD9910 option (e.g. due to extra screening enclosures on the reverse side) than the AD9912 version? If so, we'll make a revised version of the PCB with the AD9910 as the primary DDS if we need to.

jordens, 2017-05-26:
For now, yes.
I just read through the AD9910 datasheet again. It may be possible to use the digital ramp rate generator to extend the frequency resolution almost arbitrarily (but still comfortably) by doing a ramp on the phase. But I'd have questions about how the DRG accu overflows with the limits both railed. This is a pretty unconventional and I can't judge it from the datasheet. Anybody know more or willing to test?
If the AD9910 DRG can be used to extend the FTW using a fractional POW ramp, we don't need the AD9912 at all.

hartytp, 2017-05-26:
@gkasprow Would you be willing to test this? (We can buy you a dev-kit if needed).

hartytp, 2017-05-26:
Otherwise, I'll buy a dev-kit now and can test this next week.

@jordens Can you post a detailed description of the measurement you'd like taken.

jordens, 2017-05-26:
* 1 GHz DDS clock
* Create in phase lock or obtain from that 1 GHz a good 100 MHz reference clock.
* 100 MHz-0.139 Hz DDS output frequency, single tone profile 0 FTW=0x19999999
* Then configure the DRG to sweep the phase to to kill that 0.139 Hz offset with step size 0x13333 and step delta t 0x1000, DRG limits at max/min, "no no-dwell", DRG destination phase.
* Measure the output frequency to be (within a µHz) 100 MHz using a mixer or some phase noise measurement tool or SDR. Look for glitches around 7.6 kHz where the DRG accu is supposed to wrap.

hartytp, 2017-05-26:
@jordens So, how about:

- 2 Synths phase-locked together at 10MHz
- 1 generates 1GHz DDS clock
- 1 generates 100MHz LO for level 7 mixer, used as phase detector
- DDS output frequency set to 100 MHz-0.139 Hz
- DDS DRG configured as you describe above
- Record mixer output on scope and post here

Additionally,
- Look at DDS output on spectrum analyser for presence of 7.6kHz spurs

hartytp, 2017-05-26:
AD9910 evaluation board ordered. @gkasprow I can post it to you when I'm done if that helps.

jordens, 2017-05-26:
@hartytp correct. Thanks.

Actually, to just confirm the functionality, you can do a 125 MHz - 3.8 kHz  FTW (0x1fffc000) and then use the DRG to cancel the offset (0x80000000 phase step and 0x1000 ramp interval). Just look for 125.000 MHz output and clean frequency (no unexpected sidebands). Or feel free to play with that scheme to make it more obvious whether the DRG supports proper wrapping of its accumulator.

dhslichter, 2017-05-26:
If we can swing this method of finer frequency resolution (which could be implemented transparently by the ARTIQ driver), then I vote for AD9910 only, no variants, no AD9914.  If not, I vote for being able to populate either AD9912 or AD9910 -- let's just forget AD9914 for Urukul at this point.

gkasprow, 2017-05-29:
@hartytp second DDS option is not a big issue here - there are just a few control lines so we don't need another layers. Bottom shielding probably would not be needed - I placed it just in case. We can use standard 3mm cans without troubles with adjacent modules.
So it seems that we can freeze specification and have assembly option for 9912 and 9910.

jordens, 2017-05-30:
@gkasprow yes. If we can do 9912 and 9910 that's good. But I remember just 4.1 mm from the reference plane (component side of the PCB) to the max extension of components on the "solder" side. That's 4.1 mm for PCB thickness and can.

hartytp, 2017-05-30:
> @hartytp second DDS option is not a big issue here - there are just a few control lines so we don't need another layers. Bottom shielding probably would not be needed - I placed it just in case. We can use standard 3mm cans without troubles with adjacent modules.
So it seems that we can freeze specification and have assembly option for 9912 and 9910.

My AD9910 evaluation board is here and I'm setting up this measurement now. Should have data today/tomorrow (depending on what else comes up in the mean time). Can we hold off from freezing the specification until I've finished my measurements? If we decide that the AD9910 should be the default population option then it would be more convenient to have it on the top side of the board.

jordens, 2017-05-30:
AD9912 and AD9910 if #210 does not work and AD9910 only if that technique does work.

closed 2017-05-30 by jordens

gkasprow, 2017-06-01:
@jordens so what's the conclusion?  We use AD9910 only?

jordens, 2017-06-01:
@gkasprow We should still accomodate both. There are a few wrinkles in that strategy.
Let's put the AD9912 on the underside, only implement the minimum required features for the AD9912 and otherwise focus on the AD9910 features.
I have updated the wiki page accordingly.

gkasprow, 2017-06-01:
@hartytp Once you finish with your devkit, could you post it to me? I'd need it to make my board running. I can send you my 9912 after I do tests if you want to play with it.

hartytp, 2017-06-01:
Will do. @jordens let me know when you're sure you don't want me to do any more tests with it, and I'll post it to Greg.

Greg, you can keep the AD9912 kit -- I don't need it, and it might be useful for you to have it later on.