sinara-hw/sinara #196: Urukul IDC plan

jordens, 2017-05-24:
There are currently two different approaches to wiring Urukul. Details and rationale are in the wiki:
https://github.com/m-labs/sinara/wiki/Urukul#idc-usage

@gkasprow can we do both on the same board (if needed with label swapping on the (b) case) or do we need two boards?

jordens, 2017-05-24:
Just as a sidenote: DDR SPI scares me much more than quad SPI. How would one handle the remote state (phase of the slow, divided SPI clock)? Interleaving the data would also require significant changes to the SPI core. Not even the top of the line Xilinx AXI SPI core does DDR SPI.

jordens, 2017-05-24:
IO_UPDATE should be separate from shift-registers. It is very timing sensitive and should not need two SPI xfers to do a single IO_UPDATE pulse (potentially competing with other SPI stuff).

gkasprow, 2017-05-24:
@jordens We don't need to divide SPI clock in DDR mode.
All DDS chips run with same clock, only data is aligned slightly to fit
setup and hold.

gkasprow, 2017-05-24:
@jordens Yes, we can define two assembly options or use mux controlled by
jumper that selects right signals.

jordens, 2017-05-24:
But SPI already uses both edges. One to shift, the other to sample.

hartytp, 2017-05-24:
> @gkasprow Yes, we can define two assembly options or use mux controlled by jumper that selects right signals.

Great!

Can we have mux controlled by a DIP switch? Silk text would read something like "basic"/"fast" for the two different switch settings.

gkasprow, 2017-05-24:
Yes,but it defines shifting for master and sampling for slave:)
So we can still use its both edges for data. Just multiplex two stream at
master side and use DFF to demultiplex on receive side.

jordens, 2017-05-24:
I also don't see how that works with the leading or trailing SPI half cycles. And how we'd handle MISO.
Do you have a circuit where this is implemented?
And do we need it still since (for four DDS) QSPI seems to achieve the same with the same number of wires?

gkasprow, 2017-05-24:
@jordens

You simply have SPI with 2 MOSI lines that go to the mux controller by
clock signal. TO write to DDS chips you use CS address where all DDSs are
selected.
Reading goes individully, in SDR mode.
So such approach does not need serious modification of SPI core - only time
multiplexingi is added to fit 2 channels into single line.
If we go for 4 wires then there is no issue at all.

dhslichter, 2017-05-25:
My vote is against DDR SPI as well, FWIW....seems like a very substantial additional complication.

gkasprow, 2017-08-27:
I added little, low cost CPLD (XC2C128-6TQG144C) to handle all logic. It's quite probable that connectivity will change during SW development. I'll take care about CPD project.

closed 2017-08-27 by gkasprow