sinara-hw/sinara #198: Urukul power

jordens, 2017-05-24:
@gkasprow 

* Regarding VFB1 and VFB2 of that LTM4619EV#PBF: don't you want to tap them from the same side of the choke in both cases?
* Do we need that fat DC/DC just because of the AD9914? Are there more quiet, smaller, cheaper converters (2x2A instead of 2x4A) if we'd go for just the AD9912?

gkasprow, 2017-05-24:
@jordens right, the feedback path should be identical.
I used this converter mainly because we use it already on Sayma RTM and it is low noise chip since all components are integrated in it. It also has high efficiency.
We could go for discrete version but they generate more EMI due to bigger physical size and have worse efficiency. We can go for 2.5A version - LTM4622 but is has lower efficiency.

jordens, 2017-05-24:
OK. I'll leave it to you then.

closed 2017-08-26 by gkasprow