sinara-hw/sinara #199: Urukul RF amplifiers

jordens, 2017-05-24:
from @hartytp:

Should we consider using a different output amplifier? AFAICT, there's no need for a special low phase noise amp here (e.g. we're using the DDS's internal PLLs), so we could go for something which dissipates less power and is a bit cheaper.

Not sure what P1dB people want for this design -- although, if it's used with the RF PA, which has 40dB of gain, then even 0dBm is enough -- but, something like a AG203-63, which can run directly from a 3V3 rail might be a good choice. This is a little cheaper and would shave 1.6W off the design, which would make thermal management a bit easier.

Edit: anyway, there are plenty of good options for this. @jordens if you specify a P1dB and lowest frequency of operation then we can make a choice.

jordens, 2017-05-24:
Yes. Without the RF amplifier it might be fine. Do I see that correctly that with 31 mA full scale current we'd be getting some 14 dB out of the DDS, and maybe 10 dBm after the switches, attenuators, filters, balun?

Does 31 mA saturate the balun?

gkasprow, 2017-05-24:
@jordens balun is specified for 0.4W and 30mA so it should not saturate it.

dhslichter, 2017-05-25:
Be careful here -- the voltage compliance of the outputs is only +/- 0.5 V, as I read the datasheet.  Not sure how this works with the claimed 31 mA output current, but it seems it would involve using some external resistors to keep the voltage in compliance, and then you are no longer at a 100 ohm differential impedance.  You may get more power out, but it might be lost in poor match with the balun/subsequent circuitry.  

Before proceeding I suggest @gkasprow test this with his AD9912 dev kit and look at power and any performance hit.  

The idea that we get 14 dBm out of the DDS sounds very suspiciously high to me, just based on what I've seen from actual DDS chips.  Again, should be tested by @gkasprow with the dev kit before any conclusions are drawn.

hartytp, 2017-05-25:
@jordens Do you have a target max output power for this design in mind?

jordens, 2017-05-26:
10 dBm out of the DDS and 8 dBm after the analog pipeline will likely be enough.

jordens, 2017-05-26:
Doing the math, [AD-912](http://www.analog.com/media/en/technical-documentation/application-notes/AN_912.pdf) we'd get ~1 dBm out of the DDS with 20 mA full scale current and a 1:sqrt(2) transformer.

dhslichter, 2017-05-26:
That's the sort of number that feels familiar for DDS output power -- I just don't think you can make 10 dBm at the DDS output, even with 31 mA current.

In terms of output amps, it seems to me that the ERA-2+ (lower power cousin of the ERA-4XSM+) might be a reasonable way to go, and provides the desired gain.  If you're not worried about running AD9914s it definitely helps with the power budget.

hartytp, 2017-05-26:
@jordens Can I confirm what your specification for the P1dB of the final design is? Do you require +8dBm P1dB at the SMA? We should try not to over specify this, as the power consumption + cost of the output amps increases steeply with P1dB. In general, the power amplifier used to drive the AOM will supply plenty of gain, so the DDS output power shouldn't need to be too high.

Other thoughts:
- We should move the amps to after the switches and attenuators to reduce the power the amps need to drive.
- We should add a fixed T-pad attenuator before the amp to allow us to ensure that when the DDS and attenuators are set to max power, the amp is running at its P1dB.

dhslichter, 2017-05-26:
I might put the amp after the attenuator but before the switch -- I worry about transients when the amp sees sudden changes in input power (and thus output power too).  Switch (HMC349) has only 1 dB IL, and the HMC542 attenuator is only 1.2 dB.  Based on minicircuits' inventory, if you want an amp that will give you 8 dBm P1dB, you're looking at 40 mA supply current (for meaningfully lower supply current, you are down to ~6 dBm P1dB), and you can find a bunch of amps that will give you up to 13 dBm P1dB for the same supply current.  I vote for amplifiers that give us a little more operating margin, so we don't have to run the amp into compression.  The ERA-3+ fits the bill (35 mA supply current, ~22 dB gain so we can adjust max output power with a T-pad, ~12 dBm P1dB, good return loss).

hartytp, 2017-05-26:
> @dhslichter I might put the amp after the attenuator but before the switch -- I worry about transients when the amp sees sudden changes in input power (and thus output power too). Switch (HMC349) has only 1 dB IL, and the HMC542 attenuator is only 1.2 dB

If Urukul's main purpose is to drive AOMs then I don't think that transients in the pre-amp will be a serious concern, as they will be completely dwarfed by the power amplifier transients.

Having said that, as you point out, the losses in the switch are pretty low, so I don't care too much whether it goes before or after it.

dhslichter, 2017-05-26:
We keep harping on this "drive AOMs" point -- I guess I'd argue that you can drive AOMs either way, but for various other applications it might be nice to have less transient sensitivity (for example, if you're driving a microwave IQ modulator IF port).  There is value in making Urukul useful for as many potential end uses (and potential end users) as we can, with the caveat that we don't want to overload it with extra capability and features far beyond what is necessary.  So I guess I argue for ERA-3+ with attenuator and switch after the amplifier.

hartytp, 2017-05-26:
@gkasprow Whatever decision we make about this, please can we add a pad for a DNF resistor, which can be fitted instead of the amplifiers if desired? That way, users who don't want the amplifiers can leave them out and save a couple of Watts.

jordens, 2017-05-27:
Technically, P1dB is an amplifier spec (two-port) and not a "DDS/oscillator spec" but I see what you mean. Using an amplifier with a P1dB >10dBm is fine. Attenuator-amp-switch is fine. ERA-3+ in the package du jour sounds good.

gkasprow, 2017-05-29:
@hartytp OK, I added DNF resistor. 0603 fits nicely.

hartytp, 2017-09-01:
@gkasprow When do you think that Urukul will be sent off for manufacture? It looks like it's basically complete.

jordens, 2017-09-01:
We don't even have the changes from the first schematics review implemented. There needs to be a second one and then there needs to be all the layout work.

hartytp, 2017-09-01:
> We don't even have the changes from the first schematics review implemented. There needs to be a second one and then there needs to be all the layout work.

True. More accurate comment: "it looks like the first round of changes are basically complete ."

gkasprow, 2017-09-01:
@hartytp @jordens there is just one schematic page that needs update. Hope to publish it this evening.

hartytp, 2017-09-01:
@gkasprow Excellent! Thank you.

closed 2017-09-02 by hartytp