sinara-hw/sinara #201: Urukul clocking

jordens, 2017-05-24:
* As @dhslichter  and @hartytp  have suggested, dropping the ADCLK948 and the LTC6957 and using solder jumpers for the clock source (front panel or internal) sounds like a good idea to me. The clock inputs are 2.6 kOhm impedance anyway. Is there a problem with isolation, trace impedances?
* But I'd also be fine with a economic few-100mW CMOS fan-out for the reasons stated in #191. Will defer to @hartytp.
* I'd also like to see crystal pads (if needed: with solder jumpers) and load capacitors on each DDS clock input. Those seem free.

hartytp, 2017-05-24:
> The clock inputs are 2.6 kOhm impedance anyway. Is there a problem with isolation, trace impedances?

That's in parallel with 1.5pF, which is only 100Ohms at 1GHz. Plus, I'd generally want to terminate the clock inputs properly. 

> I'd also be fine with a economic few-100mW CMOS fan-out for the reasons stated in #191.
> But I do want to be able to feed up to 500 MHz directly to Urukul, preferrably 1 GHz.

I'd prefer to keep a buffer in for reasons discussed previously.

I'm also happy using solder jumpers for input source selection. @gkasprow Can you select the cheapest, lowest power consumption clock buffer that:

- Has >= 5 outputs
- Won't degrade the AD9912 PLL on phase noise
- Goes up to 1GHz
- mux would be nice, but definitely not required

hartytp, 2017-05-24:
> I'd also like to see crystal pads (if needed: with solder jumpers) and load capacitors on each DDS clock input. Those seem free.

Good idea. One thought: for a 4-channel board it might make more sense to use a single VCXO, which can be muxed into the clock buffer, rather than 4 crystals. That way, the DDS loop filter etc can be left the same for operation from an ext clock or internal Xtal, making life a bit easier for the user. 

There are tonnes of options in standard packages. There are plenty of really cheap ones, but if you want to splash out ~USD20 you can go for the same oscillator we put on Baikal: CVHD-950-100.000 which consumes 15mA*3.3V=50mW and has pretty awesome phase noise.

hartytp, 2017-05-24:
> Is it really the best solution to have the differential INT_CLK_IN (from Kasli)? And to have them go towards the back (instead of vertical for cables with angled connectors)?

I'd wondered that too. I'd rather keep this single-ended (1 cable) if possible; there is already a bit of a rats' nest inside the rack, so no need to make it worse by using 2 cables when 1 would do. Vertical next to the IDC makes more sense that RA facing towards the BP IMO.

jordens, 2017-05-24:
Ack the VCXO and keeping the fan-out.

gkasprow, 2017-05-25:
@hartytp @jordens You wrote VCXO which means that it is voltage controlled which means that we need some PLL to close the loop. Do you mean standard crystal oscillator ?

hartytp, 2017-05-26:
@gkasprow Sorry Greg, that was a typo. You're right, I meant XO.

hartytp, 2017-07-03:
Note to self:

- Anticipated DDS clock input frequencies:
  - RTIO_COARSE_CLK/CDR_CLK - 125MHz (e.g. from Kasli). see #75 for anticipated noise spectrum  also [Si5324](https://www.silabs.com/documents/public/data-sheets/Si5324.pdf).
  - XO: 100MHz [CVHD-950-100.000](http://www.crystek.com/crystal/spec-sheets/vcxo/CVHD-950.pdf)
  - SMA: user selectable, typically the same 100MHz ultra-low noise oscillator used for Sayma reference clock

Phase noise (numbers roughly read from plots and rescaled to 100MHz reference frequency):

| Source | 1kHz | 10kHz | 100kHz |
|--------|-------|-------|----------|
| XO     |  -143dBz/Hz | -157dBc/Hz | -164dBc/Hz |
| Si5324 | -130dBc/Hz |  -135dBc/Hz | -135dBc/Hz |
| AD9910 PLL (fig 16) | -125dBc/Hz | -137dBc/Hz | -139dBc/Hz |

Conclusion: 

- Give or take, we can use the same loop filter/charge pump for all clock sources, since 100MHz is pretty close to 125MHz. A high (100kHz or higher) bandwidth loop would be best for the XO/SMA_CLK, and should also work fine for the CDR/RTIO clock source.
- The CDR clock is almost as good as the DDS with the PLL on, so an external clock source shouldn't be required.
- To keep the DDS clock at 1GHz, we need different PLL multiplication factors for the CDR/RTIO clock versus the XO. Changing the multiplication factor requires a bitstream rebuild on the current servo plan. (We could pick a 125MHz XO, but I'm not sure it's worth the hassle...)
- For all of the boards that we (Oxford) buy, we'd like to use the XO as the primary clock source to minimise the wiring rat's nest.

hartytp, 2017-09-02:
Fixed in recent schematic version.

closed 2017-09-02 by hartytp