sinara-hw/sinara #202: Kasli backplane

jordens, 2017-05-24:
@gkasprow 

* The Kasli backplane looks great. Being able to just add a trivial 96-pin-DIN to 4x IDC adapter as a trivial backplane is nice.
* Could we group them so that the first EXT occupies a1,b1,c1,...,a8,b8,c8 (including its I2C and 1x3.3V mgt, 2x12 V, 3x GND) and the the next EXT the following pins etc?

hartytp, 2017-05-24:
@jordens Is this just as an overflow, since we have enough IO for 12 IDCs but not enough space on Kasli for more than 8?

How would the BP PCB mount? Is it just held in place by the 96-pin-DIN, or does it screw on to something?

How easy will it be to access those extra IDCs when this is used in a standard rack?

Might it be better to use a standard vertical medium-density connector instead, and mount those extra IDCs on a mezzanine board that sits above Kasli?

jordens, 2017-05-24:
We seem to have enough IO and there does not seem to be enough space on Kasli for four more IDC.
The racks have proper mounting mechanics for those backplane on the DIN connectors. It's fixed in the rack and Kasli plugs into it.
A different backplane (not the simple one that does the 4x IDC adaptering) would then provide DIN connectors for EEM modules that prefer those over the IDC connectors.
It would be easy to access the IDC connectors on the backplane. You just open the rack (from the top). The IDC connectors on the backplane would point towards the front. You can then optimize the backplane width for convenience.

hartytp, 2017-05-24:
Okay, sounds good to me then.

marmeladapk, 2017-05-24:
@hartytp @jordens Please note, that with XC7A50T we have enough IOs for **10** extensions, those 2 additional will be connected and routed to additional bank that is available in XC7A75T (in case we would use it). XC7A75T has same footprint and more resources.

jordens, 2017-05-24:
@marmeladapk ACK. Thanks for the reminder. Is the volume price difference roughly the 35% I am seeing online (~102 vs 68 EUR/USD)? Maybe the 75T is the way to go for the first test and first batch.

marmeladapk, 2017-05-24:
I think that @gkasprow has 50T cheaper than seen online from other source.

hartytp, 2017-05-24:
@jordens Do you think the Kasli FPGA will be a limitation when running Kasli as an ARTIQ master? (ping @cjbe who had some concerns about this).

jordens, 2017-05-24:
@hartytp We'd need to do some research to answer that conclusevely.

gkasprow, 2017-07-02:
Current version of Kasli schematics use 100T version which enables 12 EEMs. With 50T we can use only 10 EEMs. Take into account that EEMs 9-12 don't have the clock routed. I have 2 free pins on the backplane connector and we can use them for this purpose. But this would need clock IC with 12 outputs or balun connected to one of negative, unused outputs of clock IC. This approach I implemented in current version of schematics.

hartytp, 2017-07-04:
Fixed in current Kasli schematics.

closed 2017-07-04 by hartytp