sinara-hw/sinara #203: Kasli power

jordens, 2017-05-24:
From @dtcallcock:

- Can we use thread-lock barrel connectors on the boards?  These are much more robust in a lab setting and are backwards-compatible with non-locking plugs for those that don't care.  Example parts are [connector](http://www.digikey.com/products/en?keywords=sc1472-nd) and [plug](http://www.digikey.com/product-detail/en/switchcraft-inc/761K/SC1148-ND/241919).

hartytp, 2017-05-24:
@gkasprow Can we have a power budget as well, please?

hartytp, 2017-05-24:
[ ] @gkasprow Please can you change the power connector on the VHDCI carrier to match Kasli?

hartytp, 2017-05-24:
- [ ] Once we've finalised the number of IDCs on Kasli (including the backplane ones), we should make sure that we've properly specified the maximum +12V current that each IDC can draw, and that the power distribution network on Kasli (connectors etc) can handle the total current requirements of all IDCs/EEMs.

gkasprow, 2017-05-24:
I planned to change it some time ago but missed it somehow.

hartytp, 2017-07-04:
done

closed 2017-07-04 by hartytp