sinara-hw/sinara #205: Test AD9912 FSC

jordens, 2017-05-24:
Could someone with a AD9912 board (eval or custom) test the behavior of the full-scale current DAC register w.r.t. glitches, speed, and (if possible) parasitic AM-to-PM conversion?

Maybe @gkasprow or @hartytp or someone from Duke? @jbqubit ?

#195

hartytp, 2017-05-24:
We don't have an AD9912 here, so can't do this without buying an eval board first.

gkasprow, 2017-05-24:
@jordens I have a devkit.
Give me more details about measurement conditions.

hartytp, 2017-05-25:
I'd say the following:

- Set the DDS to some reference frequency. Say 100MHz.
- Toggle the aux DAC between mid-scale and mid-scale + 1LSB as fast as you can (~1MSPS?)
- Look at the DDS output on a scope and see if there are any obvious glitches
- Look on a spectrum analyser and see if the aux DAC toggling introduces any new spurs on the RF output

gkasprow, 2017-05-25:
@hartytp 
I don't know if I can click that fast :) The devkit software talks over USB so it is rather slow.
I can generate signle DAC update and see the result. I'd sync scope with SPI activity.
Isn't there any other way to toggle DAC?
Of course one can connect FPGA and quickly write to it.
I can also connect the DDS to 3U PCB with TTL outputs and do it from ARTIQ direcly but here I would need some support

hartytp, 2017-05-25:
> I can generate signle DAC update and see the result. I'd sync scope with SPI activity.

That sounds like a good plan (better than practising for MSPS clicking speed). It's not as sensitive as toggling at a constant rate and looking for spurs on a spectrum analyser, but it will pick up major glitches, which is all we're after for now.

dhslichter, 2017-05-25:
Also please test for distortion and measure output power as a function of current -- I am worried about voltage compliance of the output stages.

hartytp, 2017-05-25:
@dhslichter Good point, thanks for raising this.

@gkasprow Probably the best thing to do is:

- [ ] Measure and plot the power in the carrier and first few harmonics as a function of R_DAC_REF. Do this with the auxiliary DAC set to max, half and min. Based on this data, we can pick the highest DAC current (output power) that gives "acceptable performance". This should allow us to use the full dynamic range of the auxiliary DAC. Use a sensible carrier frequency like 200MHz for this measurement.

"acceptable performance" can be defined somewhat loosely here, since the AOMs and power amplifiers that Urukul will be used to drive are not high-linearity/low-distortion.

jordens, 2017-05-26:
@gkasprow Let's keep it simple. Note RSET (the resistor on DAC_RSET) on your eval board, use a 1 GHz DDS clock and a 100 or 200 MHz DDS output, capture (on some reasonable time scale that shows the envelope settling time and possible glitches) the DDS output transient (a) when setting the FSC (at 0x040B, 0x040C) from 0x01ff (default) to 0x0200 (i.e. the major carry), (b) the transient for 0x0000 to 0x01ff, and (c) qualitatively how much the output degrades (harmonics) when using an FSC of 0x02ff compared to the default 0x01ff. For the AM-to-PM coefficient, measure (d) the phase of the DDS output (w.r.t. the 1 GHz DDS clock) before and after doing the 0x0000 to 0x01ff transition. But I doubt you will see something beyond mrad there.

From the AD9858 I do remember seeing noticable distortion when changing RSET to higher currents and conversely some improvement when going lower, but not at worrying levels. I'd expect FSC to behave similarly here.

Interestingly (and annoyingly) the FSC registers are "live": they become active immediately without IO_UPDATE, and, as you mention, you will need to time on SPI traffic.

jordens, 2017-05-26:
Drawing conclusions from my observations: You'll probably have problems making the those transitions since the register is split and both parts are live. Then just to the 0x017f to 0x0180 transition for the major carry jump and the 0x0000 to 0x0200 for the large scale jump.

hartytp, 2017-05-26:
> You'll probably have problems making the those transitions since the register is split and both parts are live. 

That sounds like a problem for servo mode, doesn't it? Transitions around mid-scale will cause huge amounts of amplitude noise, right?

jordens, 2017-05-26:
The behavior is just conjecture. Would be good to confirm that in practice as well. But yes: for amplitude servoing you'd constrain it to the lower 8 bits only if the behavior is as conjectured. That's 4.4 dB of range left for the 0x0000 to 0x00ff mode and 2.9 dB for 0x0100 to 0x01ff.

hartytp, 2017-05-26:
Okay, we should really test that then. 4.4dB of dynamic range is getting pretty marginal -- we could probably make it work, but it would require more fiddling around with attenuator settings than I'd like.

gkasprow, 2017-05-26:
@jordens  @hartytp From devkit SW I can only set limited number of registers
![obraz](https://cloud.githubusercontent.com/assets/4325054/26494186/ead378ee-421d-11e7-807d-aca4ff7d72a1.png
Should I measure DAC output directly or the lowpass filter output?

hartytp, 2017-05-26:
@gkasprow Thanks for looking into this. Given the discussion in #195, it probably makes sense to do AD9910 DRG measurement first. If that works out then we'll ditch the AD9912, so this measurement won't be required.

I should be able to make the AD9910 DRG measurement on Monday or Tuesday.

jordens, 2017-05-26:
@gkasprow if you want to: lowpass filter output, and then for (a; the major carry glitch) look at the 11.4975 to 11.52 mA DAC current transition. And for for large scale behavior, set it to 8.64 mA (load) and then do 20 mA (hit load and record the transition). I am assuming that "LOAD" triggers the register write. Disable Auto I/O update.

Otherwise let's wait for the AD9910 DRG test.

gkasprow, 2017-05-26:
I have already all setup done with spectrum analyzer, scope and RF generator so doing any measurement is trivial.

jordens, 2017-06-01:
@gkasprow If you have the setup, could you run that test above?

gkasprow, 2017-06-01:
@jordens @hartytp I togled between 8,64 - 20mA and back.
Below is what I got. I triggered it with SPI transfer.
So 8.64 -> 20mA
[image]
[image]
[image]

20 -> 8.64mA
[image]

gkasprow, 2017-06-01:
when I switch from 8.64 to 10mA I get:
[image]

hartytp, 2017-06-01:
@gkasprow Does that mean that it takes ~50us for the auxiliary DAC to switch from 8.64mA to 10mA?

gkasprow, 2017-06-01:
@hartytp that's really surprising to me. And that's why I switched on bus decoding. I just realised that the DAC current is changed immediately after MSB or LSB is updated so we observe two steps. Switching time seem to depend on DFSC increment. I can do more measurements tomorrow, this time with IO_Update signal.

hartytp, 2017-06-01:
thanks for checking that @gkasprow. Sounds like the aux DAC is going to be too slow for the feedback applications I have in mind.

jordens, 2017-06-01:
@gkasprow 
* This is independent of IO_UPDATE.
* As described above, the best observable major carry glitch is from 11.4975 to 11.52 mA (0x07f to 0x080) please check that. But from your figures, I don't expect visible glitches.
* A reasonable large step would be 8.64 to 20.16 mA (0x000 to 0x200) it'll probably look like your second test.
* In both cases it'll touch only one register.

* It looks like it will generally slew with 6mA/20µs, glitch-free apart from the split register.

jordens, 2017-06-01:
I have seen enought. Thanks again @gkasprow for testing this and @hartytp for testing #210! Let's do AD9910 primary and AD9912 on the underside.

closed 2017-06-01 by jordens

hartytp, 2017-06-02:
Thanks @jordens and @gkasprow!

AFAICT, that's all the design decisions finalised for Urukul.

hartytp, 2017-06-02:
@gkasprow The AD9910 Eval is in the post to you. You should have it early next week.

gkasprow, 2017-06-05:
@hartytp Thanks, I got it.
Now Sayma has the highest priority so I fully concentrated on it.