sinara-hw/sinara #206: Fast Novogorny

hartytp, 2017-05-24:
Upgrade Novogorny to work with Urukul/Kasli servo.

- Dip switch to select between "basic" and "fast" modes
- Basic mode: single SPI bus shared between all ADC channels; uses 1-2 IDC(s) and standard Kasli gateware; 1MSPS shared between all channels (e.g. using CS lines)
- Fast mode: 1MSPS per channel; 2 QUAD/1 OCTAL SPI bus; 2 IDCs; ADCs acquisition to be synchronised with Urukul DDS updates in Kasli servo gateware

sbourdeauducq, 2017-05-25:
Why have the basic mode? Is it just to be compatible with the existing SPI core?

hartytp, 2017-05-25:
> Why have the basic mode? Is it just to be compatible with the existing SPI core?

Yes. Adding basic mode is very low cost (a dip switch and some gates), and allows one to use Novogorny for real-time data acquisition with only the standard Kasli gateware if one doesn't need the extra speed.

In the shorter term, before the custom servo gateware is finished, it will also allow us to do intensity stabilisation on Urukul using kernels.

sbourdeauducq, 2017-05-25:
I'm not sure if it makes sense to add hardware gates instead of pushing them into the FPGA - if we already have gateware that is flexible enough for switching between TTL and SPI on a IDC, it could as well support "muxed SPI".

hartytp, 2017-05-25:
Agreed, doing the SPI muxing in gateware would be nicer than doing it in hardware. But:

- Can we do this in a way that's flexible enough to be genuinely useful and cover the majority of use cases?
- IIRC @jordens was worried about the FPGA resources that this kind of thing would require.

I'm happy to go with whatever you and @jordens recommend on this.

sbourdeauducq, 2017-05-25:
That would not require much FPGA resources - the 7A50T is equivalent to hundreds of thousands of 74xx chips.

hartytp, 2017-05-25:
@sbourdeauducq Sounds good -- the more flexible we make the standard Kasli gateware the fewer custom builds we'll need.

Can you spell out in detail exactly how this muxed SPI should work (inc IDC pin assignments), maybe on the Wiki?

gkasprow, 2017-05-25:
@hartytp @sbouhabib 
It would be hard to make board compatible with both simple mode where 1 or 2 IDCs are used (for high density projects) and high speed dual IDC.
There must be at least one mux, but this does not complicate the design much.

sbourdeauducq, 2017-05-26:
The pinout depends on whatever the ADC we end up using requires, but the basic idea of muxed SPI is to decode CS inside the FPGA and select the relevant MISO line to make it look like a single SPI bus. This is equivalent to what your gates would be doing in "basic mode".

hartytp, 2017-05-31:
Initial plan for Novogorny II, summary of email conversation between @sbourdeauducq @jordens @gkasprow and @hartytp. @gkasprow please check below and feel free to correct/change as you see fit:

- Keep current (Novogorny I) AFE (input filter + InAmp)
- Replace ADC with LTC2320-16 8-channel, 1.5MSPS simultaneous sampling. @gkasprow can you double check that there isn't a more suitable choice of ADC before we finalize this, please?
- Use external 5V reference with ADC
- Divide InAmp outputs from +-20V to +-2.5V. Divider's accuracy should be 0.1% or better, TCR should be chosen to avoid degrading the ADC's +-15ppm/K accuracy. Either use precision resistors or a cheap divider, such as AORN1002AT3 (£2.50 in quantity from Farnel). Filter capacitor in parallel with the divider to give a second pole at 250kHz (as in current design).
- Offset InAmp outputs by +2.5V (Vref/2) using a cheap divider (2 resistors + filter capacitor, OpAmp buffer if required) and the InAmp ref input. Voltage range after divider is now 0V to +5V to match ADC inputs.
- Drive ADCs in pseudo-differential configuration as in datasheet figure 4, using the ADCs CMRR to subtract Vref/2 from the signal. Filter pole here should be set to 1MHz. Choose a suitable OpAmp for this: cheap and low power consumption; slew rate should be at least 20V/us / 4=5V/us to avoid degrading the InAmp slew rate; GBP >= 1MHz; +-12V supply. FWIW, when I looked at this the best choice seemed to be the OPA2197 (same used in Zotino).
- Do we need to do anything about power sequencing here to ensure that the ADC input doesn't exceed Vdd during startup (e.g. if a signal is connected when the device is powered on)?
- Use LVDS interface with two IDCs as follows:

IDC0:

0. ADC_SCK
1. NC
2. ADC_SDOA
3. ADC_CNV
4. SR_SCK
5. SR_MOSI
6. SR_MISO
7. SR_CS
  
IDC1:

0. ADC_CLKOUT
1. ADC_SDOB
2. ADC_SDOC
3. ADC_SDOD
4. NC
5. NC
6. NC
7. NC

- "Basic" interface mode: works with standard Kasli gateware, using only IDC0. ADCs must be read out in order. Max sample rate depends on length of cabling etc. With a 20 MHz SPI read clock (25ns round trip plus setup), we get one sample one each channel per: 30ns tCONVH + 450ns tCONV + 8 channels * 16 bits * 50ns = 6.9µs
- "Fast" interface mode for servo loops etc: use 2 IDCs source-synchronous bus (sampling on ADC_CLKOUT) using ADC_SDOA to ADC_SDOD. 1.5MSPS on all channels should be possible (200MHz clock)

closed 2017-10-01 by hartytp