sinara-hw/sinara #207: Commit and tag

hartytp, 2017-05-25:
@gkasprow There is now some prototype hardware on order or under test but, AFAICT, the latest schematics haven't been pushed yet. 

- [ ] Please would you push the latest versions of all projects that have been sent for manufacture
- [ ] Please would you tag the commit with a list of any manufactured hardware. This will make it easier for us to find the right version of the schematics later.

gkasprow, 2017-05-25:
@hartytp I've just checked, all files are up to date.
To add commit I have to modify the files. At the moment there is nothing to commit.
They are allready marked: "added production files"

hartytp, 2017-05-26:
> @gkasprow I've just checked, all files are up to date.

Okay. I though there were more recent versions of these files because a spotted a few small things like:
- On the 3U ADC, we agreed that we'd change the SMPS supply voltage to match the DAC board. It's still +-15V in the current schematic
- On the 3U ADC schematic PDF in the 3D rendering of the board, the screening enclosure doesn't have cutouts for the termination switches. But you posted a rendering on the issue tracker, which did have the cutouts. So, I assumed there was a more recent version. 

> They are allready marked: "added production files"

Next time, could you add more details to the commit message. For example, which version of the production files (prototype round 1? full production version 1?), and whether this is a "frozen" design that's going off for manufacture or schematic for final review before manufacture. This will help a lot when one tries to locate a schematic a year or so after production!

To make it easier to find schematics for production/prototyping rounds, it might be worth using the "tag" feature in the future as well as the commit message.

closed 2017-05-26 by hartytp

jordens, 2017-05-26:
@hartytp just nitpicking: you'd usually not want add "prototype round 1" or the like to the commit message because you'd eventually forget it and because you are never sure at that point in the workflow that this is actually the release or whether there will be last-minute changes. You'd just describe what you did in the commit message and then use tags for tagging things you "released" after or concomitantly with that action. Tags can be applied, removed, moved later.

hartytp, 2017-05-26:
@jordens ACK. @gkasprow from now on, each time hardware goes to production, can you tag the commit containing the relevant files, please?

klickverbot, 2017-05-26:
I'd also suggest using annotated tags (`git tag -a` on the command line, clients probably also have an option), where you could then add a quick description like "Revision <N> of <board>, as sent off to <XYZ>".