sinara-hw/sinara #208: Update the wiki

hartytp, 2017-05-25:
We should get the Wiki up to date, so that we have a clear resource to point prospective new (and eventually current) users of Artiq/Sinara to for information. It would be good to make some decent progress on this before NACTI, as I imagine that Artiq/Sinara will be a topic of conversation there. 

**Higher-level pages**: 

These need to give uses a high-level description of what hardware is available and how it all fits together, linking to the relevant card-level pages. They should explain some of the ways the hardware can be used to build an experiment, making clear what design choices have been made and why, and providing some illustrated example configurations (e.g. block diagram with: host PC; Metlino/Sayma in a rack; Kasli drtio slave in a EuroRack with Urukul, Novogorny, Zotino and BNC; Sayma DRTIO slave; etc).

- [ ] home
- [ ] uTCA overview
- [ ] Eurocard overview
- [ ] Time distribution
- [ ] Status/purchasing pages

Who will take this on? Is this something that @jordens and @sbourdeauducq can do?

**Card-level pages**: 

Should contain an overview with the following (when available):
- Brief description of what the board does and how it is intended to be used
- Photograph 
- Link to detailed WUT documentation
- Link to schematic and other files on GitHub
- Link to any relevant Artiq pages

Should also contain a bullet-pointed specification, including things like:
- Rough cost
- Power consumption
- Size
- Number of IDCs

List of pages and an (optimistic?) initial guess at who might write them (feel free to reassign) based on who funded them and has been involved in their design:
- [ ] Metlino: @jbqubit
- [ ] Sayma: @jbqubit, @dhslichter, @hartytp
- [ ] Baikal: ?
- [ ] Clock mezzanines: @hartytp, @WeiDaZhang 
- [ ] Sayma AFE: @jbqubit, @dhslichter, @hartytp
- [ ] Kasli: @jordens 
- [ ] VHDCI carrier: @jbqubit 
- [ ] BNC: @jbqubit 
- [ ] RJ45: @jbqubit 
- [ ] SMA: @jbqubit 
- [ ] Zotino: @hartytp
- [ ] Novogorny: @hartytp
- [ ] Urukul: @jordens  (@hartytp for the Servo?)
- [ ] RF PA: @hartytp

hartytp, 2017-09-06:
I'd like some high-quality photos of all existing boards to put on the Wiki (good way of reminding that this isn't just vapourware ;) ) @sbourdeauducq IIRC, you took some nice photos of Sayma. Are you happy to do this for the other EEMs? Which EEMs do you currently have complete with front panels?

sbourdeauducq, 2017-09-06:
I'm happy to take good pictures (note that the Sayma pictures were from Robert, not me). But all I have is three Sayma AMC cards without RTM that were delivered late (which is why Robert ended up taking the pics), VHDCI carrier, BNC breakout, Novogorny, and Zotino without front panel. And I do not have a Eurorack either.

hartytp, 2017-09-06:
Okay, thanks. 

Let's wait until we have nice FPs for these EEMs then.

I want to start with Zotino, and make this into a bit of a template for the other EEM pages. I'm happy to handle the text, but can someone with more advanced Markdown/html skills than me do some work on making it look nice (e.g. sorting out the images properly)?

hartytp, 2017-09-06:
Okay, done a quick tidy up on Zotino.

I'll finish it once the board's been tested, but let me know if there is anything we should add/change...

hartytp, 2017-09-24:
cf https://github.com/m-labs/sinara/wiki/StyleGuide

jordens, 2017-09-24:
Good! I also have some pandoc stuff around to make nice PDFs out of them automatically. Just need to clean it up and push it.

hartytp, 2017-10-04:
I've had a go at finishing off the Novo wiki page for this revision. There are a few blanks that will be filled in once we test the v2.0 hardware, but it's pretty much there.

Feel free to comment on anything that is unclear/missing or general style. Otherwise, I suggest we use this as a template for the other pages.

hartytp, 2017-10-04:
@jbqubit Can you have a go at doing the pages for the DIO EEMs + VHDCI_CARRIER, please? It would be nice to have a completed Wiki page for all hardware that is currently available.

hartytp, 2017-10-06:
@jordens I'd like to remove the artiq hardware page from the Wiki, since a lot of the information it contains is badly out of date and confusing to new users. Do you object?

jordens, 2017-10-06:
If you make sure that all information still in there has been integrated elsewhere, then that's fine with me.

hartytp, 2017-10-06:
AFAICT, the bulk of the info on that page is either integrated elsewhere or obsolete/incorrect. But, there are a few things, like the "Data and sample rates" section that might be of use. Can you have a quick skim over it and see if there is anything you particularly want to keep?

jordens, 2017-10-06:
* The background and objectives needs integrating into the introductory pages (_Home)
* I can't find the "System overview", nor the "Sayma gateware/runtime specification", nor the "Sayma SAWG data rate constraints" nor the "Data and sample rates" chapters anywhere else. 
* Metlino has less information in the wiki than on that page. Especially on the conceptual level.

hartytp, 2017-10-06:
ACK. I'll copy those bits into the relevant sections and then delete this page.

hartytp, 2017-10-06:
@jordens The "Sayma gateware/runtime specification" section seems to be about gateware rather than hardware. Shouldn't this all be replaced with a link to the phaser docs?

jordens, 2017-10-06:
The gateware vs hardware is the same as with nu-servo. Both should move to the code once that's done.
On the phaser parametrization, a lot of that is not (yet) implemented. It's still a design proposal.

hartytp, 2017-10-06:
Okay. So I'd prefer to move this to the phaser wiki now, as it makes more sense there than on the Sayma page. Can we do that?

jordens, 2017-10-06:
Can't you just drop the pages from _Sidebar/Home? Doesn't that have the same effect of moving them further out of reach? There is also no Phaser wiki.

hartytp, 2017-10-06:
> Can't you just drop the pages from _Sidebar/Home? Doesn't that have the same effect of moving them further out of reach? There is also no Phaser wiki.

Done. I was just doing some de-krufting while waiting on a build.

Changes:
- Removed merged the general overview section and the useful parts of the metlino section
- Deleted all the obsolete parts apart from the Sayma gateware/clocking bits
- Removed this from the sidebar
- Renamed it Sayma gateware

The clocking stuff should probably be moved into a separate page, which gives a good overview of clock distribution in Artiq/Sinara and explains the various relations/constraints between/on the clocks. However, that's a project for another day.

jordens, 2017-10-09:
@hartytp please undo the removal of the minutes.

jordens, 2017-10-09:
I went ahead and reverted that commit.

hartytp, 2017-10-09:
Okay, was just about to do that. Thanks.

The Wiki was a bit of a state, so I deleted some stuff that didn't seem relevant any more when I tidied it up. Feel free to revert as you see fit.

jbqubit, 2017-10-12:
> @jbqubit Can you have a go at doing the pages for the DIO EEMs + VHDCI_CARRIER, please?

Yes. But it will be a week or two before I can tackle this.

hartytp, 2017-10-12:
@jordens Thanks for uploading the nice EEM images.

gkasprow, 2017-10-26:
@jbqubit @jordens I took some photos of MTCA [crate](https://github.com/m-labs/sinara/blob/master/Drawings/MTCA_Front.jpg).
I'm not sure where to place them on wiki.

gkasprow, 2017-10-26:
another[ one](https://github.com/m-labs/sinara/blob/master/Drawings/MTCA_Back.jpg)

hartytp, 2017-12-07:
@jordens Thanks for the nice sampler photo!

hartytp, 2017-12-18:
@jordens next time you're taking nice photos, can you get one of the VHDCI carrier as well, please?

jordens, 2017-12-19:
I'll try

hartytp, 2017-12-30:
Thanks! They look great.

jordens, 2018-01-08:
@jbqubit @hartytp you seem to be the farthest with actually having a uTCA crate in the lab. Could you post the annotated list of P/Ns that you purchased and the kind of setup that you are aiming for (one sayma or many, with LLRF or without)? Does anyone **have** a LLRF?

hartytp, 2018-01-08:
@jordens Will do, but probably not until later in the week.

If you need info in a hurry I can probably forward you the quote I ordered from.

IIRC, we have two LLRFBPs atm. Not plugged anything into them.

dtcallcock, 2018-09-02:
Wiki moved to meta repository.  See 
https://github.com/sinara-hw/meta/issues/17.

closed 2018-09-02 by dtcallcock