sinara-hw/sinara #209: Sayma AMC - higher connector for LPC JTAG

gkasprow, 2017-05-30:
use 2mm higher J14 connector.  Wurth 621 010 210 21

hartytp, 2017-06-20:
Also, use clock capable pins on FMC if possible (#217).

gkasprow, 2017-06-20:
FMC has clock capable pins (LA00CC, LA01CC,LA17CC,LA18CC, M2C) already connected in right way.

jordens, 2017-06-21:
Are those also the respective EEM CC pins when using an FMC-VHDCI adapter to drive Kasli-style extensions?

gkasprow, 2017-06-24:
@jordens No, they are not.
VHDCI pinout follows FMC one for easier PCB routing.

gkasprow, 2017-06-24:
@hartytp @jordens 
There are 2 ways to fix this issue - modify FMC-VHDCI adapter or VHDCI breakout board.
In second case, Metlino would have to be adapted, but it is not a big deal.
SInce we are going to make corrections to VHDCI breakout, this is the right moment to fix this issue.

hartytp, 2017-06-24:
I'd modify the VHDCI carrier, and keep the FMC-VHDCI adapter as is.

jordens, 2017-06-24:
I would also alter the VHDCI breakout. The good thing is that even with the current boards the odd EEMs are wired sufficiently, see https://github.com/m-labs/sinara/issues/217#issuecomment-310850337

hartytp, 2017-07-12:
@gkasprow If you've ordered the Sayma AMCs, does that mean you've already fixed all these bugs? If so, can we close this issue now?

gkasprow, 2017-07-12:
@hartytp I mean that I assemble remaining 8 boards which are identical with 2 already produced. They still have some bugs that I have to fix once they are produced. The list of bugs is probably not complete - some of them will emerge during tests with JESD204b. Later on I will implement all fixes in new release of Sayma AMC board and then this issues will be completely fixed. You can close it and when necessary we can open it.

hartytp, 2017-07-12:
@gkasprow Sorry, I'd misunderstood you.

If the changes haven't been applied to the designs, we should keep the issue open.

jbqubit, 2017-08-24:
- [ ] Move power supply status LEDS from bottom side of board to top edge of board (like on Sayma_RTM) so that they're visible from top of crate. 

- [ ] Include FMC standoffs and screws in BOM.

abyszuk, 2017-09-26:
There are a few fixes which can applied to the ethernet PHY:
- [ ] connect JTRST_N to GND
- [ ] connect ALOS to GND
- [ ] pull GPO1 to GND
- [ ] CRS/COMMA is connected to MMC twice: (a) through MII_CRS line (b) via PHY_CFG_DDR line. Just pull it low and delete PHY_CFG_DDR wire.
- [ ] GTXCLK should be pulled low

But this just to get it to work reliably in MII mode (max 100 Mbps).
Schematics suggest that it's intended to work in RGMII mode when connected to FPGA. Following changes are required to enable RGMII mode:
- [ ] connect 125 MHz clock source to GTXCLK (maybe it's possible to use one of GPIO for this, if it's configured as clock output)

With current schematics ethernet will not work in RGMII mode, and will work in MII mode only if reconfigured via MDIO (because it starts in (R)GMII mode due to pin straps).

gkasprow, 2017-09-26:
@abyszuk The GTXCLK fix was already applied to some boards. So it works in RGMII.
I didn't test it in MII, thanks for that. I simply shorted GTXCLK with TXCLK. Did you make it running in MII mode?
Is JTRST_N critical? It works without it. In next revision we can change, now would be difficult.
The same with ALOS. For some reason it works in RMII.
COnnection of CRS/COMMA to MMC via CFG was intentional, just to make sure it is properly configured when PHY is connected to FPGA.

jbqubit, 2017-09-26:
Moved Ethernet discussion to #327.

jordens, 2017-09-27:
@gkasprow Is R253 mounted on the three boards we got for Hong Kong or did you remove it because of the RTM configuration conflict?

gkasprow, 2017-09-27:
@jordens I'm not 100% sure. I removed all resistors  that connect lines CCLK, DIN, DONE, INIT_B, PROGRAM_B to configuration pins of Kintex and connected them to ordinary IO.
R253 did not affect this functionality so probably I left it assembled.

jordens, 2017-09-27:
Could you list the differences between the AMCs that we got and the one Florent received? I can't get the second flash to work on any of ours while it worked on Florent's.

gkasprow, 2017-09-27:
Florent board, the one I sent without RTM had Artix FPGA routed to
 the config pins of Kintex FPGA. I removed R176 ... R180 and installed
R200...R204.
Are you sure your boards has both SPI FLASH memory chips soldered?
I don't remember correctly, but after you discovered that chip select is
wrongly connected, I could have decided to not mount second FLASH...
It is just about pin 1 of IC23
[image: Inline images 1]

>

jordens, 2017-09-27:
Maybe it's not soldered. @sbourdeauducq or @whitequark could you have a look?

What about pin 1 of IC23? The second flash is IC24.

gkasprow, 2017-09-27:
I just checked, only a few resistors were changed.
- R257  = DNP - to enable MMC boot without crate
- R208, R205 = DNP. I can't recall why I wanted to make them removed. They should be assembled. R208 is termination resistor for Si5324. R205 is impedance reference for DDR3 controller.
I found that wanted to remove R223 and R226, but by chance clicked on wrong instance of power schematic of Metlino where these resistors are marked R208 and R205. But this does not explain why FLASH is not working.
- R183 = DNP - to enable PUDC
- R243 =  2k0 - to correct 0.9V value
- R242 = 10k - to make 12V LED brighter

gkasprow, 2017-09-27:
@jordens If you publish some bistream with Python library to read/write this FLASH, I can debug it with scope on my board.

sbourdeauducq, 2017-09-28:
Yes, both IC23 and IC24 are there on the boards we have.

gkasprow, 2017-09-28:
So it seems I have to take care of it.
Can you provide me tools to debug it?

jordens, 2017-09-28:
@gkasprow Just our branch of [openocd](https://github.com/m-labs/openocd) and our [proxy bitstream](https://github.com/jordens/bscan_spi_bitstreams), the following script (`openocd -f sayma.cfg`):

sayma.cfg:
```tcl
source [find board/sayma_amc.cfg]
# ftdi_location 5:1
# adapter_khz 5000

init
runtest 1000
puts "JTAG scan chain:"
scan_chain
puts "Flash banks:"
flash banks

pld load 0 bscan_spi_xcku040-sayma.bit
reset halt

flash probe xcu.spi
flash probe xcu.spi1
exit

flash read_bank 0 sayma_f0.bin 0 1024
#flash read_bank 1 sayma_f1.bin 0 1024

flash write_image erase sayma_f0.bin 0
#flash write_image erase $bin $addr
flash verify_bank xcu.spi sayma_f0.bin 0
#flash verify_bank $_FLASHNAME $bin $addr

# xcu_program xcu.tap
exit
```

Gives me
```
Open On-Chip Debugger 0.10.0-00008-g95aa8855-dirty (2017-09-27-17:23)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 5000 kHz
Info : clock speed 5000 kHz
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
JTAG scan chain:
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 xcu.tap                Y     0x13822093 0x*3822093     6 0x01  0x03
Flash banks:
#0 : xcu.spi (jtagspi) at 0x00000000, size 0x00000000, buswidth 0, chipwidth 0
#1 : xcu.spi1 (jtagspi) at 0x00000000, size 0x00000000, buswidth 0, chipwidth 0
loaded file bscan_spi_xcku040-sayma.bit to pld device 0 in 3s 935790us
Info : JTAG tap: xcu.tap tap/device found: 0x13822093 (mfg: 0x049 (Xilinx), part: 0x3822, ver: 0x1)
Info : Found flash device 'micron n25q256 3v' (ID 0x0019ba20)
flash 'jtagspi' found at 0x00000000
Error: Unknown flash device (ID 0x00ffffff)
```

I.e. it only works for the first flash while it used to work for both.

gkasprow, 2017-09-28:
@sbourdeauducq the second FLASH is connected to wrong CS pin. AFAIR to be available by JTAG programmer, the CS would need to be routed to G26 pin (FCS2), not K21 (CSO_B). Maybe on previous board you fixed it by removing R253 and bridging with  R204 pad.

jordens, 2017-09-28:
As mentioned elsewhere, we have a [special proxy bitstream](https://github.com/jordens/bscan_spi_bitstreams/blob/master/xilinx_bscan_spi.py#L323) that deals with that.
We didn't do any changes on the boards. And Florent only changed the clock/jesd power/jesd termination things.

gkasprow, 2017-09-28:
@jordens What about PUDC pin? Maybe your core relies on internal pullups?

jordens, 2017-09-28:
We add bitstream pullups on pins R21 and R22 and drive the rest of QSPI1. I don't see how other pullups would affect the second flash.
There is `PUDC` near SW1:4. Does that mean anything? The schematic has that switch disconnected.
Is the R223/R226 vs R208/R205 mixup something we need to fix? Where are they (remember that we don't have expedition or gerbers/xy)?

jbqubit, 2017-11-16:
- Mark J2 on both front and back side of PCB. 
- Mark pin 1 of front and back side of J2.

jordens, 2018-03-01:
@gkasprow Which boards had the changes you mention in https://github.com/m-labs/sinara/issues/209#issuecomment-332611147 applied? And where are those resistors? Are they here on Florent's board?
[image]
That arrangement does not look like what you described in the comment.

gkasprow, 2018-03-01:
@jordens well, that looks strange. It seems no config pins are connected here at all.
What board is this? Is it the very first board I shipped to Florent?

gkasprow, 2018-03-01:
And are you 100% sure you got it in such state from me? I always use hot air gun but here it looks like somebody was playing with soldering iron :)
Anyway, to gain access to RTM FPGA you need to solder the jumpers R200-R204
[image]

gkasprow, 2018-03-01:
make sure you remove R176 as well, on the other side of the board, close to FLASH chips
[image]

sbourdeauducq, 2018-03-02:
Thanks @gkasprow I'll check this on the boards we have today.

jordens, 2018-03-02:
That was the photo of florents board I shot 7 months ago. Iirc that had only gone through yours and florents hands.

gkasprow, 2018-12-12:
@sbourdeauducq what PUDC level do you want to have on Sayma AMC?
By default pullups were enabled but I noted down that R183 that pulls this line down should be DNP.

gkasprow, 2018-12-12:
The same with POR_override.

sbourdeauducq, 2018-12-12:
I'm traveling at the moment without access to boards, @hartytp or @jbqubit can you have a look?

gkasprow, 2018-12-12:
On RTM you wanted to have pullups enabled.

hartytp, 2019-03-06:
@gkasprow I've lost track, are these issues resolved now?

gkasprow, 2019-03-06:
mechanical issues are still not resolved.

hartytp, 2019-03-11:
@gkasprow where are we with these issues?

gkasprow, 2019-03-11:
There are only mechanical and PCB left

gkasprow, 2019-03-11:
I'll move it to Sayma_AMC repo