sinara-hw/sinara #210: Test AD9910 DRG to extend frequency resolution

jordens, 2017-05-30:
We'd like to use the digital ramp generator in the AD9910 to extend the frequency resolution.

But apparently the DRG accumulator doesn't wrap around (https://ez.analog.com/thread/95552).

The other option I see is to use the DRG to do phase ramps in continuous no-dwell mode with both no-dwell bits set and DRG limits at max/min respectively. To implement a frequency offset fSYSCLK*2^(N-32)/(4*P), one would then choose a power of two as the DRG step size (2^N) and P for the DRG ramp interval counter such that the DRG accumulator exactly reaches the upper limit (2 pi \equiv 0) after 2^(32-N) DRG steps, and correctly resets to zero phase offset. That is then equivalent to wraping around.

Originally from https://github.com/m-labs/sinara/issues/195#issuecomment-304258298

@hartytp could you test this mode of operation (instead of the one suggested in #195)?

hartytp, 2017-05-30:
So, how about:

Clock: 1GHz
FTW: 0x1FFFFFC0 (124.9999851MHz)
DRG Sweep Phase0: 0x0000 (0deg)
DRG Sweep 1Phase1: 0xFFFF (359.995deg)
DRG Rising/Falling Step Size: 0x0100 (1.406deg)
DRG Step interval: 0x0001 (0.0040us)
DRG no dwellhigh/low set

hartytp, 2017-05-31:
[image]

[image]

jordens, 2017-05-31:
That will do a trangular ramp AFAICT. For a sawtooth phase, the negative step size needs to be (1<<32) - P where P is the positive step size.

hartytp, 2017-05-31:
You mean:
[image]

jordens, 2017-05-31:
* DRG limits (0x0B): FF FF FF FF 00 00 00 00
* DRG step (0x0C): FF FF FF 00 00 00 01 00 
* DRG ramp rate (0x0D): 00 01 00 01
* ASF0/POW0/FTW0 : 3F FF 00 00 1F FF FF C0 
* both no-dwell bits, DRG enable, DRG target phase set in CFR2 (0x01)

hartytp, 2017-05-31:
> DRG limits (0x0B): FF FF FF FF 00 00 00 00

AFAICT, in DRG phase ramps one can only set the upper 16 bits of the limits. Thus, we can set FF FF 00 00 00 00 00 00, but not FF FF FF FF 00 00 00 00. (That would only be possible for frequency sweeps).

jordens, 2017-05-31:
Or bigger steps:
* DRG step (0x0C): FF 80 00 00  00 80 00 00 (359.296875 deg, 0.703125 deg)
* DRG ramp rate (0x0D): 80 00 80 00 (0.131072 ms, 0.131072 ms)

jordens, 2017-05-31:
Pretty sure that's only a limitation of that evaluation software, not of the chip.

jordens, 2017-05-31:
> The phase and amplitude step size equations yield the average
step size. Although the step size accumulates with 32-bit precision,
the phase or amplitude destination exhibits only 16 or 14 bits,
respectively. Therefore, at the destination, the actual phase or
amplitude step is the accumulated 32-bit value truncated to 16
or 14 bits, respectively

jordens, 2017-05-31:
Just rail 0x0B to whatever max/min the software allows. That will be a small error but will not invalidate the test.

hartytp, 2017-05-31:
From the DRG overview section in the AD9910 data sheet. Looks like it's a limitation of the DDS itself, not the software.

> The output of the DRG is a 32-bit unsigned data bus that can be
routed to any one of the three DDS signal control parameters, as
controlled by the two digital ramp destination bits in Control
Function Register 2 according to Table 11. The 32-bit output
bus is MSB-aligned with the 32-bit frequency parameter, the
16-bit phase parameter, or the 14-bit amplitude parameter, as
defined by the destination bits. When the destination is phase
or amplitude, the unused LSBs are ignored.

jordens, 2017-05-31:
That datasheet description is consistent with mine above. And you can see from the register dump that the DRG is always 32 bits wide.
The DRG *is* an accumulator that can modulate the POW. And a triangular ramp on the POW is a frequency offset.
What is the result with the parameters above?

hartytp, 2017-06-01:
[image]
[image]

jordens, 2017-06-01:
Using the DRG in double no-dwell mode to add a sawtooth phase ramp [works](https://irclog.whitequark.org/m-labs/2017-06-01).

closed 2017-06-01 by jordens