sinara-hw/sinara #211: Metlino uTCA MCH controlling RFBP should use NAT-MCH-M4 instead of NAT-MCH-Base12-GbE

vdirksen, 2017-06-01:
Comparing the description in [https://github.com/m-labs/sinara/wiki/uTCA](url) mentioning the RFBP backplane behind the AMC backplane and [https://github.com/m-labs/sinara/wiki/Metlino](url) mentioning a single MCH NAT-MCH-Base12-GbE [http://www.nateurope.com/products/NAT-MCH.html](url) I see a small gab for the support of the RFBP. 

Desy has a LLRF backplane and it is controlled from the MCH via a NAT-MCH-RTM-BM-FPGA. For this RTM you need an MCH support Zone3 connector. Example: [http://www.nateurope.com/products/NAT-MCH-PHYS80.html](url). 

For your Metlino project you would only need the base board, which N.A.T. calles NAT-MCH-M4. If needed I can provide user manual.

Summary:
If the Metlino MCH has to support the RFBP backplane, please use NAT-MCH-M4 instead of NAT-MCH-Base12-GbE .

If no RFBP backplane support is needed, both NAT-MCH-Base12-GbE and NAT-MCH-M4 can be used.

vdirksen, 2017-06-01:
small typo corrected

jordens, 2017-06-01:
@vdirksen thanks! I fixed the links in your post.
Details seem to be here: http://www.nateurope.com/manuals/nat_mch_m4_v1x_man_hw.pdf
@gkasprow will probably be the one to triage this.

vdirksen, 2017-06-01:
thanks for fixing the links. Yes the link to the NAT-MCH-M4 manual is correct.

jordens, 2017-06-08:
c.f. #74

gkasprow, 2017-07-05:
@vdirksen @jbqubit  We do not plan to control RTM from MCH - at the moment all management is performed via AMC because uRFB is optional and won't be used in all scenarios. The RTM board has also its own power supply with possibility to use uRFB low noise supply when needed. So for now it doesn't matter which MCH we use. Moroever M4 MCH conflicts with FMC connector we may want to use in future to extend MCH Tongue 4 board to RTM .

jbqubit, 2017-07-05:
> We do not plan to control RTM from MCH
> Moroever M4 MCH conflicts with FMC connector we may want to use in future to extend MCH Tongue 4 board to RTM .

I concur with @gkasprow.

hartytp, 2017-07-12:
Looks like the consensus is that there is nothing to do here -- we should keep on using the NAT-MCH-Base12-GbE for now.

Closing this issue.

closed 2017-07-12 by hartytp