sinara-hw/sinara #214: Urukul: wire AD9910 SYNC circuitry

jordens, 2017-06-06:
Use e.g. CDCLVC1106 to fan out "DDS0:SYNC_OUT or IDC1:SYNC_IN" (solder/capacitor bridge) to DDS0-3:SYNC_IN and IDC1:SYNC_OUT.

dhslichter, 2017-06-06:
I suggest trace length matching for the SYNC_CLK and the IO_UPDATE return (should be on the same IDC, ideally also match lengths on the Urukul PCB to ~500 ps or better, referenced to the appropriate pins at the DDS).  This helps ensure the RTT can be used properly to do deterministic IO_UPDATE.

hartytp, 2017-07-05:
Do we also need to length-match these traces on Kasli/VHDCI carrier/Metlino then? Is that already done?

dhslichter, 2017-07-07:
I did not think we were planning to do DDS synchronization between multiple Urukul boards.  If we are, then this requires some more work beyond what is described above.

Either way, there is no requirement to length match the SYNC_CLK and IO_UPDATE return across multiple Urukul boards -- as long as each board provides a SYNC_CLK and IO_UPDATE return with matched lengths, the actual length is arbitrary and can vary from Urukul to Urukul.

hartytp, 2017-07-07:
Okay, so should we go for length-matching of all LVDS pairs within each IDC?

dhslichter, 2017-07-07:
We need length matching on the IDC all the way back to the FPGA which is emitting the serial commands, at least for the two lines SYNC_CLK and IO_UPDATE_RETURN.  We are only talking about length matching on the order of ~500 ps, so this is not a super stringent requirement.  It should also be possible in principle to calibrate out fixed delays (either measured or calculated) from existing PCB designs (i.e. for FMC to VHDCI breakout, if we are running Urukul from a Metlino and not a Kasli) and include those offsets manually in the calibration.  But just for good design hygiene we should length match on Urukul itself, for sure.

closed 2017-08-28 by gkasprow

jordens, 2017-08-28:
thanks @gkasprow! could you update the schematics so we can give it another review? there were so many changes.

gkasprow, 2017-08-28:
I'm still working on them. I will update ASAP.

hartytp, 2017-08-28:
Agreed, no point doing yet another review while there are still a few simple outstanding issues...