sinara-hw/sinara #216: Sayma RTM - version 1 errata

gkasprow, 2017-06-08:
- J16, J18, J55, J66 i J64  not covered vias under connectors

gkasprow, 2017-07-05:
I observed interesting lockup of analog frontend amplifier.
After power on, the 5V LDO gives -0.9V, negative -5V, VCM = 0V and outputs are at -2.5V.
Below is power sequence
[image]
yellow = VCM, green = -6V, violet = +6V. 
We clearly see that negative power goes too fast.
So I increased sot=ft start capacitance from 3.3nF to 100nF.
It got slightly better but still locks up
[image]
blue = +5V after LDO
So I increased capacitance to 1uF and it works:
[image]

But when one disconnects and quickly connects 12V supply, the SS capacitor doesn't discharge completely and -6V starts too quickly:
[image]

So we will fix it in next revision - negative DC/DC will be switched by PG output of positive one.

hartytp, 2017-07-05:
@gkasprow good catch. Are you referring to IC6 and IC7 on the AFE test board?

Out of curiosity, would it be easier/more robust to fix this by adding Schottky diodes on the LDO outputs to prevent reverse-bias-induced lockup? This is the standard trick I've used on this kind of dual supply board.

gkasprow, 2017-07-05:
@hartytp Yes, IC6 and IC7. I'll add diodes and let you know it this solves that issue.

gkasprow, 2017-07-05:
The BAS281 diode helped. Thanks @hartytp !
I saw such diodes in a few app notes so it seems it is common problem of LDOs.
[image]

hartytp, 2017-07-05:
@gkasprow Great! Glad to hear that fixed it.

I've seen quite a few subtle power sequencing problems in dual supply designs in the past -- particularly when the supplies are loaded asymmetrically. But, IME adding diodes always fixes it, so these days I always add them (diodes are cheaper than thinking/debugging time ;) ).

We should be aware of this as a potential issue for other Sinara boards and add diodes where appropriate.

T

gkasprow, 2017-07-05:
I'm doing power load tests.
And it seem to work. I loaded all supplies with 50% of rated power and for 12V voltage drop on AFE is 
 150mV for +6V and +3.3V and below 50mV for remaining rails. With single 12V fan temperature does not exceed 71 degrees in hottest spots
[image]

Then I loaded +12V and -12V rail with 100% and measured on last AFE:
-11.95V, +11.99
Then I loaded +6V with 100% and measured +5.77V on last AFE. Power consumption was 5.01A@12V
Temperature does not exceed 74deg in hottest spot
[image]
Then I loaded +3.3V with 100% and measured +3.07V on last AFE. Unloaded voltage was 3.0V, 50% loaded was 3.18V

cooling configuration:
[image]

jbqubit, 2017-07-05:
@gkasprow Regarding discussion of "lockup of analog frontend amplifier" related to the LDO IC6 (ADP7102) on AFE test board. Good catch! Related question... It looks like the LDO has a EN/UVLO pin for delayed startup and user-defined hysteresis intended for sequencing multiple power supplies. Is there a reason this feature wasn't used?

hartytp, 2017-07-05:
@gkasprow 

> And it seem to work.

Great!

> I loaded all supplies with 50% of rated power and for 12V voltage drop on AFE is 150mV for +6V and +3.3V and below 50mV for remaining rails. 

> Then I loaded +12V and -12V rail with 100% and measured on last AFE: -11.95V, +11.99

> Then I loaded +6V with 100% and measured +5.77V on last AFE. Power consumption was 5.01A@12V

I'm not sure I follow you here. Do you mean that when you load all AFE test mezzanines at 50% on the +12V supply (100mA per AFE) the +12V rail is at 11.85V, and when you load all AFE mezzanines with 100% on the +12V supply rail (200mA per AFE) the +12V rail is at +11.99V?

hartytp, 2017-07-05:
@jbqubit I don't think that the ADP7102 EN/UVLO will help here. The issue is that the negative supply switches on faster than the positive supply and ends up reverse biasing the positive supply rail (through biasing circuitry/ESD diodes/etc in the ADA4927-1 OpAmp). This causes the regulator to go into lockup.

Using the EN/UVLO to delay the positive regulator's startup further will make the problem worse. 

This is a pretty common issue in this kind of design, and adding diodes to prevent transient reverse bias of the regulators is the standard fix.

hartytp, 2017-07-05:
@gkasprow One other question while I'm thinking about RTM power supplies: at the moment, the +-12V supplies for the AFEs come from a switching converter on Sayma RTM, with some LC filtering. The +12V supply is then used on Allaki to bias the pre-amp (and, less importantly, to supply the OpAmp). Is that correct? If so, do you think that the noise on this rail will be good enough for the pre-amp if we're aiming for "ultra-low" phase/amplitude noise?

We probably talked about this before, so apologies if I've forgotten the what was agreed!

gkasprow, 2017-07-05:
@hartytp I've written that it is below 50mV, which also means close to zero :)
I measured it again with clock mezzanine plugged in:
12V rail
no load: 12.06V
50% load: 12.01V
100% load: 11.97V

6V rail:
no load: 5.95V
50% load: 5.85V
100% load: 5.71V. Had to connect another bench supply because current exceeded 5A

-6V rail:
no load: -6.04
50% load: -6.01
100% load: -5.99

-12V rail:
100% load -11.92. Due to bug on a AFE board the load is constantly connected

3.3V rail:
no load : 3.29
50% load: 3.15V
100% load: 3.03V

hartytp, 2017-07-05:
@gkasprow Nice!

Thanks for clarifying. Glad to hear that's all working!

Out of curiosity, what happens when all rails are at 100% load?

gkasprow, 2017-07-05:
@hartytp 
Yes, both 12V rails come from DC/DC converter.
It's hard to measure this noise in my noisy enviorement, but the scope shows 1-2mV RMS with probe grounded and 1-2mV RMS with probe connected to 12V rail.
So the noise is certainly below the noise floor of my scope.
I don't see any correlation between load and noise value.
Some figures are below. yellow probe is AC coupled to the rail, blue is DC coupled to the same rail. 
The scope noise floor. Probe and  clip connected to SMA ground on AFE board.
[image]

-6V rail under full load
[image]

6V rail with no load
[image]

3.3V rail
[image]

-12V rail with full load
[image]

gkasprow, 2017-07-05:
@hartytp when I load all rails with 100% , I exceed twice the power foreseen for AMC+RTM boards.
It would consume over 120W of power.
I don't want to check it now  - have only 2 boards :)

hartytp, 2017-07-05:
@gkasprow Thanks for the info.

Fluctuations in this voltage rail directly modulate the amp's bias current, so my suspicion is that using the DC/DC output directly will degrade the noise performance somewhat. If we're worried about this/if it turns out to be a problem, we can always add an extra LDO onto Allaki. Maybe @dhslichter has thoughts?

In any case, it looks like the RTM works perfectly -- this is something to think about when we do our design review for Allaki AFE.

> when I load all rails with 100% , I exceed twice the power foreseen for AMC+RTM boards. It would consume over 120W of power. I don't want to check it now - have only 2 boards :)

Fair enough, good point.

jordens, 2017-07-05:
It's not terribly sensitive. At the low frequency (for bias current/power supply fluctuations) end, you get about 0.005 dB more gain for 1% more bias current.

hartytp, 2017-07-05:
@jordens True. Depends on what noise floor you're trying to reach though. Anyway, this isn't something I'm particularly worried about as it's easy to test and easy to fix if we do decide it's problematic.

jordens, 2017-07-05:
That's assuming it's linear.... the delta is 0.12 dB for a 20% bias step.

jbqubit, 2017-07-05:
@hartytp said
> adding diodes to prevent transient reverse bias of the regulators is the standard fix.

Thanks for the explanation. Inclusion of diodes seems like a general solution. I've updated the wiki [PCBReviewChecklist](https://github.com/m-labs/sinara/wiki/PCBReviewChecklist) to reflect this practice.

dhslichter, 2017-07-07:
I thought we had an LDO on Allaki for the output amplifier supply -- if not, it would be good to add for the next iteration.  The gain of the output amp is basically linear in supply current, so fractional voltage noise is turned directly into fractional gain noise.  There is a choke and a capacitor (or at least, there should be) on the supply line which provides some filtering at higher frequencies, but I think an LDO is definitely a good idea here.

jbqubit, 2017-07-08:
@dhslichter I added the LDO to Allaki schematic review check list. #110

gkasprow, 2017-07-10:
SPI works, I had a bug in my Vivado code. For some strange reason it let me assign 2 conflicting values to same IO pin. And I didn't notice warning. But it works now - i read chip ID. I also loaded example configuration that generates 1.4GHz output.
Here is FORTH code used to init the chip 
`: config_hmc_830
0 0 $20 spi_hmc830 .
0 1 $2 spi_hmc830 .
0 2 $2 spi_hmc830 .
0 5 $1628 spi_hmc830 .
0 5 $60a0 spi_hmc830 .
0 5 $e110 spi_hmc830 .
0 5 $2818 spi_hmc830 .
0 5 $0 spi_hmc830 .
0 6 $303ca spi_hmc830 .
0 7 $14d spi_hmc830 .
0 8 $c1beff spi_hmc830 .
0 9 $153fff spi_hmc830 .
0 a $2046 spi_hmc830 .
0 b $7c061 spi_hmc830 .
0 f $81 spi_hmc830 .
;`

And I get RF at the output:
[image]

hartytp, 2017-07-10:
@gkasprow Great!

gkasprow, 2017-07-11:
I'm fighting with HMC7043 SPI.
 Don't get any response, the chip does not react to ID readout nor setting of GPIO.
I even generated example configuration using ADI tool - still no response. The chip keeps the data line Hi-Z all the time. I even configured GPIO as SPI data output but still no response.
I observe all signals at the chip leads, power supply, reset are OK.
Maybe it needs RF signal for SPI operation - this I will check tomorrow.
Any ideas?

hartytp, 2017-07-11:
Hmm, not sure. Some ideas:
- Is RESET LOW?
- Have you tried grounding pins 14 and 24? The data sheet says these "must be tied to ground". On the EVAL board, it's labelled TEST_MODE_SEL and TEST_MODE

gkasprow, 2017-07-11:
@hartytp good catch!
I will connect these pins to GND and check tomorrow.
RESET is low.
Thanks!

gkasprow, 2017-07-12:
@hartytp 
It works!
[image]

hartytp, 2017-07-12:
The joys of undocumented diagnostic modes...

gkasprow, 2017-07-12:
Another bizzare thing
When you configure HMC830 first, it works. But when you start config from HMC7043, the HMC830 does not. And it is correct behaviour. HMC830 has 2 modes of operation and as datasheet says:
>On power up both types of modes are active and listening.
A decision to select the desired SPI protocol is made on the first occurrence of SEN or SCLK following a 
hard reset, after which the protocol is fixed and only changeable by cycling the power OFF and ON.
a. If a rising edge on SEN is detected first HMC Mode is selected.
b. If a rising edge on SCLK is detected first Open mode is selected.

So one must make sure that HMC830 SEN goes high first and in our case where clock line is shared between both HMC chips it is critical.

hartytp, 2017-07-12:
uuurgh ugly. Good spot Greg.

Do we have appropriate pull-downs to ensure that these lines are in well-defined states during power up?

gkasprow, 2017-07-12:
I managed to connect 100MHz to the input, use PLL to multiply it to 1.5G and then divide it with HMC7043 to 75MHz.
So the clock chain works.

hartytp, 2017-07-12:
Whoo!

sbourdeauducq, 2017-07-12:
@gkasprow have you sent a RTM board to @enjoy-digital ?

gkasprow, 2017-07-12:
@hartytp at the moment PLL wakes up reliably, but I will add resistors in next revision.

hartytp, 2017-07-12:
@gkasprow Thanks. Normally I wouldn't bother, but if the PLL can only be reset by a power cycle it's worth being a bit more careful.

gkasprow, 2017-07-12:
@sbourdeauducq not yet. But Florent has anyway enough to play with. I plan to send it to you tomorrow or on monday. There are just a few minor things to test

sbourdeauducq, 2017-07-12:
> Florent has anyway enough to play with

He does not; this is blocking the next item in our schedule which is the AMC-RTM FPGA wishbone bridge, which in turns enables clocking, JESD204, etc.

enjoy-digital, 2017-07-12:
@gkasprow: yes that would better if you can send it tomorrow so that I have enough time to validate the wishbone bridge and basic clocking/jesd before the end of the month, thanks.

gkasprow, 2017-07-12:
@sbourdeauducq @enjoy-digital I understand it and will do my best to send this week. There are two issues so far: 
- with I2C chip on AFE that doesn't want to work
- once I switch on supply on Artix FPGA, the 1.8V supply on Kintex sometimes goes off. I have to debug it so you don't have troubles.
ADC and DAC SPI work  fine, at least I read correct ID.

gkasprow, 2017-07-12:
@enjoy-digital please correct your constraint file: DAC2 SDIO and SCLK are swapped. Easy to correct by swapping FPGA lines: J18 and J15

gkasprow, 2017-07-12:
The only thing that still does not work (apart from JESD204B) is one I2C chip on mezzanines - TCA9539RTWR.
All other I2C circuits in the chain work fine.
I verified with scope on TCA pins - SDA, SCL signals are OK, voltage levels (3.3V), frequency, pinout, supply.
And the chip does not ACK. I tried all 4 possible addresses.
The same happens on all AFE boards. RESET is also OK.
Any idea? Maybe it locks somehow with illegal I2C sequence at the startup... I'll try to use reset pin.

jbqubit, 2017-07-12:
That's great news @gkasprow! 

Are the Sayma_RTM modification simple enough that the mods can be applied to the remaining 8 boards? If so when could they be stuffed? Trying to figure out how to sequence things as NACTI approaches...

gkasprow, 2017-07-12:
@jbqubit Yes, they are simple - just need 15 mins of work per board to apply all fixes.
AMC boards are being assembled right now, should get them this week.
The only thing that worries me with RTM is JESD204B which is the only not tested functionality, so if possible, I'd like to postpone RTM assembly to the moment we make sure it somehow works.  So I'd like to focus on ADC/DAC boards evaluation, in the meantime MLabs makes ADC/DAC running and then within a few days we can have RTMs assembled. In the meantime we will produce Allaki proto boards.

hartytp, 2017-07-12:
@gkasprow Sounds like a good plan!

hartytp, 2017-07-13:
> So I'd like to focus on ADC/DAC boards evaluation, 

@gkasprow In terms of scheduling, do you think we could finish Kasli + Novogorny before testing the ADC + DAC? It would be good to send them off to manufacture asap so we don't end up waiting for the boards to arrive later on.

sbourdeauducq, 2017-07-13:
@gkasprow @jbqubit We don't have any JESD ADC code nor funding to develop it. I suppose that for testing this part, some hack based on Xilinx IP is the way to go in the short term (e.g. program the ADC and clocking from ARTIQ, then reload the Ultrascale FPGA with the Xilinx code that checks the PRBS and other things).

What do you mean by "ADC/DAC boards evaluation"?

gkasprow, 2017-07-13:
@sbourdeauducq by evaluaition I mean testing existing one and them redesign of 3U ADC board with new ADC.
Did you manage to program this ADC? It has hundreds of registers. One can use ADI tool to generate contents but there are so many options. So far I used very simple JESD ADCs that didn't need configuration.

gkasprow, 2017-07-13:
@hartytp we have 3U ADC and DAC contracts at WUT already signed and we have to finish and send them on time. Kasli can be easily replaced by Sayma AMC or KC705 for the moment.

hartytp, 2017-07-13:
> @hartytp we have 3U ADC and DAC contracts at WUT already signed and we have to finish and send them on time. Kasli can be easily replaced by Sayma AMC or KC705 for the moment.

Okay, that makes sense. Stick to your plan.

jordens, 2017-07-13:
@gkasprow Yes those JESD DACs and ADCs have hundreds of registers. It is complicated. [This](https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard/ad9154.rs) is what we do for the DACs.

gkasprow, 2017-07-13:
ADI tool for HMC7034 chip generates Python code that can be used for
configuration directly. And I used it in that way to config that chip.
SImilar tool for ADC (called SPI controller) also provides register content
but in plain text file.

hartytp, 2017-07-13:
@jordens Do you have a tool to parse that kind of thing into python?

gkasprow, 2017-07-13:
@hartytp I use programmers notepad and regular expressions :) In this way I
parse constraint files or config register contents. It takes a few minutes
of work.

jordens, 2017-07-13:
@hartytp The other way around: https://github.com/m-labs/artiq-chip-reg

gkasprow, 2017-07-13:
here is example of extreme soldering. Missing 49R9 0201 resistors were added at the DAC outputs.
[image]

hartytp, 2017-07-13:
Impressive.

jbqubit, 2017-07-13:
Wow! Nice. I consider myself school'd. :)   cf https://github.com/m-labs/sinara/issues/90

gkasprow, 2017-08-04:
@sbourdeauducq @enjoy-digital This is what I observe at the DAC clock and ref input:
[image]

[image]

gkasprow, 2017-08-04:
Below is output of the configuration script

pawel@Stanowisko-ARTIQ:~/artiq/sayma_test/test/sayma_both$ python3 test_identifier.py 
AMC fpga_id:Sayma AMC JESD Test Design
RTM fpga_id:Sayma RTM JESD Test Design
pawel@Stanowisko-ARTIQ:~/artiq/sayma_test/test/sayma_both$ python3 test_dac0.py 
dac0 configuration
dac0 present: True
LANEFIFOERR: 1, SERPLLLOCK: 0, SERPLLLOST: 0, DACPLLLOCK: 0, DACPLLLOST: 0
PRBS0: 0, PRBS1: 0, PRBS2: 0, PRBS3: 0
SYNC_TRIP0: 0, SYNC_WLIM0: 0, SYNC_ROTATE0: 0, SYNC_LOCK0: 0, NCO_ALIGN0: 0, BLNKDONE0: 0, PDPERR0: 0
SYNC_TRIP1: 0, SYNC_WLIM1: 0, SYNC_ROTATE1: 0, SYNC_LOCK1: 0, NCO_ALIGN1: 0, BLNKDONE1: 0, PDPERR1: 0
ERR_INTSUPP: 0, ERR_SUBCLASS: 0, ERR_KUNSUPP: 0, ERR_JESDBAD: 0, ERR_WINLIMIT: 0, ERR_DLYOVER: 0
DACPLL_LOCK: 0, VCO_CAL_PROGRESS: 0, CP_CAL_VALID: 0, CP_OVERRANGE_L: 0, CP_OVERRANGE_H: 0
PLL_LOCK_RB: 0, CURRENTS_READY_RB: 1, VCO_CAL_IN_PROGRESS_RB: 0, PLL_CAL_VALID_RB: 1, PLL_OVERRANGE_L_RB: 1, PLL_OVERRANGE_H_RB: 0
CODEGRPSYNC: 0x00
FRAMESYNC: 0x00
GOODCHECKSUM: 0x00
INITIALLANESYNC: 0x00
SYNC_LASTERR: 0x01f3
SYNC_LASTOVER: 0, SYNC_LASTUNDER: 1
SYNC_TRIP: 1, SYNC_WLIM: 0, SYNC_ROTATE: 1, SYNC_LOCK: 1, SYNC_BUSY: 0
LANE_FIFO_FULL: 0x01
LANE_FIFO_EMPTY: 0x02
DID_REG: 0x00
BID_REG: 0x00
SCR_L_REG: 0x00
F_REG: 0x00
K_REG: 0x00
M_REG: 0x00
CS_N_REG: 0x00
NP_REG: 0x00
S_REG: 0x00
HD_CF_REG: 0x00
RES1_REG: 0x00
RES2_REG: 0x00
LIDx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
CHECKSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
COMPSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
BADDISPARITY: 0x00
NITDISPARITY: 0x00
UNEXPECTEDCONTROL: 0x00
LANEFIFOERR: 1, SERPLLLOCK: 0, SERPLLLOST: 0, DACPLLLOCK: 0, DACPLLLOST: 0
PRBS0: 0, PRBS1: 0, PRBS2: 0, PRBS3: 0
SYNC_TRIP0: 0, SYNC_WLIM0: 0, SYNC_ROTATE0: 0, SYNC_LOCK0: 0, NCO_ALIGN0: 0, BLNKDONE0: 0, PDPERR0: 0
SYNC_TRIP1: 0, SYNC_WLIM1: 0, SYNC_ROTATE1: 0, SYNC_LOCK1: 0, NCO_ALIGN1: 0, BLNKDONE1: 0, PDPERR1: 0
ERR_INTSUPP: 0, ERR_SUBCLASS: 0, ERR_KUNSUPP: 0, ERR_JESDBAD: 0, ERR_WINLIMIT: 0, ERR_DLYOVER: 0
DACPLL_LOCK: 0, VCO_CAL_PROGRESS: 0, CP_CAL_VALID: 0, CP_OVERRANGE_L: 0, CP_OVERRANGE_H: 0
PLL_LOCK_RB: 0, CURRENTS_READY_RB: 1, VCO_CAL_IN_PROGRESS_RB: 0, PLL_CAL_VALID_RB: 1, PLL_OVERRANGE_L_RB: 1, PLL_OVERRANGE_H_RB: 0
CODEGRPSYNC: 0x00
FRAMESYNC: 0x00
GOODCHECKSUM: 0x00
INITIALLANESYNC: 0x00
SYNC_LASTERR: 0x01f3
SYNC_LASTOVER: 0, SYNC_LASTUNDER: 1
SYNC_TRIP: 1, SYNC_WLIM: 0, SYNC_ROTATE: 1, SYNC_LOCK: 1, SYNC_BUSY: 0
LANE_FIFO_FULL: 0x08
LANE_FIFO_EMPTY: 0x00
DID_REG: 0x00
BID_REG: 0x00
SCR_L_REG: 0x00
F_REG: 0x00
K_REG: 0x00
M_REG: 0x00
CS_N_REG: 0x00
NP_REG: 0x00
S_REG: 0x00
HD_CF_REG: 0x00
RES1_REG: 0x00
RES2_REG: 0x00
LIDx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
CHECKSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
COMPSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
BADDISPARITY: 0x00
NITDISPARITY: 0x00
UNEXPECTEDCONTROL: 0x00
pawel@Stanowisko-ARTIQ:~/artiq/sayma_test/test/sayma_both$ python3 test_dac1.py 
dac1 configuration
dac1 present: True
LANEFIFOERR: 1, SERPLLLOCK: 0, SERPLLLOST: 0, DACPLLLOCK: 0, DACPLLLOST: 0
PRBS0: 0, PRBS1: 0, PRBS2: 0, PRBS3: 0
SYNC_TRIP0: 0, SYNC_WLIM0: 0, SYNC_ROTATE0: 0, SYNC_LOCK0: 0, NCO_ALIGN0: 0, BLNKDONE0: 0, PDPERR0: 0
SYNC_TRIP1: 0, SYNC_WLIM1: 0, SYNC_ROTATE1: 0, SYNC_LOCK1: 0, NCO_ALIGN1: 0, BLNKDONE1: 0, PDPERR1: 0
ERR_INTSUPP: 0, ERR_SUBCLASS: 0, ERR_KUNSUPP: 0, ERR_JESDBAD: 0, ERR_WINLIMIT: 0, ERR_DLYOVER: 0
DACPLL_LOCK: 0, VCO_CAL_PROGRESS: 0, CP_CAL_VALID: 0, CP_OVERRANGE_L: 0, CP_OVERRANGE_H: 0
PLL_LOCK_RB: 0, CURRENTS_READY_RB: 0, VCO_CAL_IN_PROGRESS_RB: 0, PLL_CAL_VALID_RB: 0, PLL_OVERRANGE_L_RB: 0, PLL_OVERRANGE_H_RB: 0
CODEGRPSYNC: 0x00
FRAMESYNC: 0x00
GOODCHECKSUM: 0x00
INITIALLANESYNC: 0x00
SYNC_LASTERR: 0x01f1
SYNC_LASTOVER: 0, SYNC_LASTUNDER: 1
SYNC_TRIP: 1, SYNC_WLIM: 0, SYNC_ROTATE: 1, SYNC_LOCK: 1, SYNC_BUSY: 0
LANE_FIFO_FULL: 0x00
LANE_FIFO_EMPTY: 0x00
DID_REG: 0x00
BID_REG: 0x00
SCR_L_REG: 0x00
F_REG: 0x00
K_REG: 0x00
M_REG: 0x00
CS_N_REG: 0x00
NP_REG: 0x00
S_REG: 0x00
HD_CF_REG: 0x00
RES1_REG: 0x00
RES2_REG: 0x00
LIDx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
CHECKSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
COMPSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
BADDISPARITY: 0x00
NITDISPARITY: 0x00
UNEXPECTEDCONTROL: 0x00
LANEFIFOERR: 1, SERPLLLOCK: 0, SERPLLLOST: 0, DACPLLLOCK: 0, DACPLLLOST: 0
PRBS0: 0, PRBS1: 0, PRBS2: 0, PRBS3: 0
SYNC_TRIP0: 0, SYNC_WLIM0: 0, SYNC_ROTATE0: 0, SYNC_LOCK0: 0, NCO_ALIGN0: 0, BLNKDONE0: 0, PDPERR0: 0
SYNC_TRIP1: 0, SYNC_WLIM1: 0, SYNC_ROTATE1: 0, SYNC_LOCK1: 0, NCO_ALIGN1: 0, BLNKDONE1: 0, PDPERR1: 0
ERR_INTSUPP: 0, ERR_SUBCLASS: 0, ERR_KUNSUPP: 0, ERR_JESDBAD: 0, ERR_WINLIMIT: 0, ERR_DLYOVER: 0
DACPLL_LOCK: 0, VCO_CAL_PROGRESS: 0, CP_CAL_VALID: 0, CP_OVERRANGE_L: 0, CP_OVERRANGE_H: 0
PLL_LOCK_RB: 0, CURRENTS_READY_RB: 0, VCO_CAL_IN_PROGRESS_RB: 0, PLL_CAL_VALID_RB: 0, PLL_OVERRANGE_L_RB: 0, PLL_OVERRANGE_H_RB: 0
CODEGRPSYNC: 0x00
FRAMESYNC: 0x00
GOODCHECKSUM: 0x00
INITIALLANESYNC: 0x00
SYNC_LASTERR: 0x01f1
SYNC_LASTOVER: 0, SYNC_LASTUNDER: 1
SYNC_TRIP: 1, SYNC_WLIM: 0, SYNC_ROTATE: 1, SYNC_LOCK: 1, SYNC_BUSY: 0
LANE_FIFO_FULL: 0x00
LANE_FIFO_EMPTY: 0x00
DID_REG: 0x00
BID_REG: 0x00
SCR_L_REG: 0x00
F_REG: 0x00
K_REG: 0x00
M_REG: 0x00
CS_N_REG: 0x00
NP_REG: 0x00
S_REG: 0x00
HD_CF_REG: 0x00
RES1_REG: 0x00
RES2_REG: 0x00
LIDx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
CHECKSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
COMPSUMx_REG: 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
BADDISPARITY: 0x00
NITDISPARITY: 0x00
UNEXPECTEDCONTROL: 0x00
pawel@Stanowisko-ARTIQ:~/artiq/sayma_test/test/sayma_both$

enjoy-digital, 2017-08-05:
@gkasprow: thanks for the captures and tests. In case you want to modify the configuration of the PLL, the configuration script is here:
https://github.com/enjoy-digital/sayma_test/blob/master/test/sayma_both/clocking_config.py

[225 more replies]