sinara-hw/sinara #217: allocate clock capable pins on IDC connectors

jordens, 2017-06-12:
For schemes like source synchronous data transfers, used e.g. in [fast Novogorny](https://github.com/m-labs/sinara/wiki/Novogorny#fast-novogorny) there should be a clock-capable pin on the IDC connectors.
Let's ensure that pin 0 (also SPI clock) is on a CC pin. And if there are more CC pins available, also put pin 3 on a CC pin.

hartytp, 2017-06-20:
If possible, please can we make sure that this is done for: all Kasli IDCs; the FMCs on Sayma AMC (for when they're used with FMC->VHDCI adapter + VHDCI_CARRIER); and, the VHDCIs on Meltino?

jordens, 2017-06-20:
That would be wise. Yes.

gkasprow, 2017-06-20:
@hartytp  Sure, this we can still do on Metlino and Kasli.

jordens, 2017-06-24:
On [VHDCI from FMC](http://www.ohwr.org/attachments/download/4489/FMC_DIO_32LVDS.PDF) AFAICT VHDCI pairs 1/35, 2/36, 18/54, 19/55 are CC, that's the usual FMC CCs on LA 1,2,18,18.

gkasprow, 2017-07-15:
@jordens The rev 1.0 signal assignment is following:
 FMC - VHDCI - EEM no - EEM signal
LA00CC - L1/L35 - EEM1:1
LA01CC - L2/L36 - EEM1:2
LA17CC - L20/L54- EEM3:2
LA18CC - L21/L55- EEM3:3
So in rev 1.1 I replaced pins 32/36 with 9/43 
and L19/53 with 20/54
and 21/55 with 27/61

closed 2017-08-20 by hartytp