sinara-hw/sinara #221: VHDCI carrier version 1 errata

gkasprow, 2017-06-13:
- mark connectors with names of LVDS signals
- add front panel LED
- R210, R187 = DNP

gkasprow, 2017-07-16:
All issues fixed in rev 1.1

closed 2017-07-16 by gkasprow

jordens, 2017-07-16:
Hmm. isn't tying the shield on the metlino/fmc side **only** the better choice for ground loops?

gkasprow, 2017-07-16:
@jordens we have only single wire for gnd signal which is not sufficient for fast LVDS signals. Anyway, we can use connectors with shield connection or without it.

gkasprow, 2017-07-16:
@hartytp please have a look ad BNC, SMA, RJ45 and VHDCI board documentation.
If there are not comments, I start production.

jbqubit, 2017-07-16:
Documentation discussion in #235.

hartytp, 2017-07-17:
@gkasprow All look good to me! Please start production.

Thank you for doing that.

FWIW, the RJ45 schematic still has a large red comment saying "Add TVS!!". AFAICT, the TVS has been added, so this comment can now be removed.

hartytp, 2017-07-18:
@gkasprow When you're ready, could you send me a quote for these boards and also confirm the expected delivery date, please?

On a related note, what's the status with the RF PA prototype? Did it go to manufacture yet?

hartytp, 2017-07-19:
@gkasprow Can we order an extra VHDCI_carrier for M-Labs, please?

Total quantity of boards for us is thus:

VHDCI carrier: 6
BNC: 15
RJ45: 6

jordens, 2017-07-19:
@gkasprow oh. are we already tallying numbers for the first big batch for those EEM boards or will there be another prototype round? if yes, we'd like a few days to rally the troops to maximize batch size.

hartytp, 2017-07-19:
@jordens My understanding is that the following 3U boards have been fairly well tested, and all known issues fixed:

VHDCI_CARRIER
BNC_DIO
SMA_DIO
RJ45_DIO

The plan is now to produce a relatively small batch (~20 of each) which people can start using in their labs. Once these have been used in a few labs for some time (few months, say), we'll see if any other issues need fixing before moving onto a larger production round.

In the mean time, we need quite a few of these in our labs asap to run our experiments (we're getting to the stage where not having these is causing us problems).

npisenti, 2017-07-19:
If it's going to be of order few months + fab time, we'd also like to get quoted for:

* 6x BNC_DIO
* 1x DAC EEM (once debugged/tested)
* 1x ADC EEM (once debugged/tested)

jordens, 2017-07-19:
@hartytp that notion of "we need quite a few of these asap" is common. we would like to have kasli/novogorny/urukul plus some of those boards within the next couple months as well. from what i have seen so far planning for about a month turn around sounds reasonable. also as you say, these boards have been fairly well tested and all known issues fixed. why do a "small batch" if there are others wanting them as well? either we expect remaining bugs or we expect not to encounter them.
not having hardware when it is needed (for whichever reason) is a planning issue. that's exactly what i would like to preempt.

hartytp, 2017-07-19:
> either we expect remaining bugs or we expect not to encounter them.

Right now, I do think that there is a decent chance of encountering further bugs or things we need to change. e.g. Greg found an issue on the BNC DIO EEM with the DC to DC converters failing after a while, which required a change of part + footprint. These kinds of issues sometimes only show up after months of testing. Moreover, there is still the possibility that as the other EEM designs progress, we'll realise that we need to make modifications to these boards (I'm thinking of things like your spot that VHDCI_CARRIER needs to route the cc pins properly).

> why do a "small batch" if there are others wanting them as well? 

In general, I'd welcome others buying hardware asap as well, as it increases the number of testers. However:

1. People have to realise that this is v2 prototype hardware rather than v1 production hardware. 
2. I need to get this order placed fairly soon (ideally, few days, week or at the latest). So, if people know what they want and can tell Greg now, that's great. But, I don't want to get into the situation of people asking for quotes and then taking ages to decide what they actually want.

jordens, 2017-07-19:
Most groups won't wait months until those bugs have been identified and another month until the next batch is done. They are in a very similar situation to yours and will need to just accept the risk.
Pending information how @gkasprow wants to rig this and how the bill is split, I would propose that the first one (i.e. you) just set a deadline for batch size freeze and then everybody plan around that.

hartytp, 2017-07-19:
@gkasprow Can you confirm a few things:

- Have you already placed the order? Or, is there still time for other people to add to this order?
- What is the cost of these boards?
- What is the lead time/manufacturing time?
- What is the ordering process? Does everyone have to give Technosystem a PO/payment before the manufacture can begin? Or, can people confirm quantities with you and sort payment out later? We can now place orders with Technosystem within 24 hours, but the first order we placed took us a couple of weeks as they had to be added to Oxford's "suppliers database", which is a pretty tortuous process. I want to make sure we don't end up with large delays due to having to get multiple Universities' finance departments to all do the paperwork before the manufacturing can start.

Other than that, I'd be happy to set a batch size freeze on Monday 9am GMT.

hartytp, 2017-07-19:
@gkasprow One other thing I forgot to ask: will these boards come with proper, fully labelled front-panels?

- It'd be nice to have the name of the PCB on each panel BNC DIO, RJ45 DIO, SMA DIO, VHDCI EEM Carrier
- It'd be nice to have channel numbers labelled
- If there is room, it would be nice to have the function of the LEDs labelled (e.g. "red=input", "termination" or wherever is appropriate)
- Ideally, but not necessary, it'd be nice to have an ARTIQ logo somewhere if there is space

jbqubit, 2017-07-19:
This is waaaay off the topic of "VHDCI breakout version 1 errata". I created new conversation #237.