sinara-hw/sinara #222: SMA DIO version 1 errata

gkasprow, 2017-06-13:
- use 500mA DC/DC
- add GND/I2C testpoints

gkasprow, 2017-07-14:
All issues except for RA IDC fixed in rev 1.1

closed 2017-07-14 by gkasprow

gkasprow, 2017-07-15:
All issues fixed in rev 1.1

sbourdeauducq, 2018-03-29:
@gkasprow How serious is the DC/DC converter problem on SMA DIO v1.0? Are there serious problems with v1.0?

hartytp, 2018-03-29:
Related to this, I've been wondering recently if it's worth creating a version of the SMA/BNC DIO boards that doesn't have isolation. My thinking here is:
1. If I remember the history correctly, the decision to add isolation there was made early on when we were primarily thinking about systems with a single Metlino as master and no Kasli. As a result, it seemed important to isolate the DIO grounds. Now that Kasli is available and relatively cheap, one can achieve isolation by adding Kasli DRTIO slaves (effectively, use 1 Kasli per ground point). So, the need to isolate the DIO modules seems greatly reduced. I'd argue that the isolated DIO modules are still potentially useful and definitely worth stocking, but may not be the best option for most use cases.
2. The non-isolated version is likely to be a bit cheaper, simpler and more robust (no horrible grounding issues to get right, fewer SMPSs etc)
3. Currently, the jitter between a pair of TTLs is about 1.5ns pk-pk, so something like 750ps pk-pk jitter per TTL. AFAICT, this is dominated by the isolators. This jitter is about acceptable, but a factor of few less would be nice (ARTIQ's timing resolution is about 1ns). This could be improved a bit by changing for a lower jitter isolator, but the isolator we have isn't bad compared with other available options.

Thoughts?

hartytp, 2018-03-29:
> @sbourdeauducq How serious is the DC/DC converter problem on SMA DIO v1.0? Are there serious problems with v1.0?

Well, if you use a floated board with switching supplies then it's always prone to charging up. One generally needs to ensure that the PCB and load grounds are connected somehow. The *usual* way that would happen in this kind of system IME is that the main PCB ground would be connected to an earthed chassis via its front panel. The load would also have a single ground bond somewhere. If you do that you'll be fine. If not, you may see some issues.

I believe we placed 100k resistors between the output ground and the main PCB ground on these DIO boards. The idea of that was to limit the potential difference between the two grounds and avoid this kind of issues. In hindsight, it seems that the ground currents are larger than anticipated, so that resistance is too high. But, if you're worried about this/don't need isolation and want the safest solution, you can replace that 100k resistor with a short to un-isolate the load.

sbourdeauducq, 2018-03-29:
> In hindsight, it seems that the ground currents are larger than anticipated, so that resistance is too high.

Where are those ground currents coming from?

hartytp, 2018-03-29:
Switching supplies + capacitance + rectification. It's a standard issue with isolated SMPSs, which we did anticipate to a certain extent (hence the 100k resistors) although clearly we didn't put quite enough time/thought into it

hartytp, 2018-03-29:
PS if you're interested, there's a really good analysis of this in the Ott EMC book.

gkasprow, 2018-03-29:
There are ways to do ideal isolation. One needs to install 2 LiOn batteres and 2 mechanical relays that periodically toggles between them. When one is working, another one is charged, then they swap :)
Another one is solar array, but this works only with low power devices.
Yet another one is resonant DC/DC converter. Resonant converters have very high efficiency even with poor inductive coupling between primary and secondary windings. So one can separate them even with 1cm air gap and still get 90% of efficiency . I've build such one recently for HV supply where 20kV isolation was needed and very low capacitance between primary and secondary side.

gkasprow, 2018-03-29:
@hartytp there are 100k resistors in new revisions.

sbourdeauducq, 2018-03-30:
> Resonant converters have very high efficiency even with poor inductive coupling between primary and secondary windings. So one can separate them even with 1cm air gap

Don't resonant converters also use a full-bridge rectifier that would reduce the ground current significantly, compared to a flyback?

dtcallcock, 2018-03-30:
> Currently, the jitter between a pair of TTLs is about 1.5ns pk-pk, so something like 750ps pk-pk jitter per TTL. AFAICT, this is dominated by the isolators. This jitter is about acceptable, but a [...]

I agree with these concerns of @hartytp.  In our current setup we have a few places where we have to meet quite tight setup conditions for digital triggers and a few 750ps added up would start getting a bit hairy.  If a factor of 2 or 3 improvement is available with a different chip that would be ideal.  On the other hand one could just avoid using this board in the few places where it matters (trivial if a non-isolated version is produced).

gkasprow, 2018-04-02:
How many non-isolated DIO boards would be needed?
Maybe it would be easier to add a jumper that would short the DC/DC converter grounds on primary and the secondary side?
If isolator jitter is an issue, dedicated board design would be needed.
We can also think about assembly variant that simply shorts isolator inputs with outputs.

hartytp, 2018-04-02:
> If isolator jitter is an issue, dedicated board design would be needed.

For us, I think the jitter is mildly annoying, but not a major issue, as we aren't likely to use the DIO board for any critical signals. For very low jitter inputs (e.g. APD sampling with <<ns resolution) we'll use a TDC like stamper. For DDS sync lines etc, we'll use EEMs/AMCs which connect the SYNC directly (well, via a CPLD etc) to the FPGA.

> Maybe it would be easier to add a jumper that would short the DC/DC converter grounds on primary and the secondary side?

We already have the 100k resistors for that; they can be replaced with 0R to connect the grounds so long as users are happy getting out a soldering iron.

> We can also think about assembly variant that simply shorts isolator inputs with outputs.

If you can add room for 0R jumpers without making the layout too much of a mess, that might be a good idea.

hartytp, 2018-04-02:
Hmmm...actually, it's not clear to me that removing the isolator would make a huge difference to the jitter:
- 1.5ns pk-pk measured between two outputs. So, assume 750ps pk-pk jitter for a single output
- Isolator is SI8651BB-B-IS1 datahseet specifies 350ps peak-peak jitter
- So, that's less than half the measured jitter. I assume the rest comes from the LVDS interface/FPGA/etc so the improvements here with changing the isolator may not be that large
- However, if we want to squeeze the last bit of jitter out, we could also use one of the NVE isolators with only 100ps of (peak-peak??) jitter, but I'm not sure it's worth bothering without understanding the other sources of jitter a bit better.

gkasprow, 2018-04-02:
To limit the output jitter , one can add resync DFF at such output and feed 100MHz clock. This would have to be dedicated low jitter DIO with dedicated outputs. In this way we get rid of most of the jitter components but limit the timing resolution to only 10ns.

jbqubit, 2018-04-02:
It seems unnecessary at this point to create a new PCB variant or factory-stuffing option. Soldering 0R is a fine solution. Let's add pads for this to the next version. 

> I'm not sure it's worth bothering without understanding the other sources of jitter a bit better.

Agreed. Better if this is championed (and funded by) a user whose application depends critically on jitter. As @gkasprow points some approaches might entail significant design changes.