sinara-hw/sinara #224: Sayma hardware testing checklist

jbqubit, 2017-06-21:
# Key Dates
__5/29__:  2 of Sayma_RTM delivered to WUT
__6/8__: 2 of Sayma_RTM delivered to WUT
__6/21__: 1 each of Sayma_RTM & Sayma_AMC shipped to M-Labs (Florent)

# AMC
- [ ] power supplies, sequencing 
- [ ] AMC+RTM power dissipation under load in chassis 
- [ ] oscillators 
- [x] MMC 
- [x] MMC update over USB 
- [x] FPGA UARTs 
- [x] FPGA programming 
- [ ] JTAG over USB
- [x] LEDs
- [x] SMA outputs 
- [x] AMC Ethernet connectvity 
- [ ] SDRAM 
- [ ] GTH transceivers and SFP 
- [ ] Silabs clock chip 
- [ ] SFP
- [ ] FLASH 
- [x] I2C tree (only AMC side)
- [ ] FMC

# RTM

- [ ] oscillators 
- [ ] basic clock distribution
- [ ] FPGA UART 
- [ ] FPGA programming 
- [ ] JTAG over USB
- [ ] I2C tree excluding analog front end (AFE) mezzanines 
- [ ] I2C tree including AFE
- [ ] AFE

jbqubit, 2017-06-21:
@gkasprow I've had several inquires as to the status of Sayma hardware testing. Here's a checklist for people to follow who want to track progress.

gkasprow, 2017-06-21:
The SDRAM 32 works perfectly
[image]

jbqubit, 2017-06-22:
Shift date for Sayma shipment to M-Labs. Now tomorrow.  

> 6/22: 1 each of Sayma_RTM & Sayma_AMC shipped to M-Labs (Florent)

gkasprow, 2017-06-24:
Second DDR 64 controller working. I really hate Vivado! It took 2 days to synthesize and make running Xilinx example design...
[image]

gkasprow, 2017-06-24:
16 GT links running in parallel at 12Gbit. Chip temperature 40deg without heatsink and with air cooling. After a few minutes of work no errors detected.
[image]

gkasprow, 2017-06-24:
Example eye scan using IBERT tool. All tests done with RTM loopback that mimics poor matching.
[image]

gkasprow, 2017-06-24:
SFP to SFP connectivity with 30cm copper cable
[image]

gkasprow, 2017-06-24:
No troubles with PRBS31 as well
[image]

jordens, 2017-06-24:
Just as a sidenote: That eye doesn't look physical to me. It is also unlike anything from #166. There are features much shorter than 0.01 UI, i.e. 1 ps in there. I'd suspect some Vivado bug in there.

gkasprow, 2017-06-24:
@jordens This has very low resolution - 8 units vertical, 8 units horizontal. In ISE it looked better.

gkasprow, 2017-06-28:
I tested Sayma with FMC->VHDCI adapter -> VHDCI carrier -> 3U RJ45 in a loopback. Works fine.
The signal enters via SMA, passes 32 times through FMC and 3U RJ45 extender and goes to the SMA output. Below is screenshot.
[image]

gkasprow, 2017-06-28:
SPI FLASHx4  works fine. FPGA boots after ~one second.

[image]

gkasprow, 2017-07-03:
There was issue with SCANSTA chip. I didn't work.  I figured out that switching it to mode with master port B1 (power on with JTAG probe disconnected) and then pulling TRSTB to GND made it working with master port B0. It seems that it needs some time after power ON. And indeed, configuration happen too early during power on. 200ms delay in MMC software solved that issue,
There was the same issue with Silabs was that it needs some time after power on. I2C initialisation failed when started too early and locked the chip. Initial delay made it working.

gkasprow, 2017-07-03:
JTAG on RTM works
[image]

gkasprow, 2017-07-03:
Silabs on RTM is also working. I set it to free-run at 125MHz and will test ethernet connectivity over GTP and SATA cable.

gkasprow, 2017-07-07:
Status update:
Mux tree works fine.
I managed to communicate with PLL chip from little FORTH program that does bit-bang SPI. I discovered that PLL chip responds with right ID but logic levels are compromised. The HMC7043 does not release SPI bus for some reason so I will have to either add 3-state buffer or route dedicated line to FPGA.
[image]

hartytp, 2017-07-07:
@gkasprow Great! Did you get RF out of the HMC830?

sbourdeauducq, 2017-07-07:
What does Analog Devices tech support have to say about the HMC7043 not releasing the SPI bus? Maybe it just needs some clock pulses with CS deasserted?

gkasprow, 2017-07-07:
This pin is bidir, and I keep CS high.

Maybe there is something else that block this line…

Clock line is also common for both chips.

gkasprow, 2017-07-07:
@hartytp Not yet, I was fighting with SPI...

jbqubit, 2017-07-10:
@gkasprow I added "SPI" checkbox for RTM and checked it since you reported it working in #216. Nice!

gkasprow, 2017-07-10:
@jbqubit One SPI chip works.
I'm trying to make others running and it is tricky.

gkasprow, 2017-07-12:
RTM clock switching, PLL, distribution works. This SPI is really tricky and needs correct init sequence.

gkasprow, 2017-07-12:
ADC and DAC SPI works
[image]

gkasprow, 2017-07-13:
Bizzare I2C behaviour.
The TCA9539 needs to have any write operation performed first to respond to read operations later on.
All other chips ACK read operations.
once you make write operation, the TCA9539 responds later on to any read operations.
If you don't to write operation first, the chip does not ACK its own address...

hartytp, 2017-07-13:
Great! So, it's just the DACs and ADCs left to test then...

gkasprow, 2017-07-13:
power supply issue is known - it was caused by too rapid switching of RTM supply - main 12V rail went to 6V for a moment and then caused 1.8V overvoltage event. I added R=1.5k C=1uF to the PMOS switch to slow down RTM 12V slope and it works fine. This was already added as an issue by Kermit - he noticed it on my previous AMC board.

gkasprow, 2017-07-13:
@sbourdeauducq  @enjoy-digital @jordens @jbqubit AMC+RTM are tested and working, I'm not aware of any issues. I didn't test:
- USB JTAG
- JESD204B to DAC and ADC
- Artix FPGA programming by Kintex.
- MMC access to Eternet
 I'm ready to send it to France.
Tomorrow morning I will send it.
On Monday I will receive 8 AMCs. I will pass one AMC+RTM set to Creotech, they will test ADC JESD204B and remaining less critical parts. I will focus on 3U boards now.

hartytp, 2017-07-13:
@gkasprow Massive congratulations to you and your team for designing and testing those boards!

gkasprow, 2017-07-19:
8 Sayma AMC are produced. On Friday I will test them.

gkasprow, 2017-08-08:
Front panels for Sayma AMC and RTM finally arrived. I will take them with Saymas to NACTI.

jbqubit, 2017-08-08:
​Excellent! ​

hartytp, 2017-12-01:
This issue covered basic functionality testing, which is complete.

Let's move to a new issue for more detailed analog testing (phase noise etc).

closed 2017-12-01 by hartytp