sinara-hw/sinara #226: 3U ADC (Novogorny) tests

gkasprow, 2017-06-30:
ADC boards arrived
[image]

hartytp, 2017-07-01:
@gkasprow What tests do you plan to do on this board? If we're going to make a new version with a different ADC, I don't think we need to test this version too carefully.

gkasprow, 2017-07-01:
That's true, but we have to make sure that AFE, supply and reference are working fine.
So I'd like to test/measure:
- [ ] SNR for all gain settings
- [ ] spectrum with full scale and mid scale input signal. It will show us power supply performance.
- [ ] CMRR
- [ ] channel isolation
- [ ] susceptibility to external EMI sources (i.e. neighbouring boards). I will test it with EMC equipment.
- [ ] gain switching.
ADC chip itself is not that critical here. We know its characteristics or can measure them with devkit.
I will use ADI software to to all evaluation.

hartytp, 2017-07-01:
Sounds like a good plan, thanks Greg!

gkasprow, 2017-07-07:
ADC with shielding boxes
[image]

gkasprow, 2017-07-13:
Today I powered it on. The DC/DC converter generates +3/ -2.5V without load (output chokes removed). Will debug it next week.

gkasprow, 2017-07-14:
power supply works. There were two tantalium capacitor with wrong polarisation.

gkasprow, 2017-07-18:
I managed to communicate with devkit and acquire samples but it works only for DC.
I ordered DC890 interface to do AC characterisation. Now switching to Zotino debugging.

hartytp, 2017-07-18:
@gkasprow Thanks for the update. What tests do you want to do before you send us some of the ADC/DAC the boards?

hartytp, 2017-07-18:
(I'm keen to give M-Labs a DAC board asap so that they can write the software for it.)

gkasprow, 2017-07-18:
I'd like to test ADC AC performance to if see if gain change works and do multichannel  tests at high SPI rate.
DAC supply works correctly. Temperature regulator has some issues with stability - PI controller needs some component value optimisation. Once I connect to the devkit and check functionality, I would be able to send both boards to Mlabs.
Later on I will continue tests when Mlabs ARTIQ support gets ready

jordens, 2017-07-18:
In the absence of Kasli, and metlino we'll have to develop with the vhdci carrier and fmc-vhdci and kc705. Do we have them or could you send some as well?

gkasprow, 2017-07-18:
Recently batch of fmc-vhdci boards were produced for Joe and shipped to US. There were more boards produced and are kept on stock so you can buy them.
Pls contact @mmroz88

jbqubit, 2017-07-18:
@jordens I can send you one of the fmc-vhdci I ordered. What address should I use?

hartytp, 2017-07-18:
Thanks @jbqubit

hartytp, 2017-07-18:
@jbqubit Do you have a spare VHDCI_Carrier for @jordens card as well?

jbqubit, 2017-07-18:
I do not have a spare VHDCI_carrier.

sbourdeauducq, 2017-07-19:
Please send the adapter(s) to M-Labs HK, we'll also need them for Zotino.

M-Labs Limited
Workshop 15B 6/F Block B1
17 Ko Fai Road, Yau Tong
Hong Kong
+852-59362721

jordens, 2017-07-19:
Might be better to send them to QUARTIQ.

gkasprow, 2017-07-19:
I have one spare VHDCI carrier.

hartytp, 2017-07-19:
Thanks @gkasprow. NB this has a slightly different pin-out to the new version, due the the reallocating of cc pins...

jbqubit, 2017-07-19:
@jordens @sbourdeauducq Am I sending qty 1 fmc-vhdci to Hong Kong or Germany?

jordens, 2017-07-19:
one to HK.

gkasprow, 2017-07-22:
Supply works fine. Just reversed caps caused all the problems. DAC died on of the boards probably due to reverse biasing. Shottky diodes needed at the LDO outputs to prevent such situation.

jbqubit, 2017-07-24:
@jordens For reference, what is mailing address of QUARTIQ?

hartytp, 2017-07-31:
@gkasprow What's the status of the tests on the ADC/DAC boards? When do you anticipate being ready to ship the prototypes to us/M-Labs?

gkasprow, 2017-08-01:
the boards are on a way to mLabs

>
>

hartytp, 2017-08-01:
Great! Can you also ship us a couple of each to test out, please?

gkasprow, 2017-08-01:
Once I test them I will do so.

jbqubit, 2017-08-01:
​Please send one my way too for testing if there's enough of the prototype
boards. ​

gkasprow, 2017-08-01:
@jbqubit I produced 5 proto boards. You have to talk with @hartytp  about it:)

hartytp, 2017-08-02:
@gkasprow That's fine, we don't need all 5, so do send one to @jbqubit.

Can you remind me where we are with this. AFAICT:
- Both boards power on, and have had basic functionality confirmed with eval software
- For AC tests on the DACs, you're waiting for M-Labs to write their driver, so you can drive the DAC from a KC705. (Then you can use the DMA feature in ARTIQ to play back a long sinusoidal waveform for testing).
- For AC tests on the ADC, you've ordered a faster devkit to make the measurements.

Is that correct? What's the ETA for the ADC tests?

gkasprow, 2017-08-02:
I'm doing ADC tests today. I plan to finish them till Friday.

hartytp, 2017-08-02:
Great! Thanks for the update!

gkasprow, 2017-08-02:
I have some issues with ADC readout while reading massive data.
It seems that it is some silicon issue and have to consult Linear support.
The ADC SDO output clock gets some strange glitches.
It is created from SDI clock and simply divided by 2.
[image]
[image]
The glitches occur from time to time - had to configure scope to trigger on them.
I do not observe them on devkit.
All voltages are fine.
I thought that too steep clock edges may cause this so added 100R series resistors to SDI and CLKI but it did not help.

jordens, 2017-08-03:
I don't understand the setup. Who divides the clock? The ADC? What's the scope traces? SCKi and SCKO?
Otherwise I can see slow edges being a problem. Not steep ones.

hartytp, 2017-08-03:
I think that in CMOS mode, the SCKO pin isn't just a simple divider. e.g. the data sheet states that in CMOS mode, SCKO is driven low at the falling edges of BUSY.

This pin isn't used in CMOS/SPI mode, so can we just ignore these glitches?

gkasprow, 2017-08-03:
The scope shows SCKI and SCKO. The clock signals are driven by LVDS receiver and SCKO is driven by ADC chip itself.
For some reason the devkit does not receive the data correctly - it says it is corrupted from time to time so this could be the problem. In general, D-flip flop may not toggle correctly when clock edge is not fast enough. It seems that the devkit uses SPI in DDR CMOS mode and needs thic clock for data capture.

gkasprow, 2017-08-03:
Clock edges are steep, look at scope sample rate;)

gkasprow, 2017-08-04:
I got it.
It is caused by translation from CMOS to LVDS and back to CMOS.
The devkit FPGA generates too long slope so LVDS receiver interprets it as one or more signal transitions.
[image]
legend:  purple - SCKI on devkit
green - SCKI on LVDS
blue - SCKI on ADC chip 
yellow - SCKO

hartytp, 2017-08-04:
Nasty. Good spot Greg!

gkasprow, 2017-08-04:
I added LVC14 buffer and it works now.
[image]

gkasprow, 2017-08-04:
channels 3..8
[image]

gkasprow, 2017-08-04:
input signal is 80mV, gain is 100

gkasprow, 2017-08-04:
But there is an issue with BNC connectors. I've just noticed thab bottom and upper connector grounds are SHORTED!
it means we cannot use common mode chokes or have to replace connectors with separated grounds.
This introduces crosstalk to neighbouring ADC channel, in this case channel 2

gkasprow, 2017-08-04:
channel 2, F=10k, gain = 1000
[image]

gkasprow, 2017-08-04:
With TERM=OFF, connected channel does not change much but on  floating channels i observe some signal on 25th FFT sample. Fs=1MHz/8=125kHz so the signal must be 6.1kHz
[image]

gkasprow, 2017-08-04:
It is also visible on active channel but only with gain = 1000

gkasprow, 2017-08-04:
DC/DC converter works at 418kHz so if we sample it with 125kHz there is no way to get 6.1kHz

gkasprow, 2017-08-04:
EMC shield and unplugging of BNC cable does not help

gkasprow, 2017-08-04:
When I touch BNC conectors by hand I get such result
It seems this 6kHz is induced from "the air"
[image]

gkasprow, 2017-08-04:
For this configuration it does not matter if the signal is connected to BNC hot pin or ground -they are fed to same amplifier. And that's why BNC works as antenna ...
With G=1000 10mV saturates the ADC and that's why when I touch it by hand we get spectrum of square signal

gkasprow, 2017-08-04:
And that's why I get such strange signal on neighbouring channel that share same GND with BNC connector
Common mode signal extracted from first channel is sampled by second channel.
This makes sense now.
[image]

gkasprow, 2017-08-04:
@hartytp this should not be a problem with EEM. The edge on FPGA with LVDS signals are much steeper.
But we may experience it ON DIO boards with TTL inputs. There is '245 buffer at the input and isolator but still, such situation may happen.

hartytp, 2017-08-04:
That doesn't surprise me. 

In most labs there is so much EMI (switching supplies, fluorescent light, conducted emissions on the mains wiring etc) that the combination of high gain and floating inputs usually causes this kind of issue...

In practice, this won't be an issue, since we will always drive the inputs from a reasonably low impedance (say a photodiode receiver).

Shall we agree to do all tests with all inputs terminated in a test load (say, 1k)?

hartytp, 2017-08-04:
For the second revision, can we try to find BNCs with independent ground connections?

gkasprow, 2017-08-04:
@hartytp These connectors we use are hell expensive. There are some low cost versions from non-name Chinese company but quality may be an issues.
Another quick fix would be additional vertical board where straight BNC connectors are installed.
Moreover, We can try to make it in such way that can be plugged to the IDC on ADC board.
Anyway, I will find some solution.

hartytp, 2017-08-04:
> Another quick fix would be additional vertical board where straight BNC connectors are installed.
Moreover, We can try to make it in such way that can be plugged to the IDC on ADC board.

That sounds like a good solution, particularly if it lets us avoid expensive connectors.

gkasprow, 2017-08-04:
Here is what I get from ADC on devkit.
It seems I have to have a look or build good sine-wave source. Now my generator defines measurement limit.
The noise floor with generator connected.
[image]
the noise floor with generator disconnected
[image]
crosstalk on CH0 while CH3-7 are connected to 20kHz generator
[image]

hartytp, 2017-08-04:
> the noise floor with generator disconnected

What does that translate to in terms of LSB RMS noise?

> crosstalk on CH0 while CH3-7 are connected to 20kHz generator

What what the signal level on channels 3-7? (i.e. what's the cross-talk in dB?) Looks like it's not a problem though, which is good.

sbourdeauducq, 2017-08-04:
> There are some low cost versions from non-name Chinese company but quality may be an issues.

I found a company in Shenzhen that sells RF connectors (and cable assemblies) at low prices, and I'm quite happy with them. I can give you the contact, though you may need a Chinese-speaking person (it's easier to order when you can point at things, but not so much remotely...)

[175 more replies]

closed 2017-11-16 by hartytp