sinara-hw/sinara #227: 3U DAC (Zotino) tests

gkasprow, 2017-06-30:
DAC boards arrived
[image]

jordens, 2017-06-30:
Nice!
If you managed to build a switched DC-DC DAC/ADC with noise as low as envisioned, that would be a massive achievement.
I am especially looking forward to graphs/number on:

* Noise (really wideband, different output loads)
* Crosstalk
* Temperature coefficient
* Glitches on major carry

(same for #226)

hartytp, 2017-06-30:
Great job Greg, that looks really good!

How many of these boards have arrived?

What tests are you thinking of doing (beyond confirming basic functionality)? A few things I thought about were:

- Measure step response
- Measure cross-talk (full-scale square wave at ~50kHz on one channel, another channel at 0V)
- Measure noise PSD (0.1Hz to 10MHz)
- Measure spurs due to SMPSs, digital-cross talk, pickup etc. Have one channel updating as fast as possible and measure noise on another channel
- Confirm that temperature controller works. Measure temperature stability of DAC in lab conditions
- Measure drift of channels with temperature controller running. Set one channel to Vmax/0V, and look at drift over ~24 hours to confirm gain and offset drifts.
- Check for thermal issues (drifts) when outputs are slewing at max rate (e.g. testing for issues due to self-heating in the OpAmps). What’s the best way to test this?

Configuration for measurements is: DAC->3m SCSI cable->HD68 to IDC->IDC to BNC->scope.  CMCs and a ~500kHz LPF on the IDC-BNC board as previously discussed.

gkasprow, 2017-06-30:
I produced 5 sets of all boards.
Please make a checklist with all measurement you want me to perform.
Greg

hartytp, 2017-06-30:
I've edited the top post with an initial list of tests to perform. Anyone should feel free to edit this if there are more tests they think we should do, or if they can think of different/better tests we should do instead.

> I produced 5 sets of all boards.

Ace. Presumably, you need 1 for testing. Could you send one to @jordens for software development, and post the other 3 to me, please? @jordens do you need a VHDCI carrier board?

gkasprow, 2017-07-01:
Yes, I will post boards once I make them running.

gkasprow, 2017-07-01:
@hartytp @jordens One more thing about tests - I will do some tests that ADI evaluation software lets me perform. For rest of them I will need ARTIQ test software running on KC705 or Sayma. This approach will also validate boards operation in final environment.

gkasprow, 2017-07-07:
DAC with shielding boxes
[image]

hartytp, 2017-07-07:
very nice!

gkasprow, 2017-07-18:
Status update:
Power supply works
Thermal controller needs some tuning of PI regulator 

one issue so far: DAC heats up and after a minute causes LDO overcurrent protection. Board consumes 240mA when cold and 0.4A after one minute when DAC gets hot. Needs further investigation. I will check if the issue is on other boards.

dhslichter, 2017-07-18:
>one issue so far: DAC heats up and after a minute causes LDO overcurrent protection. Board consumes 240mA when cold and 0.4A after one minute when DAC gets hot. Needs further investigation. I will check if the issue is on other boards.

yikes!  keep us posted.

hartytp, 2017-09-03:
@jordens @sbourdeauducq What's the status of ARTIQ support for Zotino? IIUC, @gkasprow's current plan is to test Novo using a DevKit (and some hacks to program the PGIAs), but to use Artiq to test Zotino. This will require the Artiq Zotino driver.

If possible, I'd like to have Zotino and Novo prototypes tested and shipped before Kasli/Urukul prototypes arrive to avoid building up a backlog.

jordens, 2017-09-03:
We need the hardware to write ARTIQ support. Writing that blind is something we want to avoid. The driver for the AD53[67][0123] family is already there and tested.

Also we have the backlog from Sayma. We really need to focus on that first.

hartytp, 2017-09-03:
> We need the hardware to write ARTIQ support. Writing that blind is something we want to avoid. 

Sorry, I thought that @gkasprow had already shipped you a Zotino and a Novo. @gkasprow Could you send M-Labs/Quartiq 1 of each of these boards, please?

> Also we have the backlog from Sayma. We really need to focus on that first.

ACK, that should take priority. But, I don't want to put off testing the Zotino prototypes for too long. So, if Artiq Zotino support is more than a few weeks off, we should look into options for testing Zotino without Artiq (e.g. use a microprocessor eval board)...

jordens, 2017-09-03:
We may already have them in HK.
That plan is ok.
Ping @sbourdeauducq @mntng @whitequark

gkasprow, 2017-09-03:
@hartytp I've sent them already with Saymas.

hartytp, 2017-09-03:
@gkasprow Thanks for confirming!

@sbourdeauducq et al, let me know what you think the timeline for Artiq support of Zotino is likely to be, and we'll come up with an appropriate plan for hardware testing.

hartytp, 2017-09-03:
@gkasprow One other thing: are there any other things that can be done on Zotino/Novo without Artiq support. e.g. IIRC, the temperature controller still needs debugging. It might be worth sorting those kinds of issue out first, while M-Labs write drivers.

gkasprow, 2017-09-03:
Yes, I'm just preparing HDL for little CMOD7 FPGA devkit to be able to take ober the SPI bus and configure registers in ADC in Zotino and Nobo boards. I will use little FORTH CPU to do the job.
The temperature controller needs adjustment of regulator time constant - now it oscillates with sub-Hz rate.

hartytp, 2017-09-03:
@gkasprow Thanks for the update! That sounds great. Let me know how it goes.

T

sbourdeauducq, 2017-09-04:
We have a Zotino hooked up in HK to a KC705 using the FMC DIO and VHDCI carrier. @mntng is testing it. There is some bug/regression in the compiler/runtime: https://github.com/m-labs/artiq/issues/828 (nothing that sounds difficult to fix, though)

hartytp, 2017-09-04:
@sbourdeauducq Thanks for confirming.

arpitagrawal23, 2017-09-06:
Hello Everybody
I'm working with @jbqubit and have a Zotino board with me (connected via KC705 and VHDCI). I'm aiming to contribute to writing driver for Zotino. If any of you have some code in your git branch, would you let me know please? I can use it to avoid duplication of efforts and build something that is not yet done. Thank you.

sbourdeauducq, 2017-09-06:
The driver itself is there, it just needs "glue". You should talk with @mntng, I'll write you both an email.

dhslichter, 2017-09-25:
@gkasprow @hartytp we made a duplicate of the Zotino board filters, running from an AD5360 DAC, for some work in our lab.  Measured the output noise, which appears to be strongly influenced by environmental pickup after the filters, but with ferrites on the cable from DAC box to spectrum analyzer we measured the output noise to be ~2-3 nV/rtHz in the 1-10 MHz band, with some structure but nothing really exceeding 4 nV/rtHz.  Don't have the data traces handy, but it'd be a good sanity check to compare once @gkasprow gets the DAC testing set up.

gkasprow, 2017-09-26:
I'll start DAC evaluation soon, once I finish tests with ADC. Hopefully this will happen this week.

hartytp, 2017-10-04:
@gkasprow How many of these boards do you have populated atm? If you have more than one, can you send me one now, please? That way we may be able to help you with some of the testing.

gkasprow, 2017-10-04:
@hartytp We have 3 at WUT, one in MLabs and one in Joe's hands.

gkasprow, 2017-10-04:
@hartytp I just assembled test suite with original ADI devkit. Unfortunately it allows only DC tests, so we will need ARTIQ diver very soon.

jordens, 2017-10-04:
AFAIK @mntng @sbourdeauducq @arpitagrawal23 @jbqubit are all working on that driver.

hartytp, 2017-10-04:
> @hartytp We have 3 at WUT, one in MLabs and one in Joe's hands.

- [ ] @gkasprow Can you post us 1 asap then, please? That way, we can help with some of the testing.

-----

- [ ] @gkasprow @cjbe Please can you double check the list of tests above and confirm if there are any other tests you think we should do (or if there are any we shouldn't bother doing)?

> Unfortunately it allows only DC tests, so we will need ARTIQ diver very soon.

Most of the tests currently on the list can be done with a DC signal from the DAC. If you confirm the DC performance and the step response (90%-10% rise time), we can probably help with the AC measurements.

> AFAIK @mntng @sbourdeauducq @arpitagrawal23 @jbqubit are all working on that driver.

That's quite a lot of collective programming might! Any idea when this will land?

gkasprow, 2017-10-04:
@hartytp Once I cnfirm it cooperates with devkit, will post it ASAP. Hopefully will be able to post on Monday.

hartytp, 2017-10-04:
@gkasprow Great, thanks! Can you post some of the adapter boards as well, please (BNC->IDC and HD68 to IDC)? That will help with testing.

gkasprow, 2017-10-04:
@hartytp sure.

gkasprow, 2017-10-13:
@hartytp  Today I finally did tests with DAC boards. Digital part works well and I'm able to set voltage on specified output. So  @jbqubit 's student should be able to make it running.
I found some strange issue. With wrongly soldered decoupling capacitor on -12V rail, the voltage is around -2.5V and DAC gets damaged in such conditions. It happened on 2 of the boards.
This would not happen in correctly working boards with correctly installed capacitors, but I'm curious what is the mechanism of the damage.

gkasprow, 2017-10-13:
@jbqubit please make sure that your student won't connect supply of the board with wrongly soldered caps.

hartytp, 2017-10-13:
> @hartytp Today I finally did tests with DAC boards. Digital part works well and I'm able to set voltage on specified output. 

Great! Can you ship us a board next week then, please? Let me know when you're ready to start  analog testing.

gkasprow, 2017-10-13:
@hartytp we will manage it. With devkit I can do only DC tests (crosstalk, stability)

arpitagrawal23, 2017-10-15:
> @jbqubit please make sure that your student won't connect supply of the board with wrongly soldered caps.

I was not aware of this and had connected power supply several times over the last few weeks. Now, even after correcting the capacitor polarity, the board is not working. My code seems fine, I've verified SPI signals on the oscilloscope, but I'm not getting output at terminals of Zotino board. While there are chances that my code is buggy, but can we eliminate the possibility of Zotino being broken by doing some test on it?

gkasprow, 2017-10-15:
Is your DAC chip getting hot?

sbourdeauducq, 2017-10-17:
@gkasprow since we don't have Altium and this is not on the silkscreen: where exactly are those caps? I measured the voltage on the three tantalum caps on the top side and they are polarized correctly on the HK board.

gkasprow, 2017-10-17:
The capacitors are mounted correctly according to PCB silkscreen. They are placed in wrong way on the schematics.
Here are pics with wrongly mounted capacitors:
[image]
[image]

You can also use free Altium viewer.

sbourdeauducq, 2017-10-18:
Ok, the HK board had been fixed.

gkasprow, 2017-10-19:
A few more measurements:
[image]
[image]

gkasprow, 2017-10-19:
As a next step I switched heating on and noticed that regulator disabled heating MOSFET at 50 degrees but IC kept heating itself until it failed.
On two other boards I had same situation.
So for some reason, heating of DAC chip abve 50 degrees causes some internal latchup which starts consuming too much current from -12V source until LDO overheats and disables its output.
So the DAC boards work providing that we don't heat the DAC chip. There must be a reason behind that.
Any ideas?

jordens, 2017-10-19:
You will want to load the offset DACs with 0x2000 for all tests.

gkasprow, 2017-10-19:
I measured what happens during power on cycle.
And it's quite interesting.
3V3 wakes up much earlier than rest of rails and is't obvious.
Then goes -12 and 12V and then reference. And this reference causes spike on negative rail. Such non-monotonic supply may cause latchup and such effect is observed. 

[image]

gkasprow, 2017-10-19:
there are visible oscillations and it is caused by the fact that LC1964 is missing ceramic cap at its output. There is such cap close to the DAC but it seems the equivalent ESR is too high. So I added 1uF 0805 cap and oscillations are gone. But the spike is still there...

[image]

gkasprow, 2017-10-19:
I lowered negative power supply to -4.8V. And there is no spike so far. DAC stays cold.
[image]

hartytp, 2017-10-19:
@gkasprow What's happening on the other voltage rails? e.g. the +-13V rail and the OpAmp supply +-12V?

gkasprow, 2017-10-19:
They are stable. Do you think they could case any issue? The output amp is conneted by 2.8k resistor to the DAC so nothing should happen.

gkasprow, 2017-10-19:
I connected shutdown pin of VSS to reference:
[image]

And this is how the opamp rail looks like.
blue - positive one:
[image]

And negative one (blue)
[image]

hartytp, 2017-10-19:
hmm...the data sheet doesn't really mention any special sequencing requirements. e.g the digital supply must turn on before we apply signals to the digital pins. But, that's about it.

hartytp, 2017-10-19:
@gkasprow To check that I understand you, the issues you see are:

1. When the reference switches on, there is a large current draw on the DAC's negative supply, which causes that rail to sag by a couple of volts. (But no obvious signs of latch up etc).
2. When you heat the DAC above 50deg it draws a large amount of current from the negative rail until it self destructs.

gkasprow, 2017-10-19:
@hartytp That's correct. I have a suspicion that the DAC is already on the edge of latchup which gets accelerated when it's hot. And such latchup condition may happen when there is wrong power sequence or supply rises in non monotonic way.

gkasprow, 2017-10-19:
Another, standard condition for latchup are digital signals before power supply. But this should be the case only with DVCC rail which in our case is same as other logic.

sbourdeauducq, 2017-10-19:
FWIW @mntng also had an episode where the DAC got very hot after powering the board. Did not happen in subsequent power-ups AFAICK.

gkasprow, 2017-10-19:
I already damaged 3 DAC chips which are quite costly so would like to find a cause first before I burn another ones.

gkasprow, 2017-10-19:
So I will modify the board in such way that opamps supply will go first,then VSS, VDD, DVCC as it is recommended by msamera[ here](https://ez.analog.com/message/164258).

gkasprow, 2017-10-19:
@hartytp this will take a few days, so I can send you Novo boards first. OK?

hartytp, 2017-10-19:
@gkasprow Yes, please do send Novo asap.

> there are visible oscillations and it is caused by the fact that LC1964 is missing ceramic cap at its output. There is such cap close to the DAC but it seems the equivalent ESR is too high. So I added 1uF 0805 cap and oscillations are gone

Did you check the supplies on Novo out of curiosity? Does this issue with the negative rail oscillating affect Novo as well? (We can check this when the boards arrive if necessary)

[148 more replies]

closed 2018-01-15 by hartytp