sinara-hw/sinara #228: Kasli SFP pin9 (RS1)

jordens, 2017-07-04:
On an SFP+ ([SFF-8431 R2.1](https://www.10gtek.com/templates/wzten/pdf/SFF-8431-(SFP+%20MSA).pdf)), pin 9 is RS1, i.e. TX rate select. Could we either provide a solder bridge for a 10k pullup or an FPGA pin to drive it?

gkasprow, 2017-07-04:
We will have solder bridge like on Sayma board. There are not free pins on FPGA.

jordens, 2017-08-10:
@marmeladapk @gkasprow Looking at the current schematics, there is a resistor on the SFP pin 9 now (albeit unnumbered, I assume that means it's not in the layout yet). Can we close this?

jordens, 2017-08-10:
Ah no. RS1 needs to be high for "TX signalling rate greater than 4.25 GBd". I.e. it should be left open (there is a 30k pull down in the module) for slow operation and pulled high for fast operation. When we want to operate faster than 4.25 GBd, we need the pullup. Can we get both options, the 0R to GND and a 4k7 pullup?

jordens, 2017-08-10:
This also applies to the Sayma SFP+ cages when we are trying to do >4.25 GBd DRTIO links @sbourdeauducq

gkasprow, 2017-08-10:
@jordens This was already implemented - 0R to GND and pullp to VCC in Kasli and Sayma.

gkasprow, 2017-08-10:
Anyway, most SFP+ modules ignore this pin

jordens, 2017-08-10:
OK

closed 2017-08-10 by jordens

marmeladapk, 2017-08-11:
@jordens Do you want this pin pulled up or down by default? Currently it's pulled down.

jordens, 2017-08-11:
@marmeladapk 0R "pulldown" and 10k pullup both populated by default. We can just remove the 0R pulldown when necessary.

marmeladapk, 2017-08-11:
@jordens I'll change it in the next commit.