sinara-hw/sinara #232: FMC DIO 32ch LVDS is missing screw

jbqubit, 2017-07-12:
The FMC to VHDCI adapter shipped by Creotech to UMD is missing the pair of nuts customarily on boards (circled in image). Please update the BOM to include this part. AFAICT it's not included in MOLEX_71430-0013. I've found nuts locally in a junk drawer to use with my adapter. 

[image]

Including a pair of stand-offs and 4 screws for attaching the FMC side would be one less thing for end users to source.

dhslichter, 2017-07-12:
+1 on the FMC standoffs and screws, that is great to include with any FMC daughtercard...

gkasprow, 2017-07-12:
Joe, these are not the boards I tested and let the guys to ship to you
Mine were brown and had 32 channels, these are some old 8channel versions
that should never go to client.
And the boards I tested had both screw.
I'll explain it.
Greg

jbqubit, 2017-07-12:
They sent four copies of the following board.

[image]
[image]

jbqubit, 2017-07-12:
I had pasted an image from OHWR and circled the missing screw since I didn't have a camera handy. The boards Creotech sent are fully stuffed as illustrated in the two photos in the previous post.

gkasprow, 2017-07-12:
OK, these boards are fine, I tested them and did this little modification
to fit pinout of Metlino/VHDCI carrier.
I thought they were brown, sorry, my fault.
The one you shown to me is different because has only 8 channels assembled.
Greg

gkasprow, 2017-07-12:
OK

hartytp, 2017-08-20:
@gkasprow Is this being handled by Creotech? If so, can we close this issue?

closed 2017-08-30 by gkasprow