sinara-hw/sinara #233: "Smart" backplane for EEM/Kasli ecosystem

dhslichter, 2017-07-12:
Moving my comment from #204 to a new issue:

Might it be better in the long haul to have Kasli's functionality actually just integrated on the backplane? You would have a slim "dumb" card that brings SFP and coax connections to the front panel, and connects them to the backplane via some high-performance but lower-pin-count connector, suitable for gigabit signals. Then the Artix FPGA, clock distribution, etc all live on the backplane, and you don't need crazy fat connectors just to route all the EEM signals and clocks off the Kasli and on to the backplane. This would also free more front panel space, because if you are adding 96-DIN mezzanines to Kasli then this obviously will make it wider, which may be undesirable for some.  

The only downside I can see is that if your FPGA on the backplane goes bad for some reason, it's more work to change it out....but this seems like it should not be a very common occurrence.

gkasprow, 2017-07-12:
@dhslichter Take it into account that to place such big FPGA on backplane, you need at least 12 layers. And such big 19" board would be hell expensive. Moreover, DIN connectors are THT so it means you cannot place BGA under it so some of slots would be wasted. At least 12..14 HP. 
It's much cheaper and more convenient (maintenance) to keep backplane simple and make Kasli complex.
We don't have to use DIN96 connector for Kasli. We can easily use any type of backplane connector, i.e. 170 pin AMCone we use in Sayma/Metlino. We can stack them to have 340 or even 680 pins and they are dedicated to high speed signalling. These backplane connectors are cheap, from Kasli side you don't need any (PCB fingers).
We can make current Kasli compatible with such ones.
look [here](http://d2lupdnmi5p5au.cloudfront.net/i__src189536162cb3fa5c582207b5cdf44c87_para1457fb4c00698144390d6358806ef8f.jpeg)

dhslichter, 2017-07-12:
ack @gkasprow, sounds very logical.

gkasprow, 2017-11-10:
There is interesting initiative here: http://easy-phi.ch/ 
They use different approach, based on active backplane with USB as main interface.
There is plenty of work they did and it would be nice to find common denominator between our EEM and their backplane. Example board schematic is [here](http://easy-phi.unige.ch/application/files/5414/7306/5683/Low-cost_Dual_Discriminator_-_Schematics.pdf). On page 2 in upper right edge there is DIN connector. 
I see potential of marrying our solution with their. In case of the backplane, we can simply use pins which are grounded for our LVDS lanes.
In this way such backplane would support both DRTIO and non-DRTIO boards with USB interface where simple controller based on USB hub or some RPI/Beaglebone do the job. In case of DRTIO extensions we simply plug there Kasli. Or we can mix the two and mix boards within crate.
We could build all [planned hardware](https://github.com/m-labs/sinara/wiki/PlannedHardware) to be compliant with this approach.
It would be nice to join our and their efforts and not duplicate the work.
5U is not an issue, one can simply place 1U dummy panels in between ours.
This is just an idea, nothing more.

jordens, 2017-11-10:
Pretty sure that's been dead for a while. I chatted with them back in '15 but haven't seen any movement [since](https://github.com/easy-phi/main) then.

gkasprow, 2017-11-10:
@jordens at least we can reuse some ideas for slow control things :)

jordens, 2017-11-10:
Yep.

jordens, 2018-01-18:
The connection between the panel board and the smart backplane could be 4x SATA plus a ribbon for all of USB/Power/SFP-Management/LEDs and maybe MMCX-SMA pigtail for the clock.

gkasprow, 2018-01-18:
The question is if we really want to have smart backplane.
It is part that is difficult to replace and also the most expensive.
It's much easier to break active backplane, while passive one can be broken only mechanically.
I'd keep the backplane dummy and put all the smartness to the Kasli or other controller.
Today we had discussion with @marmeladapk and he asked simple question - why we really need this backplane. I know that it's nice looking, we get rid of cable nest but we add: cost, all boards must be bigger to have access to it so they will be more expensive, the backplane does not come for free, we limit the slot availability in case we use 8HP boards. We can of course make some slots 8HP and some 4HP.
Another aspect is crate size - I saw users having different HP size. Some use table-top crates, others prefer 19" rack mount ones. We would need dedicated backplanes here as well.

If we really want to go for backplane, it would be nice to keep some compatibility with existing EEM boards.
In case of short ones, it's easy - just add some extender with IDC socket and backplane connector.

We can do the same to existing Kasli - extend IDC cables to the [backplane connector](http://docs-europe.electrocomponents.com/webdocs/14a4/0900766b814a4339.pdf) attached directly to another end of the IDC cable. It has 64 pins (96 pin version also has 64 connected pins) so would split to 2 EMMs. But this means we need 4 additional such backplane IDC connectors!

If we go for 96 pin ones, we would need only two of them, but dedicated adapter board would be needed that would translate 8 30 pin IDCs to two straight DIN 41612.
Another issue is clock routing.

Yet another option that do not use any cables is to add to Kasli two  mezzanines with clock distribution and female connectors that:
- first one mates with 4 EEMs and second backplane connector 
- second one mates with next 4 EEMs and third backplane connector.

In this way we keep compatibility with  all existing boards
Urukul, Sampler and Mirny would need to use 8 HP slots and dedicated mezzanines - EEM to backplane adapters

jordens, 2018-01-19:
I agree with that.

hartytp, 2018-01-19:
@gkasprow I agree completely: while the idea of a BP sounds nice, I'm not sure we'd use one even if it were available:
- Maybe I'll change my mind once I've used these boards more, but I don't think the ribbon cables will be that much of a pain. Should be fine from a SI/PI perspective
- Even a passive BP will be more expensive, a fiddle to mount and stock than the current solution.
- Kasli is already quite limited in terms of IO, so we cannot afford to waste any. However, we have boards with different widths and different numbers of EEM connectors required. I can't see any efficient way of ensuring that all Kasli IO can be used from the BP in a sensible way (e.g. not involving lots of unused space in the rack).
- BP unlikely to be compatible with lots of case sizes. A lot of our Kasli will be in relatively small crates which are unlikely to have a BP, so we will still have the IDC there.
- Have to increase size of boards to mate with BP connector.

I'm going to (perhaps optimistically) close this issue for now. My reason for doing this is that currently there does not seem to be a clear plan for the BP, or even necessarily a clear use case for it. Moreover, no one is currently taking the lead on it and I'm not sure we have the resources to think about it properly. My suggestion is that we keep this closed until someone has the time/motivation to take an active lead on this.

closed 2018-01-19 by hartytp

dhslichter, 2018-01-19:
Following on, I agree with the above that the backplane does not seem to provide important benefits, and introduces a lot of extra complication and cost, at this point.  The big points of the Kasli ecosystem are low cost, flexibility, adaptability, and simplicity, and the backplane idea runs counter to all of them.

gkasprow, 2018-06-19:
For future reference - there is a COTS backplane that seem to fit perfectly to our needs - the Compact PCI Serial standard. With most basic configuration it has 8x LVDS pairs, I2C, 12V and management power rail. And it is 3U Euro crate. ButiIt supports only 8 cards. The modules are 4 HP. 
Short introduction is [here](https://www.arcobel.com/datasheets/CompactPCI%20Serial%20-%20Compendium.pdf)
Of course that will be more expensive than existing solution, mainly due to connectors.

jordens, 2018-12-04:
Let's continue the cpci discussion here.
A couple more pieces of info from [Schroff](https://schroff.nvent.com/wcsstore/ExtendedSitesCatalogAssetStore/Attachment/SchroffAttachments/Documents/63972-333.pdf) [Hartmann](https://www.hartmann-electronic.com/products/backplanes/cpci-serial.html?file=files/hartmannelectronic/Support/Downloads/Documentation/Backplanes/Compact-PCI-Serial-Backplane-Catalog.pdf)