sinara-hw/sinara #234: PCB_3U_VHDCI_breakout add VHDCI cables to BOM

jbqubit, 2017-07-12:
Each VHDCI_breakout board needs two (2) VHDCI cables. Please update BOM to include a pair of the required cables 1 meter length. It's nice if end users don't have to figure out how to source these somewhat-special items. 

As per offline conversation with @gkasprow the following P/N works.

Tech Choice- 708 0.8 mm/0.8 mm VHDCI 0.8mm SCSI Cable 3ft Offset
Brand: Tech Choice           P\N: 708 

Below shows what happens when cables are not "offset" variety. FYI

[image]

cjbe, 2017-09-18:
The long bushing power jacks (L712RA) are problematic to use with normal length plugs - if the plug is too short, the ground connection is not made, but the plug outer still connects to the body of the jack, which can provide a ground connection via the rack and front panel to other EEMs!

There should be a note somewhere in the documentation warning of this, and suggesting the Switchcraft  761K as a default safe connector to use.

hartytp, 2017-09-19:
@gkasprow Do we want to pick a different power connector for these boards, which is compatible with the connectors used on standard 12V wall warts.

jordens, 2017-09-19:
@dtcallcock advocated these.
But @cjbe with these connectors you should only use "floating" power supplies because the inner conductor will connect first most of the time. I am not particularly worried about that.
I'd like to go to the more common ID 2.1mm x OD 5.5mm type (still inner connector positive), i.e. [722RA](http://www.jameco.com/Jameco/Products/ProdDS/1772934.pdf)

hartytp, 2017-09-19:
> I'd like to go (for Kasli and Clocker) to the more common ID 2.1mm x OD 5.5mm type (still inner connector positive), i.e. 722RA. Those have hundreds of matching power supplies on Farnell/Reichelt etc.

Let's make sure that we fully specify the connector by giving the length dimension as well.

I'm happy with any connector that works with standard +12V PSUs available on Farnell and elsewhere.

FWIW, 2.5mm seems more common than 2.1mm from a quick search (but, as always, crap metadata on Farnell doesn't help this). I didn't see much that was obviously compatible with the 722RA -- certainly not hundreds of PSUs. Can you give me some examples?

cjbe, 2017-09-19:
@jordens I was using a floating supply, and the barrel of the plug was not making contact with the sleeve connection (ground point) on the jack - the jack sleeve connection is electrically isolated from the front panel mounting point in this jack design. I was only seeing grounding through the front panel, via an RJ45 EEM, to the VHDCI breakout.

I tried ~5 different 'standard' 2.1x5.5 plugs I had around the lab, and all had this problem.

jordens, 2017-09-19:
@cjbe How does that work? The L712RA has a 2.5 mm pin. That won't fit the 2.1 mm hole in your connector. The usual problem is 2.1mm pins and 2.5mm plugs. They are just too loose and may not make contact to the outer because it ends up being off-center.

@hartytp I can find both. But apparently you guys apparently have a lot of 2.1 mm in the lab.

* "Mean Well" seems to be mostly 2.1 mm: https://www.reichelt.de/Netzteile-Festspannung/MW-GST60A12/3/index.html?ACTION=3&GROUPID=4950&ARTICLE=169684&OFFSET=100&

* "XP Power" is 2.5 mm: http://www.farnell.com/datasheets/2333577.pdf, http://www.farnell.com/datasheets/2015793.pdf

gkasprow, 2017-09-19:
@jordens @hartytp  it may be not the best idea to let people use random wall wart. Most of cheap Chinese ones do not have any CMC filters nor meet any EMI standards.
It's better to relay on certain verified power supplies which you can easily buy and don't care about connector compatibility. Another thing is power rating. Power supplies are so cheap these days that it makes sense to use overrated 5A 12V adapters in all cases. Low cost ones sometimes don't have any protection and may catch fire when overloaded.

cjbe, 2017-09-19:
@jordens that was a typo - I used plugs with the same ID and OD as specified in the L712RA datasheet, that is 2.5 x 5.5.

cjbe, 2017-09-19:
@gkasprow completely agreed - my concern is that we bought some nice PSUs ([XP Power AFM30US12C2](http://uk.farnell.com/xp-power/afm30us12c2/power-supply-2-5a-12v-30w-iec/dp/2319724)) that looked from the datasheet like they should mate OK, and felt like they mated OK, and powered the device OK. However the only path for the return current was via the rack chassis and another EEMs front panel.

gkasprow, 2017-09-19:
@cjbe you can always replace the plug :) You can also remove some plastic part of the plug using sharp blade to make the ground contact longer.

cjbe, 2017-09-19:
@gkasprow I have already replaced the plug :) - I was just surprised by this failure mode as I had not seen it before, and thought others should be aware of it.

hartytp, 2017-09-19:
> @jordens @hartytp @cjbe @dtcallcock it may be not the best idea to let people use random wall wart. 

> Another thing is power rating. Power supplies are so cheap these days that it makes sense to use overrated 5A 12V adapters in all cases. 

Agreed, let's find a nice, commonly available +12V, 5A power supply and recommend that as the "standard" PSU to use. We should also ship this with Kasli (and potentially with the VHDCI carrier and clocker as well depending on the cost).

> It's better to relay on certain verified power supplies which you can easily buy and don't care about connector compatibility. 

Forcing users to change the connector on power supplies wastes a lot of time. Let's pick a connector that is compatible with high-quality COTS PSUs.

jordens, 2017-09-19:
I'd like to take a common connector (short 2.1 or 2.5 ID barrel, locking) and dialing in on a power supply that we then also ship with Kasli/VHDCI (the locking connectors are so uncommon that the PS cord needs to be modified). There are a bunch of them (XP Power e.g.) with medical approvals.
People are fully able and capable of breaking Kasli anyway.

hartytp, 2017-09-19:
> People are fully able and capable of breaking Kasli anyway.

True, but I'm not sure I follow you in this context...

jordens, 2017-09-19:
That was aimed at "not the best idea to let people use random wall wart": IMHO choosing connectors based on non-technical reasons is misguided.

hartytp, 2017-09-19:
Aaah. Okay.

Sure. Let's just pick a standard PSU that works with this connector and I'm happy. If we can't find a good COTS PSU that mates properly with this connector, that **would** be a good reason to reconsider the connector.

gkasprow, 2017-09-19:
Such quality power adapters with suitable plug can be shipped with VHDCI and Kasli. So this would solve the issue and prevent people from plugging random power source.

dhslichter, 2017-09-19:
I agree with choosing a clean, COTS power supply with no modifications necessary.  However, I think the locking connector is important, and this was the main concern that @dtcallcock was raising.  Thus I like what @jordens is proposing.

hartytp, 2017-09-19:
@dhslichter Are you suggesting bulk buying COTS PSUs and having them modified by e.g. technosystems to add locking connectors?

gkasprow, 2017-09-19:
@hartytp This is the way we would do.

hartytp, 2017-09-19:
How much would that add to the cost of the PSU?

gkasprow, 2017-09-19:
depends on quantity, something like 2EUR@100pcs

hartytp, 2017-09-19:
Okay, that's fine then.

sbourdeauducq, 2017-09-20:
> However the only path for the return current was via the rack chassis and another EEMs front panel.

:) :) :)

Those barrel connectors are generally a bit crappy like this and unprofessional. They also cause short circuits sometimes.
On ionpak we use [Phoenix Contact 1746978](https://www.digikey.com/product-detail/en/phoenix-contact/1746978/277-6765-ND/2511356) and its associated [terminal block plug](https://www.digikey.com/product-detail/en/phoenix-contact/1765551/277-8689-ND/3605987). It's easy to cut the cable of the PSU, screw it in there, and then this sort of shenanigan doesn't happen.

jordens, 2017-09-20:
Those don't have bend protection, or strain relief, are much larger, and are non locking.

gkasprow, 2017-09-20:
For my MTCA cassettes I use another locking [DC DIN](http://pl.mouser.com/ProductDetail/Kycon/KPJX-4S-S/?qs=%2fha2pyFaduiEsOXZH1hheQ5K0vEgCjGZ5IjeCQdzNWsSTkvOPjVxYA%3d%3d&utm_source=octopart&utm_medium=aggregator&utm_campaign=806-KPJX-4S-S&utm_content=Kycon) connectors. Farnell has PSUs with such connectors on stock.
And they sustain 10A easily.

hartytp, 2017-09-20:
It is a good point: barrel connectors aren't great mechanically, so we could consider something else.

Do we have enough room on, e.g. the Kasli/Clocker front panels for that Kind of DIN connector?

jordens, 2017-09-20:
I would usually stay away from those round small DIN connectors for their tricky assembly, general flimsiness, and non-locking. But maybe those are OK. What would you recommend @gkasprow the locking barrel connectors above (short version) or those?

gkasprow, 2017-09-20:
These are locking ones. One have to pull the shell to unlock.

gkasprow, 2017-09-20:
@jordens This [one](http://www.mouser.com/ds/2/222/KPPX-365228.pdf) I use and it is locking

dhslichter, 2017-09-20:
At the cost quoted I think we're in good shape, and I think the connectors @gkasprow is proposing seem fine, if he has good experience with them.  I tend to favor a screw lock over a pull lock, all else being equal, but we probably need a relatively small form factor.  

Getting away from barrel connectors seems like a good idea for a number of reasons, including making people think a little carefully before putting a power supply on, in addition to all the mechanical/electrical reliability issues.

gkasprow, 2017-09-20:
The DC DIN jack is much bigger, takes roughly twice more space on PCB than barrel type DC jack.
So may be an issue with Kasli where space is limited. And that's true - assembly of the plug is problematic. Once you assemble, it is very hard to dismantle without breaking it.
In addition, we wouldn't be able to offer assembly variants with such DIN DC connectors on rear side of EEM boards. The only argument is that some PSUs already have such plug and it is rated for 15A.

jordens, 2017-09-20:
Weighing the arguments, I'm leaning towards sticking with the 712RA 5.5mm/2.5mm/shirt bushing locking barrel (or 722RA if someone convinces me that there is a majority of 2.1mm ID barrels out there).

cjbe, 2017-09-20:
The annotation on the PCB and front panel regarding upper and lower VHDCI connections is swapped around:

The lower VHDCI connector (front panel label EEM 5..8) connects to headers labelled LVDS1-4, which are labelled 'Upper VHDCI'.

The upper VHDCI connector (front panel label EEM 1..4) connects to headers labelled LVDS5-8, which are labelled 'Lower VHDCI'.

gkasprow, 2017-09-20:
Thanks @cjbe As itermediate fix we will glue sticker on the PCB.

jbqubit, 2017-09-20:
Start channel count at 0. Use same nomenclature for front panel labels, headers on board: "EEM 0..3" not LVDSxx or Upper/Lower VHDCI.

dhslichter, 2017-09-20:
@jordens we use the 712RA on our TTL breakout boxes and it seems to work well.  Soldering on the mating connector to a power supply is very straightforward.  The connection is robust and simple.  The only downside is the temptation for users to use their own wall wart.

gkasprow, 2017-09-20:
@dhslichter we can always isolate the power connector :)

dhslichter, 2017-09-20:
The placebo power connector!  I love it :)

gkasprow, 2017-09-20:
I mean insulation washer between panel and connector ground. In this way not appropriate power plug won't make contact and won't supply the board.

dhslichter, 2017-09-21:
OK, I see :)  Also a good idea!

gkasprow, 2018-08-06:
@cjbe To isolate the power connector from the panel we will probably need existing, long barrel (L712RA) one. We want to enable connection of the panels with board ground using 0R resistors. Since the power connector barrel is before the CMC, we want to isolate it.

gkasprow, 2018-08-06:
@hartytp @jordens I'm going to leave existing connector and add insulating washers. Want to finish new release today.

hartytp, 2018-08-06:
I'm happy with that. Let's just make sure it's correctly documented so that anyone who wants to buy the locking connector buys the right one. Any objections @cjbe

jordens, 2018-08-06:
I am OK with the long one.

gkasprow, 2018-08-06:
What about adding default VHDCI cables to the BOM? Wouldn't it be better to order them separately, i.e. from same company who sells VHDCI carriers?

gkasprow, 2018-08-06:
@hartytp whad exactly do you mean as "Enable this to be used as a general purpose IDC <-> VHDCI adapter"
Do you want the I2C to be routed in a way that Kasli can control/detect remote devices?

jordens, 2018-08-06:
Adding them to the BOM would be OK if there was no question of quantity (1 or 2?) and length (1, 3, 5, 10m?). And it's the same problem as with the HD68 cables on Zotino. In retrospect it may have been better to go for VHDCI there as well to reduce variation.

gkasprow, 2018-08-06:
So  let's leave it as it is now.

hartytp, 2018-08-06:
> What about adding default VHDCI cables to the BOM? Wouldn't it be better to order them separately, i.e. from same company who sells VHDCI carriers?

I'd prefer not to ship the VHDCI cables by default, as everyone will want different lengths and they're not that cheap. If a supplier could stock the VHDCI cables in a few lengths (we use 10m in a few cases) that would be great.

> In retrospect it may have been better to go for VHDCI there as well to reduce variation.

I'm happy with HD68 on Zotino. VHDCI is a pain to hand solder and work with. HD68 is also much easier. Also, we felt at the time that it was good to use different connectors to avoid users plugging the wrong connector into the wrong EEM (particularly when Zotino had +-12V on its HD68).

hartytp, 2018-08-06:
> @hartytp what exactly do you mean as "Enable this to be used as a general purpose IDC <-> VHDCI adapter"
> Do you want the I2C to be routed in a way that Kasli can control/detect remote devices?

Essentially, yes.

We want to be able to connect: Kasli -> IDC -> VHDCI_CARRIER -> VHDCI cable -> VHDCI_CARRIER -> IDC -> EEM and have it behave as if we just had Kasli -> IDC -> EEM.

hartytp, 2018-08-06:
We use this quite a bit, for example in our Sampler-Kasli-Urukul servo to place Sampler next to photodiodes and Urukul next to AOMs, which are on a different optics table.

closed 2018-08-06 by gkasprow