sinara-hw/sinara #235: EEMs: VHDCI breakout, BNC, SMA, RJ45 documentation check

jbqubit, 2017-07-16:
New Issue created from #221. 

https://github.com/m-labs/sinara/wiki/EEM
- "Extension modules connect to the Carrier via an IDC ribbon cable (2x15 100 mil). Cable pitch is 1.27mm, connector pitch is 2.54."
    - shouldn't this read "2x15 with 100 mil pin-to-pin spacing"?
    - what is cable pitch? 
- "pin header" 
    - state LVDS voltage
    - rename "comment" column to "default use"
    - add new column for "pin designation if used for SPI"
- "EEM Backplane" 
    - Since EEM_backplane is still a draft, consider creating a new page called EEM_backplane and moving backplane-related content to it. 

https://github.com/m-labs/sinara/wiki/EEM_BNC
- State maximum frequency for digital wavetrain with 50% duty cycle. WUT tested this. 

https://github.com/m-labs/sinara/wiki/EEM_SMA
- "Extension module supplying 16 digital IOs via front-panel SMAs." should be 8 digital IOs
- State maximum frequency for digital wavetrain with 50% duty cycle. WUT tested this. 

https://github.com/m-labs/sinara/wiki/EEM_RJ45_LVDS
- "This board connects to up to two Kasli extension headers." These are "EEM headers" not "Kasli headers" as they're used by both Sinara and Kasli. 
- State maximum frequency for digital wavetrain with 50% duty cycle. WUT tested this. 
- [ ] comment on short-circuit protection
- Image is old... should show two EEMs and 
- link in "Design files are located here." is to SMA not RJ45_LVDS. 
- missing table showing pinout for each RJ45 and pin assignment if used for SPI

gkasprow, 2017-07-17:
@jbqubit @hartytp I updated wiki.

jbqubit, 2017-07-17:
@gkasprow Awesome! Looks great! PCB photos show nice LEDs too. :)

closed 2017-07-17 by jbqubit