sinara-hw/sinara #237: ordering 3U VHDCI_breakout, SMA, BNC, RJ45

jbqubit, 2017-07-19:
Move discussion started in #221 here. I've updated wiki so R1 corresponds to only the PCBs ready for ordering now. I propose teams update R1 quantities in the wiki. 
https://github.com/m-labs/sinara/wiki/Purchasing

jbqubit, 2017-07-19:
@hartytp said in #221... 

@gkasprow Can you confirm a few things:

- Have you already placed the order? Or, is there still time for other people to add to this order?
- What is the cost of these boards?
- What is the lead time/manufacturing time?
- What is the ordering process? Does everyone have to give Technosystem a PO/payment before the manufacture can begin? Or, can people confirm quantities with you and sort payment out later? We can now place orders with Technosystem within 24 hours, but the first order we placed took us a couple of weeks as they had to be added to Oxford's "suppliers database", which is a pretty tortuous process. I want to make sure we don't end up with large delays due to having to get multiple Universities' finance departments to all do the paperwork before the manufacturing can start.

Other than that, I'd be happy to set a batch size freeze on Monday 9am GMT. 

@gkasprow One other thing I forgot to ask: will these boards come with proper, fully labelled front panels?

- It'd be nice to have the name of the PCB on each panel (e.g.BNC DIO, RJ45 DIO, SMA DIO, VHDCI EEM Carrier)
- It'd be nice to have channel numbers labelled
- If there is room, it would be nice to have the function of the LEDs labelled (e.g. "red=input", "termination" or wherever is appropriate)
- Ideally, but not necessary, it'd be nice to have an ARTIQ logo somewhere if there is space

It's not a huge problem if this isn't done, as we'll just do the labelling ourselves, but it would be nice.

jbqubit, 2017-07-19:
In the USA most of us are setup to place small value orders via credit card. @gkasprow Please confirm that technosystem is setup for this.

gkasprow, 2017-07-19:
@jbqubit I got info that they are working on it.

gkasprow, 2017-07-19:
@jbqubit All boards are produced in such way that PCB are printed in much higher volume (50 or 100 pcs) and boards are manufactured with some margin, let's say 10 pieces and kept on stock. Once the stock is low, another batch is assembled. So unless you need massive amount of them, a few pieces will be always available immediately. The longest lead time is for bare PCB, then assembly takes one week (providing parts availability) + a few days of tests. We do tests after batch production so single pieces, providing they are on stock, are shipped next day.
I ordered aluminium panels and will make labels using laser. Depends on result, I will keep on doing this way or order much more expensive panels from Schroff, with long lead time. Before labelling, I will ask you for panel graphics approval.

hartytp, 2017-07-19:
@gkasprow Thanks for the update, that all sounds good. When do you think the first PCBs will be delivered to us?

@jordens Do you want to send something around the ARTIQ list to inform people about this purchasing round?

jbqubit, 2017-07-19:
@gkasprow If I understand correctly... The implication of your approach is that there's no need for Sinara users to get super-organized and place advance orders for these inexpensive boards. If that's the case please provide official technosystem part numbers and ordering information. Then individual labs can order any time.

gkasprow, 2017-07-19:
@jbqubit that's true for simple boards.
For AMC/RTM this needs to be somehow organized in advance because Creotech must reserve manpower and assebly line.

hartytp, 2017-07-19:
@gkasprow I'm also a little confused about the ordering process for these EEMs? Do we email Creotech to get a quote/place the order? If so, what part number should we use?

As we've never ordered from Creotech, we'll need a formal quote to place our initial order...

gkasprow, 2017-07-19:
I'm organising it in such way that you could order it easily. So PN will be specified with assembly variants (i.e. URUKUL). We have such system for other open hardware products so don't worry. At the beginning you will simply send email but later on we will think of some web-shop.
There are many other details like VHDCI cables, SMP bullets, various cables and adapters that would be nice to have in one place.

hartytp, 2017-07-19:
@gkasprow That sounds great! Thanks!

Has Oxford's order for these EEMs already been logged? Or, do I need to email the quantities we'd like to someone before I get the quote?

gkasprow, 2017-07-19:
@hartytp Batch or PCBs are ordered, give me a few days and I will provide all info.

hartytp, 2017-07-19:
@gkasprow Perfect, thanks for the update. I'll wait to hear from you.

Note that, as I mentioned before, due to the bureaucracy here, it would help to get the quote soon so that I can begin the paperwork process...

jordens, 2017-07-20:
I was also thinking of new board revisions. If people need/want those, the lead time goes up and there needs to be organizing, and old stock needs to be cleared.
Otherwise greg's approach sounds fine to me. The numbers in the wiki then at least provide a good indicator for technosystem/creotech to size their batches and stock.

gkasprow, 2017-07-23:
@jbqubit  @hartytp Front panels are [here](https://github.com/m-labs/sinara/blob/master/ARTIQ_ALTIUM/PANELS/Project%20Outputs%20for%20PANELS/Panels_3Dview.PDF)
Looking for comments.

jbqubit, 2017-07-25:
> Batch of PCBs is ordered, give me a few days and I will provide all info.

Great! But this Issue got closed prematurely. Outstanding questions include the following. 

- [ ] What is the cost of these boards?
- [ ] What is the ordering process?
- [ ] When do you think the first PCBs will be delivered to us?
- [ ] What part number should we use?
- [ ] Update https://github.com/m-labs/sinara/wiki/Purchasing to reflect the purchasing workflow once it's been worked out.

jordens, 2017-08-10:
https://ssl.serverraum.org/lists-archive/artiq/2017-August/001133.html

hartytp, 2017-08-10:
Closing this issue, as nothing to do here.

Plan is to update the Wiki with this info.

Greg wants to finish sorting out a few details before we do this.

jbqubit, 2017-08-31:
Reopening since part numbers are not yet on wiki. Vendor does not get accept credit card purchases.

hartytp, 2017-08-31:
> Vendor does not get accept credit card purchases.

Not even via PayPal?

closed 2017-09-18 by hartytp