sinara-hw/sinara #238: EEM front panels

hartytp, 2017-07-24:
Comments of @gkasprow front panel designs. They look good!

- [ ] Please use EEM code names where possible. Let's keep the amount of text on the front panel to a minimum so it doesn't get over crowded. e.g. use "Zotino" or "Zotino 32ch DAC" rather than "Zotino 32ch 16-bit 1MSPS DAC".
- [ ] Fix ARTIQ logo
- [ ] For Zotino temperature LEDs, can you try to adjust the spacing so that HTR and OK are spaced further apart (otherwise, they look like a single word!). Maybe use two letters to indicate that the heater is on (HT)?
- [ ] For Zotino LEDS, what do MODE and STAT mean? (They're not called that on the schematic). If these LEDs don't have a fixed purpose, can we just label them USER LEDs or similar?
- FWIW, I would have preferred to keep everything zero-indexed. However, the front panels should match the schematics, so maybe it's too late to change that for these designs?
- It's a shame that the LEDs on the LVDS DIO EEM aren't in the same order as the RJ45 connectors? I wish we'd spotted that before we sent the boards off to manufacture. It's not a disaster, but it could be very confusing for the users!
- [ ] on the VHDCI carrier should that be "EEM 1...4" rather than "LVDS 1...4"?
- On the SMA DIO board, it's a shamge that the outputs aren't ordered in a more logical fashion (e.g. numbers monotonically decreasing along the length of the board)! Again, I wish we'd spotted this before going to manufacture!

dhslichter, 2017-07-24:
Adding on to (with some duplication) what @hartytp has listed:

- [ ] I know this may cause consternation, but I would strongly push for nomenclature of "Zotino 32ch DAC" instead of just "Zotino". 
- [ ] It might be helpful to have a small box around shared banks (e.g. on BNC or SMA IO) to give a visual reminder that channels are distributed into banks which have a shared direction, for example. 
- [ ] On TTL IO tester with IDC, what does "outputs enabled" mean?  Is this a direction indicator?  If so, I would call it "direction".
- [ ] Some things which should perhaps be fixed on the subsequent board designs:
    - [ ] Fix numbering of channels to be zero-indexed on board schematics
    - [ ] Fix ordering of LEDs on LVDS DIO EEM
    - [ ] On the BNC IO EEM, I would move the two banks of 4 channels apart a bit more to give the visual indication of two banks, where direction is shared within the bank.
    - [ ] On SMA IO, I would move the bank direction LEDs to the end (rather than in the middle)
    - [ ] Change ordering of SMA outputs on SMA IO to be monotonically increasing.

jbqubit, 2017-07-24:
[Link](https://github.com/m-labs/sinara/tree/master/ARTIQ_ALTIUM/PANELS) to panels PDF.

hartytp, 2017-07-24:
@dhslichter Some of the points you raise/reiterate would require a new revision of the PCBs rather than just changes to the front panels, so I've moved them to new issues.

hartytp, 2017-07-24:
Should we put a hardware revision on the front panels (e.g. bottom right)?

jbqubit, 2017-07-24:
>Should we put a hardware revision on the front panels (e.g. bottom right)?

Revision number on PCB is sufficient. That way there's no need to print a new front panel for minor PCB changes. Besides, front panel lead time is >> PCB lead time.

hartytp, 2017-07-24:
@jbqubit IME, it's always a good idea to have the hardware revision on the FP. Otherwise, one gets into a situation where there are several different revisions of a PCB in identical-looking enclosures, but all with slightly different functionality. This can easily cause hard to debug issues in the lab. In general, if a change is worth printing a whole new batch of PCBs, it's probably big enough to bump the version number on the FP.

The front panels should be fairly cheap, so IMO this is not a good place to try to save money. Also, AFAICT, the lead time for the panels isn't >> the month it takes to print and stuff the boards. And, in any case, if you really want to keep the same FP with a new PCB revision you can always put a sticker on it!

@gkasprow Even if we don't do it for the other boards, I'd like to add the hardware revision on the FP for Novogorny and Zotino.

dhslichter, 2017-07-24:
I agree with @hartytp on this one.  You don't want to have to pull cards to figure out what revision the hardware is inside.  And, as pointed out, it's generally the case that if you make revisions to the PCB, you will have to be making changes to the front panel too, so it's not like this hurts us in most instances.

jbqubit, 2017-07-24:
Let's go with version numbers on the FP then.

gkasprow, 2017-07-24:
Front panels are not printed so we can place HW revision as well.

hartytp, 2017-08-20:
@gkasprow Have all items here been completed for the first batch of front panels that are being produced? If so, can you close this issue, please?

gkasprow, 2017-08-29:
I updated panel designs - check pdf

hartytp, 2017-08-29:
They all look good to me! Thanks for sorting that.

jordens, 2017-08-29:
- [ ] The channel numbering on the panels is still not zero-indexed (apart from the user leds on Zotino).

hartytp, 2017-08-29:
@jordens For the rev1 boards, the silk screen isn't zero-indexed. So, do we want to keep the panels consistent with the silk and fix this in the next revision? Or, just fix it on the panels now and let them be inconsistent with the silk (could cause mild confusion when setting DIP switches etc)? I'm not fussed either way so long as it does get fixed in the next revision...

jordens, 2017-08-29:
Well. If this is "rev1", then the panel label "rev 1.1" is wrong.
If everybody would just stick to the versioning scheme that was originally suggested... Otherwise whoever prefers other versioning and naming schemes needs to step up, work on a consensus, execute the vote, and clean up issues, milestones, and the wiki, pcb layouts, schematics.

Also IMHO it is more important to fix the panels than the silkscreen. Important enough to have the silkscreen be out of sync and slap a sticker on the pcb.

hartytp, 2017-08-29:
> Well. If this is "rev1", then the panel label "rev 1.1" is wrong.
If everybody would just stick to the versioning scheme that was originally suggested... Otherwise whoever prefers other versioning and naming schemes needs to step up, work on a consensus, execute the vote, and clean up issues, milestones, and the wiki, pcb layouts, schematics.

Sorry, yes, these are Rev1.1 my mistake. Can you remind me what versioning scheme you favour?

> Also IMHO it is more important to fix the panels than the silkscreen. Important enough to have the silkscreen be out of sync and slap a sticker on the pcb.

Fine by me. Greg, do you mind fixing this?

jordens, 2017-08-30:
From the [original plans](https://github.com/m-labs/sinara/milestones) we had 0.1 and 0.2 as the two prototype rounds and then 1.x as the sequence of first production rounds. I don't really care how we do it as long as it is consistent.

* Front panels, PCBs, purchase orders, wiki pages, issues, milestones, schematics all need to match.
* The releases also need tagging in git.
* If we do reviews, then we need to do them on well-defined git hashes and not with moving targets.

jbqubit, 2017-08-31:
I agree with @jordens. @gkasprow Somebody on your team is best positioned to do this. Please assign.

hartytp, 2017-09-03:
@jbqubit What is the action you want completed before this issue can be closed? 

If it's about tagging releases properly, then let's open a new issue for that.

closed 2017-09-04 by hartytp