sinara-hw/sinara #240: SMA DIO Errata

hartytp, 2017-07-24:
- [ ] Change ordering of SMA outputs on to be monotonically increasing.
- [ ] Space permitting, move the bank direction LEDs to the end (rather than in the middle). If this will reduce the pitch between adjacent SMAs too much (i.e. to the point where they get awkward to do up) then don't worry about it.

jordens, 2017-09-18:
* [ ] SMA clearance to panel is too small
* [ ] SMA connector clearance to mounting brackets and screws is too small

jordens, 2017-09-27:
@gkasprow Also, from my boards, only the SMA boards were shipped with corrugated washers under those screws (PCB to mounting angle block). That makes the underside of the PCB thicker than allowed and collides with the left adjacent front panel. Don't know from the top of my head whether these are supposed to use washers but it needs to be (a) consistent among all boards and (b) be below the ~2.27 mm allowed max protrusion on that side.

gkasprow, 2017-09-27:
well, such washers were probably attached to Schroff mounting kit. The [manual](https://schroff.pentair.com/wcsstore/ExtendedSitesCatalogAssetStore/PentairDataLoad/EPSMedia/ASIM/webcat/subgroup/mi/60817-182.pdf) says they should be there. Probably I used wrong ones. The kit contains much more components than are needed :)

jordens, 2017-09-28:
Also what does the QC sticker actually mean? On my 3U_DIO_RJ45 two pins of the RJ45 cage were not soldered.

hartytp, 2017-10-12:
@gkasprow Should this have the same PSU as the BNC DIO board? Wasn't there a reliability issue with the SMPS you're using on this board?

gkasprow, 2017-10-12:
I replaced these DC/DC converters with different type ones on the boards that were shipped. 
SMA DIO design was already modified to support Murata converter

hartytp, 2017-10-12:
@gkasprow Okay, good.

I just asked because I saw the photos that Robert posted https://github.com/m-labs/sinara/wiki/EEM_SMA which have the old converter. But, maybe this was just an older revision of the pcb...

gkasprow, 2017-10-12:
@hartytp It looks like old one but model is different. I installed one with twice higher current.

jordens, 2017-10-13:
Even with the more powerful ones, the claim that the "outputs can supply 5V into 25Ohm" is at the very least misleading, because that would be 800 mA per 5V rail. And "tolerate an indefinite short-circuit to ground" only makes sense if it includes the DC/DC converters shutting down. Could someone clarify what is actually meant by those statements?
And why did technosystem distribute rev1.0? Should I send them back?

hartytp, 2017-10-13:
@gkasprow Thanks for the update! But, I'm confused, should these have the Murata converter or the higher current rated Traco power one?

hartytp, 2017-10-13:
@gkasprow PS, not sure if I already said this, but: I set a rack up in my lab like the one Robert posted photos of. Modulo the small bugs we've posted, it all works really nicely and looks great! Thanks again for all the great work you guys have done on this.

gkasprow, 2017-10-13:
The SMA version is v1 but with higher power dc/dc converter. Murata is same power.

gkasprow, 2018-01-13:
@jordens do you insist on adding supporting insulating washers for SMA connectors? They conflict with LEDs... I can place them from inner side after cutting their edges, but on the outer side they will cover LEDs.

jordens, 2018-01-13:
I don't insist. My conclusion is that the connector spacing is too small and that we need to go to 8 HP and a pattern similar to DIO_BNC. But I don't think we can do this in the resent revision.

gkasprow, 2018-01-13:
@jordens going to 8HP involves adding vertical SMA connectors and additional PCB mounted vertically.
So let's leave it as it is.

jordens, 2018-01-14:
@gkasprow or pigtails and panel mounted connectors. or even some non-standard mechanical design where the angled connectors are on both sides. ack, not something for now.

gkasprow, 2018-01-14:
done.

hartytp, 2018-01-15:
@gkasprow If the boards have gone to manufacture, are all items at the top of this issue done? If so, can we close the issue?

closed 2018-01-15 by gkasprow