sinara-hw/sinara #241: BNC DIO Errata

hartytp, 2017-07-24:
- [ ] Move the two banks of 4 channels apart a bit more to give the visual indication of two banks, where direction is shared within the bank.

hartytp, 2017-08-19:
- [ ] Add 100k resistors between the isolated grounds and the main circuit ground. This will prevent the boards from charging up, which could damage sensitive instruments connected to them. 

To do: add this to the SMA board list as well!

jordens, 2017-09-18:
- [ ] The rack extraction handle should be 8 HP wide (not 4). There is an alignment hole in the panel.

jordens, 2017-09-29:
I am confused about the connections between IC28, IC22, IC29.

1. When set to input (DIR_CTRL* high, IC22:A5 low, DIR*_RD low, IC29:DIR high) isn't IC28:B* driving the same signals against IC29:B*? I.e. don't you want to drive IC28:EN2 with IC22:A5?
2. For completeness of the handshaking that you have started to build in, don't you want to use IC28:A5/DIR1_RD (inverted) to drive IC21:*DE to turn on the LVDS driver as late as possible?
3. Also don't you want to drive IC28:EN2 using IC28:A5/DIR1_RD (inverted) as well to save some power?
4. Finally on IC13, can we just use one pin for DIR_CTRL1 and DIR_CTRL2 each? Do we need to drive them so heavily that we need four? When using just two pins, the rest could be used for e.g. DIR[12]_RD monitoring to double check the switches are not wrong and overriding and to double check that there is at least some indirect indication that the power supplies might be working working.

gkasprow, 2017-09-29:
@jordens 
1. IC29 is assymetrical, A ports has higher load than port B. That's why we need to invert the signal. Of course we could get rid of T27 and use DIR_CTRL directly. Such scheme was evolution of yet another primary idea where direction of each channel could be changed independently.
2. This could be done. It can be also used to drive LEDs to use them as indicator or correct isolator operation.
3. Sure, let's do it.
4. I wanted to keep compatibility with other IDC boards. Each pin is direction of corresponding pin. In this way when you plug DIO TTL, DIO SMA, DIO LVDS, just read this register to know port directions.
We can of course change it and implement more advanced diagnostics. With such loop we will check DC/DC converters operation indirectly.

jordens, 2017-09-29:
@gkasprow thanks.

1. I don't understand your comment. My concern is about IC28 and IC29 both being enabled (each driving their B port) and driving the same copper (e.g. IC29:14 and IC29:15 driving against IC28:15) at the same time when the set of channels is configured as input. This won't happen for long since ultimately they will drive the same value. But it will happen transiently when switching between input and output. And it will exceed absolute max ratings. Better to turn of the unused coupler (disable IC28 like in (3))?
4. Ok. That makes sense. Let's leave it.

gkasprow, 2017-09-29:
@jordens OK, I got it. This needs fix in next revision. Disabling of IC28 would do the job. One can also do a quick fix lifting and connecting pin 10 with pin 11 of IC28/IC11.

jordens, 2017-09-29:
Right.
It will also happen transiently for a about 10+5+5+10 = 30 ns each time the input changes (explanation above). This is a bit worrying.

gkasprow, 2017-09-29:
Fortunately the buffer has much higher driving current so will dominate the
isolator chip. I didn't notice anything suspicious when I looked with
scope. I will have a look one again to see if operation of existing boards
is safe.

hartytp, 2017-10-12:
@cjbe IIRC, you had a noise issue with this board. Can you remind me what it was?

The data sheet for the PSUs say the noise is 85mVpk-pk, but recommend adding an inductor to reduce it to 5mVpk-pk if desired.

gkasprow, 2017-10-12:
@hartytp this value is under assumption that converter is fully loaded which is not the case. Anyway, if we want to have clean TTL signal, adding of LC filter will solve the problem. One has only make sure to use high ESR tantalium capacitors in parallel with ceramic ones  to damp oscillations of such LC filter.

hartytp, 2017-10-12:
@gkasprow Okay. Maybe it's a bit worse under very light loads? IIRC @cjbe mentioned seeing >200mV of ripple on his digital signals, which was a bit annoying.

Assuming the LC filter is small and cheap, it's probably worth adding it for the next revision IMO.

gkasprow, 2017-10-12:
@hartytp I did not observe such problem. There are plenty of measurements done within #187 and ripples are much smaller. @cjbe how did you connect the ground of the scope?

cjbe, 2017-10-24:
@gkasprow I am using a BNC DIO connected to a VHDCI breakout, powered by a floating PSU. Hence the only grounded point in this system is the scope used to measure the BNC outputs.

With one output connected to a BNC output at logic high, and no termination on the scope, I see ~170 mVpp: [image]

With one bank of outputs (4) all terminated and at logic high, I see something pretty similar: [image]

gkasprow, 2017-10-24:
@cjbe Let me have a look in my lab. You can also do quick check and measure the noise induced on the probe. Just connect the scope probe ground clip to the BNC GND ring and touch the probe tip to the same BNC GND ring.

gkasprow, 2017-10-26:
@cjbe You are right, I looked carefully with termination on and there is an issue.
[image]
It appears only when termination is on on the output.
One can easily fix it by adding 47uF capacitor at the DC/DC converter output.
With this fix it looks much better
[image]
here one can add caps. I added 2x 22uF 0805 but  47uF or even 100uF 1206 or 1210 would be better
Cap voltage - the higher the better because effective capacitance would be bigger.
[image]

gkasprow, 2017-10-26:
There are three pads. Gnd is on the most left, vcc is on the most right. I shorted middle one with the most right one. WIth 1210 cap one can solder it directly between left and right pads.

gkasprow, 2017-11-02:
I've noticed interesting feature of DIO board.
During live switching of port directions with generator connected to the output the generator beeps with warning because it discovered another driven signal instead of the load and disconnects the output.
But the driver IC starts heating a lot probably due to latchup effect. So one should consider adding overcurrent detector to next board revision.
With current revision please make sure to disconnect power supply during changing direction of the ports.

hartytp, 2017-11-02:
@gkasprow I'm fine if we just ban live switching of directions. But, we should add a clear note on the schematic/silk.

jordens, 2017-11-02:
IMHO we should also consider putting a 50R series resistor between the output driver and the switchable termination. It seems bot unnecessary and unrealsitic to drive eight times 5V into 50R on board and 50R remote in parallel. Having the additional series 50R would make the output always be 50R impedance (with the temrination disabled) and the inputs would be either high impedance or 50R (parallel termination). It would also reduce worries about power supplies, ripple, and EEM connector load.

gkasprow, 2017-11-02:
@jordens good idea. it should be smaller than 50Ohm due to internal output resistance of the driver.

jordens, 2017-11-02:
Yesh. Make it ~30R.

gkasprow, 2018-01-10:
@jordens I released v1.2 of DIO-BNC.
Please have a look.
Main difference is power supply filtering and delayed buffer enable.

hartytp, 2018-01-11:
@gkasprow This all looks good to me! Thanks.

The schematic is a bit cramped, which is a symptom of how the complexity of this circuit has increased over time :(. Otherwise, looks good.

Only one thing I would potentially like to change: the way the direction control is done. It would be nice if the PCF8574ADW could also be used for direction control (as well as readout). That way one can do all configuration from Kasli if desired and doesn't have to set DIP switches.

To do that, it's important that the BNCs all start off as inputs until the PCF8574ADW is programmed. However, in the current design, if the switch is left open (so that the PCF8574ADW can control the direction) the BNCs all start as outputs until they are programmed. This can cause damage to equipment. Could we change it so that all channels start as inputs, and shorting either the switches or the PCF8574ADW changes them to outputs?

A couple of minor comments:
- If there is room, can you add a short note that the reset ICs prevent glitches while the isolators are starting up (otherwise, it's not clear why they are there)
- If there is room on the schematic can you add a note stating that grounding/closing the switches sets the channels to inputs. Or, would it be easier just to add some 0201 20R resistors to protect the ICs in case of live switching? That way we can just remove the warning.

hartytp, 2018-01-11:
@gkasprow One other question: what's the power rating out those 30R output resistors? For this kind of lab equipment, it's very nice if the outputs can be shorted to ground indefinitely, so they should be able to handle that much current if possible.

gkasprow, 2018-01-11:
@hartytp the PCF chip CAN do direction control. DIP switches must be off.
PCF chip is open drain, weak pullup (aka 8051 IO). So it can read and short to ground.

gkasprow, 2018-01-11:
@hartytp where shall I add these resistors? I'm not sure why these isolators are killed while changing direction...

hartytp, 2018-01-11:
@gkasprow Right, but that wasn't quite my point. To do direction control with the PCF chip, one has to have the DIP switch off, as you say. That means that all channels begin as outputs until the PCF chip is programmed. That can cause damage to a load. It would be better to have it so that when the DIP switches are off all channels start as inputs. Then either closing the DIP switches or programming the PCF chip can turn them into outputs.

Just a matter of changing the default state at power on to make things safe.

gkasprow, 2018-01-11:
@hartytp OK, good point.

hartytp, 2018-01-11:
> I'm not sure why these isolators are killed while changing direction...

Okay, so have you actually seen an isolator dying? I wasn't sure if this is a hypothetical issue or a practical one.

I was thinking of putting them between IC28 and IC29. I can imagine both chips transiently being set as outputs at the same time when the direction switch is set. That could potentially damage the isolators.

But, it might be worth re-reading the isolator datasheet to check for other potential sources of damage. 

It's better to make the board robust than to rely on the user seeing a warning printed on the underside of the PCB!

gkasprow, 2018-01-11:
@hartytp I saw it dying but live direction change was only my suspicion. The 30R resistors are 0.25W ones (0805) The ICs have internal 20R resistors so the max voltage drop would be 3V which will damage them.
I will use 0.5W ones. Good point.

hartytp, 2018-01-11:
> I will use 0.5W ones. Good point.

Good! It's a bit ugly to have to use 0.5W resistors for this, but better than having dead boards when people accidentally short channels.

> @hartytp I saw it dying but live direction change was only my suspicion. 

Well, the isolation resistors might fix that. What do you think?

gkasprow, 2018-01-11:
OK, 20R won't harm :)

hartytp, 2018-01-11:
:) Although, I still find it a bit surprising how complicated this board has become for a simple TTL breakout!

gkasprow, 2018-01-11:
I also moved the power supply section to another sheet.

hartytp, 2018-01-11:
If we do that, please can we remove the warning about live switching potentially causing damage (both from the schematic and silk)?

This should work! And, I don't think those warnings will necessarily prevent people from doing it (particularly since the warning is on the wrong side of the PCB).

Also, we will plan to do live switching in practice if we use the PCF chip to change the direction. So, I'd prefer to just test this carefully and make sure it works. If it still happens with the isolation resistors then I'll take a closer look at the PCB.

gkasprow, 2018-01-11:
@hartytp at the moment, when switches are off, the board is in input state. There are 2 inverters :)
So we don't have to change anything, just add protection resistors.

hartytp, 2018-01-11:
> at the moment, when switches are off, the board is in input state. There are 2 inverters :)
So we don't have to change anything, just add protection resistors.

Apologies: you're right about that, I missed an inverter. The current design is correct.

I missed an inverter. Would be good too put a couple of extra schematic annotations:
- switches open means inputs
- switches should be open to allow direction control by I2C-- note that current annotation incorrectly states that I2C can be used for direction readout only (or, at least, it's meaning is a bit ambiguous).

gkasprow, 2018-01-11:
@hartytp I commited DIO-BNC with recent changes.

hartytp, 2018-01-11:
Looks good!

a-shafir, 2018-01-11:
What is the maximum expected voltage drop over the isolation?
The isolation components itself are good up to 3KV.
The layout gap is not as wide as can be but certainly few hundreds V.
Looks like the bottom-line will be R18/R25. It is 2012 package so 200V max according the common specs. But the value and package/power limit will be ~110V DC/RMS.
I think it makes sense to use high voltage resistors like Stackpole RNV and also increase it to 220K or 470K.

hartytp, 2018-01-11:
I'd prefer not to do that @a-shafir. As far as I'm aware, the only intended use for this design is breaking ground loops, where it's expected that the potential difference across the isolation barrier is a volt or less.

Designing for kV is complete overkill for the kinds of uses this board will have.

gkasprow, 2018-01-11:
For kV applications, one need to have different mechanical construction, it's not only PCB that matters.

hartytp, 2018-01-12:
@gkasprow One tiny nitpick. I noticed that the annotation about the EEM connector specifies the maximum current as being 500mA. That's outdated/incorrect isn't it? IIRC, the max current for the EEM +12V is 2A.

One final thing: this board doesn't need a complete power budget, but it would be good to give users a rough estimate of the max +12V current draw when all channels are terminated and driving 50R. This number is extremely useful for users who want to make sure they don't exceed the maximum current Kasli can provide.

gkasprow, 2018-01-12:
If we assume that 8 outputs are shorted with 50Ohm, this gives 800mA@5V. DC/DC converters are rated for 2W each so will survive. They have roughly 70% efficiency, so this adds about 2W to the bill. Current consumption of remaining logics depends on signal toggle ratio but total power consumption should not exceed 7W in the worst case.

jordens, 2018-01-12:
@hartytp please make sure that the changes to the direction logic also work if the FPGA is unaware. I.e. if it always uses output channels as outputs as it does now.

jordens, 2018-01-12:
@hartyp To elaborate: there is also the direction of the FPGA pin playing into this. If you have all channels as inputs by default then the FPGA will see complete LVDS bus contention until the channels are reconfigured. As we don't have any tooling to do that soft reconfiguration this will be an issue. I only see the following solution: instead of having only two states (input, output) plus the delayed power on (which is only a band-aid AFAICT), have three states (input, output, and high-Z on both sides: BNC and EEM). Then you can use the I2C register (it has several unused pins still) to enable the EEM after setting its channel configuration, or you can hard-enable it using a dip-switch when using the dip-switches to determine direction.
But let's delay that for v1.3 or later.

jordens, 2018-01-12:
@gkasprow otherwise the design looks good to me. Thanks

gkasprow, 2018-01-12:
LVDS is current source, so you can connect two LVDS outputs with no harm.

jordens, 2018-01-12:
Ok. And this true independent of power supplies (2.5 V vs 3.3 V) and even for the most extreme type1/type2 combinations?

gkasprow, 2018-01-12:
At the LVDS output there is just MOSFET bridge supplied by 2 current sources + 2 protection diodes. Providing that we connect 3.3V EEM output with 1.8V FPGA output, the 3.3V output common mode will be still within 1.8V range so the current will be limited to a few mA. The [datasheet](http://www.ti.com/lit/ds/symlink/sn65mlvd040.pdf) specifies the MLVDS output loaded with 50 Ohm as Vh = 1.3, Vl= 0.7V.

hartytp, 2018-01-12:
> @hartytp please make sure that the changes to the direction logic also work if the FPGA is unaware. I.e. if it always uses output channels as outputs as it does now.

We didn't make any changes to the direction logic in the end IIRC. Greg convinced me that the current setup is fine.

gkasprow, 2018-01-13:
@jordens there is no space for any washers between front panel and BNC connector body. So only nut is needed to fix it correctly. Can you confirm? I don't have any v1.1 DIO-BNC board to check it.

jordens, 2018-01-13:
https://github.com/m-labs/sinara/wiki/DIO_BNC
Confirm. Nut and spring washer IIRC.

closed 2018-01-13 by gkasprow

hartytp, 2018-01-14:
Thanks for doing that Greg! Hopefully, this is now a production quality run rather than prototyping.

@jordens @gkasprow When the new boards arrive, please can you have a quick look at the outputs on a scope and check that the noise issues are gone and that there are no turn-on glitches on the outputs.

It would also be good to try live direction switching a few times and check that this doesn't damage the board (just flip the switch ten times or so).