sinara-hw/sinara #242: LVDS DIO Errata

hartytp, 2017-07-24:
- [ ] Fix ordering of LEDs to match RJ45 connectors

hartytp, 2017-07-24:
@gkasprow Sorry for not picking up these issues in the previous design review!

gkasprow, 2017-07-24:
I noticed them doing panel design. This should have been done in rev1 but there is so much work...

hartytp, 2017-07-24:
No worries Greg -- you guys are tackling an intimidating number of hardware designs at the same time and making great progress, so a few minor things are bound to slip through!

These issues are only cosmetic and don't affect the boards' functionality. In any case, these boards are still prototypes, rather than production, so these kind of minor issues are expected :) 

At this point, I think the top priority is getting prototype hardware delivered to a few labs for thorough testing. We can always fix these small issues in rev2.

jordens, 2017-09-28:
What does the QC sticker actually mean? On my 3U_DIO_RJ45 two pins of the RJ45 cage were not soldered.

gkasprow, 2017-09-28:
@jordens could you take photo of this and send me? I'll ask Technosystem about it. QC means that it passed functional test.

jordens, 2017-09-28:
Out of curiosity: what functional tests are being done?

I already soldered it myself. Photos of that probably won't help. Bottom corner of the cage.

[image]

gkasprow, 2017-09-28:
We tested propagation of signal through all ports in both directions and power consumption. Technosystem does AOI tests so we will investigate it.

jordens, 2017-09-28:
Ok. Those tests sound good. Thanks!

marmeladapk, 2018-06-13:
_Migrated to sinara-hw/DIO_RJ45#1_

closed 2018-06-13 by marmeladapk