sinara-hw/sinara #243: RTM-SFP for Sayma

gkasprow, 2017-07-26:
I have interest from CERN about Sayma board. They want to use it as general purpose AMC carrier with plenty of (Q)SFP channels. So the idea is to build RTM with SFP cages.
Initial plan is to use 4x SFP + 3x QSFP to utilise all 16 gigabit channels. There are easily available QSFP to SFP+ copper cables ideal for low distance connections. 4 SFP cages let us use low cost WDM transceivers for DRTIO longer distance connections.
So the question is - would such board be attractive for ARTIQ users?
Exisitin project of 8 channel RTM SFP carrier could be adopted easily.

jordens, 2017-07-26:
Great to see the interest from CERN! Why not just 4xQSFP? Then the poor SMA connectors won't be so cramped.
I can't imagine a use case for it in Sinara/ARTIQ within the next couple years. Maybe if we get to 1k qubits ;)

gkasprow, 2017-07-26:
Guys want to connect at least 4x White Rabbit WDM SFPs.

hartytp, 2017-07-26:
Yes, AFAICT the main place we want SFPs is Metlino/Kasli. They're much less useful on Sayma.

gkasprow, 2017-07-26:
The same board can be used as Metlino extension

jbqubit, 2017-07-27:
ARTIQ uses bi-chromatic SFP modules as used by white rabbit. Unclear if we would use QSFP. 

ARTIQ use of Metlino RTM remains open. See #74.

hartytp, 2017-07-31:
Let's close this, as the relevant discussion is already in #74, as @jbqubit points out.

closed 2017-07-31 by hartytp