sinara-hw/sinara #245: ARTIQ build for Sinara testing based on Sayma core device

jbqubit, 2017-08-03:
@sbourdeauducq @gkasprow @hartytp  Proposed hardware configuration for testing Sinara without dependence on Metlino. Minimal setup consists of two Sayma_AMC + EEM peripherals.  The aim is to permit testing by several labs using M-Labs supported builds of ARTIQ. 

- Sayma core device
    - 1 Sayma_AMC  -- acting as ARTIQ core device
    - 2 SFP modules (front panel)
    - 2 SFP fibers to Sayma RF 1 and Sayma RF 2
    - 1 FMC-VHDCI mezzanine
    - ethernet interface vai SPF to RJ-45 (or SATA to RJ-45) 

- Sayma RF 1
    - 1 Sayma_AMC -- acting as Sayma peripheral 
    - 4 Allaki 
    - 1 FMC-VHDCI mezzanine
    - 1 SFP module (front panel) 

- Sayma RF 2 (optional) 
    - 1 Sayma_AMC -- acting as Sayma peripheral 
    - 4 Allaki 
    - 1 SFP module (front panel) 

- Eurocard Chassis
    - 1 VHDCI_breakout 
        - VHDCI 1 driven by core device
            - 1 BNC EEM 
            - 1 SMA EEM 
            - 1 RJ45 EEM 
        - VHDCI 2 driven by Sayma RF 1
            - 1 BNC EEM 
            - 1 Zotino EEM
            - 1 Novogorny EEM
            - 1 Urukul EEM

hartytp, 2017-08-03:
I'm hoping to use Kasli as the master instead of Sayma AMC, and to add few extra Kasli DRTIO slaves, for example for the NU-servo. But, other than that, this sounds like a good plan.

hartytp, 2017-08-03:
One question though: what's the issue/actionable item here? Is this a discussion we should have on the ARTIQ list?

jbqubit, 2017-08-03:
> I'm hoping to use Kasli as the master instead of Sayma AMC, and to add few extra Kasli DRTIO slaves

I'm planning on a different in-lab configuration too. A default hardware configuration for a couple weeks of testing will help track down bugs faster.

sbourdeauducq, 2017-08-04:
We will only support continuous integration, autobuilder, conda packages, etc. on the final system that has Metlino and a MicroTCA chassis. For intermediate hacks like this, please build from source.

sbourdeauducq, 2017-08-04:
And with the hack it is possible to have more satellite Saymas, Greg built a RTM-SFP adapter board that can be used to get more transceiver channels out of one Sayma.

jbqubit, 2017-08-04:
Let's discuss how to proceed on this at NACTI.

hartytp, 2017-08-20:
Joe, can we close this now?

closed 2017-08-24 by jbqubit